bigtreetech/BIGTREETECH-SKR-mini-E3 #614: [BUG] SKR MINI e3 v3 Controller Fan always on

Roonster78, 2022-02-01:
### Description

I have 3 machines with the SKR mini E3 v3.0 and the exact same firmware. After powering on the machines the controller fan of two of the machines stay silent unless a strepper gets activated. On one machine, the controller fan runs permanently and doesn't respond to any command.

Firmware is the actual Marlin 2.0.9.3 bugfix branch from the 31st January 2022

### Steps to reproduce

1. Switch on the printer
2. Check if the controller fan runs
2.1 try to control the speed in the menu (if configured)

**Expected behavior**
Should not run/spin like the other two machines

**Actual behavior**
Runs permanently unlike the other two machines

### Additional Information
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/7977777/Configuration.zip)

radek8, 2022-02-01:
Probably a damaged mosfet that spins a fan.
A frequent defect ....

mfaccin, 2022-03-04:
Same issue here, I can't control FAN_1 and FAN_2 using the gcode from Marlin. 
How can I address the configuration to have them set as P1 and P2? @radek8

radek8, 2022-03-05:
@mfaccin 
What board do you use?
Which fans do you want to control on connectors FAN1 and FAN2?
What do you mean P1 and P2?
Submit your configuration.

mfaccin, 2022-03-05:
@radek8 I'm using the SKR E3 mini V3.0

I intend to use two 5010 in FAN1 and FAN2 and an axial fan in FAN0.

According to the https://marlinfw.org/docs/gcode/M106.html I can select the speed of the FAN0 using `M106 P0 S_`, but it doesn't work with the other two.

radek8, 2022-03-05:
Do you use all 3 fans to cool the printed part?

radek8, 2022-03-05:
@mfaccin
Send me your configuration files (Configuration.h and Configuration_adv.h) from which you compiled fw.

radek8, 2022-03-05:
@mfaccin
If you want to control multiple fans with command M106 with index P0 to P2, you must define the number of fans.
#define NUM_M106_FANS 3

w7boz, 2023-07-21:
My SKR mini e3 v3 Fan 0 is always on. It has a bad Mosfet. How do I get it fixed or replaced???