bigtreetech/BIGTREETECH-SKR-mini-E3 #615: V3 X and Y DIAG PIN

Sallynator, 2022-02-05:
can someone tell me the pin name3s of the x and y diag pins for klipper configuration ?
ive been searching for hours and cant find anything...

radek8, 2022-02-05:
I found it in 3 minutes :-)

SKR-mini-E3-V3.0
#define X_DIAG_PIN                         PC0  // X-STOP
#define Y_DIAG_PIN                         PC1  // Y-STOP

[image]

radek8, 2022-02-05:
[image]

Sallynator, 2022-02-05:
thank u for this but it seems it wont accept those pins like this

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
diag_pin:PC0
microsteps: 16
rotation_distance: 40
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_endstop: 0
position_max: 235
homing_speed: 20
homing_retract_dist:0

did i do anything wrong ?

Sallynator, 2022-02-05:
[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
diag_pin:PC0
microsteps: 16
rotation_distance: 40
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_endstop: 0
position_max: 235
homing_speed: 20
homing_retract_dist:0

[tmc2209 stepper_x]
uart_pin: PC11
tx_pin: PC10
uart_address: 0
run_current: 0.580
stealthchop_threshold: 1
driver_SGTHRS: 50

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
diag_pin:PC1
microsteps: 16
rotation_distance: 40
endstop_pin: tmc2209_stepper_y:virtual_endstop
position_endstop: 0
position_max: 235
homing_speed: 20
homing_retract_dist:0

[tmc2209 stepper_y]
uart_pin: PC11
tx_pin: PC10
uart_address: 2
run_current: 0.580
stealthchop_threshold: 1
driver_SGTHRS: 50

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
# endstop_pin: ^PC2
# position_endstop: 0.0
endstop_pin: probe:z_virtual_endstop
endstop_pin: ^PC2
position_endstop: 0.0
position_max: 250

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: 2.0
stealthchop_threshold: 1

radek8, 2022-02-05:
I wrote you the pin number on the processor.
I use Marlin. I don't know the writing script for klipper.

radek8, 2022-02-05:
https://www.klipper3d.org/TMC_Drivers.html

do3cc, 2026-01-04:
I sincerely hope, @Sallynator found the solution years ago, but while double checking my configuration, I stumbled over this and thought to document the solution.

The pins names are correct and can be used like this in a Klipper config.
The example contains one mistake. The parameter `DIAG_PIN` belongs to the `tmc2209 stepper_x` section, not the `stepper_x` section.

Then a few words of warnings:
1. Your sensitivity value of `50` for parameter ` driver_SGTHRS` is rather low. For the `TMC2209` drivers on this board, the range is from `0` to `255` and it is best to start with the highest sensitivity of 255.
2. The manual of Klipper warns against using sensorless homing for the z axis. it is too unreliable and will likely damage your bed. This applies to both Ender style bed slingers as the typical CoreXY printers.

I followed the [Voron Sensorless Guide](https://docs.vorondesign.com/tuning/sensorless.html) which also works for Bedslingers.
Then the only `SKR mini E3` specific setting is the parameter `DIAG_PIN` in the `tmc2209 stepper_$x` settings:

```
...
[tmc2209 stepper_x]
...
DIAG_PIN: PC0
...
[tmc2209 stepper_y]
...
DIAG_PIN: PC1
...
```