bigtreetech/BIGTREETECH-SKR-mini-E3 #616: [BUG] BLTouch/CRTouch configuration issue in SKR-mini-E3-V3.0-klipper.cfg

markebert, 2022-02-07:
### Description

The provided SKR-mini-E3-V3.0-klipper.cfg file contains the incorrect settings for using a BLTouch/CRTouch as a virtual z endstop pin.

### Steps to reproduce

1. Attempt to home Z-axis while using a BLTouch or CRTouch without a Z endstop pin.

**Expected behavior**
The axis should home and deploy and utilize the probe to zero out the z axis.

**Actual behavior**
X and Y axis will home and then when the Z axis attempts to home the probe will not deploy, even though a self_test and pin_up and pin_down commands all work, and klipper will report back "BLTouch failed to verify sensor state" and the z axis will not be homed.

### Additional Information
The issue can be corrected by changing line 13 of the provided config file (SKR-mini-E3-V3.0-klipper.cfg) from "sensor_pin: PC14" to "sensor_pin: ^PC14".

tless1217, 2022-02-16:
I don't know the answer to this ,but how did you get kipper to work at all? Mine keeps getting an error

markebert, 2022-02-17:
Are you using an otherwise stock Ender 3/3Pro? Also are you using a BLTouch or CRTouch or just the default Z endstop?

markebert, 2022-02-17:
> 

Are you using an otherwise stock Ender 3/3Pro? Also are you using a BLTouch or CRTouch or just the default Z endstop?

markebert, 2022-02-17:
> I don't know the answer to this ,but how did you get kipper to work at all? Mine keeps getting an error

Are you using an otherwise stock Ender 3/3Pro? Also are you using a BLTouch or CRTouch or just the default Z endstop?

tless1217, 2022-02-17:
Im using a stock ender 3, im using a bl touch.

markebert, 2022-02-17:
> Im using a stock ender 3, im using a bl touch.

Here's my default printer.cfg. Hopefully it works for you. All you should need to do it set your z-offset (should be positive) and make sure your serial id is set correctly.

`# This file contains common pin mappings for the BIGTREETECH SKR mini
# E3 v3.0. To use this config, the firmware should be compiled for the
# STM32G0B1 with a "8KiB bootloader" and USB communication.

# The "make flash" command does not work on the SKR mini E3. Instead,
# after running "make", copy the generated "out/klipper.bin" file to a
# file named "firmware.bin" on an SD card and then restart the SKR
# mini E3 with that SD card.

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC0
#position_endstop: 0
#position_max: 235
position_endstop: 0
position_min: 0
# Use position_max: 210 for normal operation but position_max: 230 for bed calibration
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PC11
tx_pin: PC10
uart_address: 0
run_current: 0.580
stealthchop_threshold: 999999

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC1
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_y]
uart_pin: PC11
tx_pin: PC10
uart_address: 2
run_current: 0.580
stealthchop_threshold: 999999

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
#endstop_pin: ^PC2
#position_endstop: 0.0
endstop_pin: probe:z_virtual_endstop
position_min: -2
position_max: 250

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: 0.580
stealthchop_threshold: 999999

[safe_z_home]
home_xy_position: 150, 92.5 # Change coordinates to the center of your print bed
z_hop: 10 # Move up 10mm
z_hop_speed: 5

[virtual_sdcard]
path: ~/gcode_files

[respond]
default_type: command

[pause_resume]
recover_velocity: 50

# Pin mappings for BL_T port
[bltouch]
sensor_pin: ^PC14
control_pin: PA1
x_offset: -45
y_offset: -5
# CHANGE THIS TO WHATEVER Z OFFSET YOU NEED
z_offset: 0.0

[bed_mesh]
speed: 120
horizontal_move_z: 5
mesh_min: 10, 10
mesh_max: 185, 185
probe_count: 5,5

[screws_tilt_adjust]
screw1: 0,0
screw1_name: front left screw
screw2: 210,0
screw2_name: front right screw
screw3: 210,195
screw3_name: rear right screw
screw4: 0,195
screw4_name: rear left screw
horizontal_move_z: 10
speed: 50
screw_thread: CW-M4

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD1
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PC11
tx_pin: PC10
uart_address: 3
run_current: 0.650
stealthchop_threshold: 999999

[heater_bed]
heater_pin: PC9
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC4
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[heater_fan nozzle_cooling_fan]
pin: PC7

[heater_fan controller_fan]
pin: PB15
heater: heater_bed
heater_temp: 45.0

[fan]
pin: PC6

[mcu]
# This may need to be changed
serial: /dev/serial/by-id/usb-Klipper_stm32g0b1xx_5C002A000550415339373620-if00

[printer]
kinematics: cartesian
max_velocity: 200
max_accel: 2000
max_z_velocity: 5
max_z_accel: 100

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5,  EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PD6, EXP1_10=<5V>

# See the sample-lcd.cfg file for definitions of common LCD displays.
[display_status]

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2

[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
gcode:
    ##### set defaults #####
    {% set x = params.X|default(230) %}      #edit to your park position
    {% set y = params.Y|default(230) %}      #edit to your park position
    {% set z = params.Z|default(10)|float %} #edit to your park position
    {% set e = params.E|default(1) %}        #edit to your retract length
    ##### calculate save lift position #####
    {% set max_z = printer.toolhead.axis_maximum.z|float %}
    {% set act_z = printer.toolhead.position.z|float %}
    {% set lift_z = z|abs %}
    {% if act_z < (max_z - lift_z) %}
        {% set z_safe = lift_z %}
    {% else %}
        {% set z_safe = max_z - act_z %}
    {% endif %}
    ##### end of definitions #####
    PAUSE_BASE
    G91
    {% if printer.extruder.can_extrude|lower == 'true' %}
      G1 E-{e} F2100
    {% else %}
      {action_respond_info("Extruder not hot enough")}
    {% endif %}
    {% if "xyz" in printer.toolhead.homed_axes %}    
      G1 Z{z_safe}
      G90
      G1 X{x} Y{y} F6000
    {% else %}
      {action_respond_info("Printer not homed")}
    {% endif %}

[gcode_macro RESUME]
description: Resume the actual running print
rename_existing: RESUME_BASE
gcode:
    ##### set defaults #####
    {% set e = params.E|default(1) %} #edit to your retract length
    #### get VELOCITY parameter if specified ####
    {% if 'VELOCITY' in params|upper %}
      {% set get_params = ('VELOCITY=' + params.VELOCITY)  %}
    {%else %}
      {% set get_params = "" %}
    {% endif %}
    ##### end of definitions #####
    G91
    {% if printer.extruder.can_extrude|lower == 'true' %}
      G1 E{e} F2100
    {% else %}
      {action_respond_info("Extruder not hot enough")}
    {% endif %}  
    RESUME_BASE {get_params}

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
    TURN_OFF_HEATERS
    CANCEL_PRINT_BASE
`

markebert, 2022-02-17:
> Im using a stock ender 3, im using a bl touch.

Here's my default printer.cfg. Hopefully it works for you. All you should need to do it set your z-offset (should be positive) and make sure your serial id is set correctly.

[printer.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8085933/printer.zip)

R0BATZEN, 2022-02-20:
I have experienced the same problem upon first installing Klipper on this board.
After finding out what the issue was, I opened a [pull request](#612 ) to get the correct pin declaration (^PC14) changed in the default config file.
So far there has been no reaction from the devs that would implement my fix.

closed 2022-03-12 by bigtreetech

numindast, 2022-08-31:
Hello, this does not seem to be reflected in the current example printer config. I found markebert's solution to be what I needed after starting from the stock config for mini e3 v3 today and experiencing the exact same issue and the fix provided works. Thanks

Pd1099, 2023-01-23:
You saved me. Ive looked far and wide for an answer to fix my crtouch issue. This was it. Your the man!

Samedwin, 2023-02-27:
WOW!!!! THIS DID IT. I tried every configuration of all the suggested fixes
In my printer.cfg klipper file - EVERY combination of true/false and output modes - I tried ALL the Klipper suggested fixes, this is 2^4 which was 16 combinations
I tired all of these, and every combination of all of these - very tedious! And your ^ suggestion fixed it!! I was naive in thinking that !PC14 did the same thing as ^PC14..... NOPE! UGH!!!!
Here's all the suggestions Kliiper made (only "pin_up_reports_not_triggered: True" - which WAS necessary afterall
I tired TF for all of these:
pin_up_reports_not_triggered:      
pin_up_touch_mode_reports_triggered:
set_output_mode
probe_with_touch_mode

The FIRST thing Klipper needs to check is that the Z axis is homing in the right direction. unfortunately you can't check Z until you home it annoying... but ... that was my first issue of the probe not reporting state error. That took me a half dozen hours to realize z was moving up instead of down

after I realized that - I was able to diagnose that NONE of the documentation out there has the^ suggestion. I tried ! in front of the pin, but that didnt do it.

If you ever want to get a BL Touch or CR Touch working on SKR Mini E3 V3 - you HAVE TO USE THIS CONFIGURATION FOR THE BL Touch section of printer.cfg:

[bltouch]
sensor_pin: ^PC14
control_pin: PA1
x_offset: -40
y_offset: -30
z_offset: 0.3
pin_up_reports_not_triggered: True

Good luck out there!

ydooWoody, 2023-04-27:
> If you ever want to get a BL Touch or CR Touch working on SKR Mini E3 V3 - you HAVE TO USE THIS CONFIGURATION FOR THE BL Touch section of printer.cfg:
> 
> [bltouch] sensor_pin: ^PC14 control_pin: PA1 x_offset: -40 y_offset: -30 z_offset: 0.3 pin_up_reports_not_triggered: True
> 
> Good luck out there!

Man. Ive been struggling for like 12 hours over here. I dont know how to repay you but I owe you

Xcion54, 2023-06-19:
> ### Description
> The provided SKR-mini-E3-V3.0-klipper.cfg file contains the incorrect settings for using a BLTouch/CRTouch as a virtual z endstop pin.
>
> ### Steps to reproduce
> 1. Attempt to home [...]

This Worked Perfectly thank you!!!!!!!!!!

sdylewski, 2024-11-01:
works for me to.  thanks!  Disappointing that the Klipper files for the BTT skr mini e3 STILL haven't been updated to be correct.

Trig0r, 2024-12-25:
This doesnt work for me.
I've tried the BLT on the Zstop pins as well as the correct 2 of the 5 pins on the SKRe3v3 board, pin goes up and down as it should when you test it, but when you home all 3, G28, I get failed to verify sensor state.

PR #612 (2022-03-12): Fix pin declaration for bltouch
### Description
Adding '^' enables the built-in hardware pull-up resistor for the pin. 

### Benefits

Using '^' will make sure the bl touch functions as expected.
Without it the sensor will not be able to detect anything.

### Related Issues
This fixes #616