bigtreetech/BIGTREETECH-SKR-mini-E3 #618: [BUG] M1 Stop , print paused

Seelenkind, 2022-02-08:
### Description
I have successfully operated an SKR Mini E3 V2.0 with a TFT E3 V3.0 on an Ender 3 with BL Touch for months.
After converting to an SKR Mini E3 V3.0, I tried flashing the preconfigured firmware from Github. The BL Touch is plugged into the probe slot. When that particular firmware didn't work, I plugged the BL Touch's two wires into the Z endstop and flashed the other firmware that was specifically made for it. Again no success. Unfortunately, a quick test was not possible.
So I configured and flashed the firmware myself. Now everything works as it should be. On this occasion I flashed the current firmware for the TFT display and I was happy that everything went well and is up to date.
The joy didn't last long. During a longer print I came to the printer and noticed that the print was paused with an M1 stop and the display asked me to press the button to continue. The nozzle had melted into the component and the print was ruined. The print was made from the SD card slot on the TFT display. The error does not appear when printing from the SD card slot on the board.
I then did a little research and came across indications that machine codes that begin with M1.. (e.g. M117) may not arrive completely at the printer and only an M1 is evaluated.
I helped myself in this regard by deactivating M0 & M1 in the Marlin firmware. Even if an M0 or M1 is sent to the printer by mistake, it is not evaluated and a "Command unknown" message appears on the TFT display.
However, this is only a temporary solution and should not remain so in the long term.
Is this problem generally known or does it only appear on a few devices?
Soon I want to flash an older firmware on the TFT display and try out whether the cause is the update to the display or whether the board tends to errors when receiving serial data.
Does anyone have a final solution?

### Steps to reproduce

randomly during print

### Additional Information
config & binary at : https://github.com/Seelenkind/Ender-3-SKR-Mini-E3-V3-Marlin-Firmware

[image]
[image]

radek8, 2022-02-08:
Yes
the problem you described has recently started to appear on printers based on SKR 2 and SKR mini E2 V3.0
Apparently this is related to the use of the processor.

For someone, it helped fix the size of the Marlin buffer.
Someone was helped by a fix prepared by @kisslorand
PR is ready. Awaiting Bigtreetech approval.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2365

You can read more about the problem and how to edit the cache here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2295

Seelenkind, 2022-02-08:
OK. Thanks for the answer. I think I've stumbled across these posts before and can remember some of the details.
January 2022 bugfix is now master. But it still gives me the same error.
I'll try the [Buffersize ](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2295#issuecomment-1000204084)settings now. If that doesn't help then maybe an older firmware from the TFT display. January 2021 maybe. Otherwise, uncommenting M0_M1 in gcode.cpp helps.

kisslorand, 2022-02-08:
If you come across new information please move the discussion to the above mentioned topic as it is monitored by more people and it is more centralized. Anyway the, solutions given, even the PR @radek mentioned just avoid the bug. The bug has been traced to Marlin, but that's where we hit a wall.

Seelenkind, 2022-02-09:
> If you come across new information please move the discussion to the above mentioned topic as it is monitored by more people and it is more centralized. Anyway, the solutions given, even the PR @radek mentioned are not a fix, they just avoid the bug. The bug has been traced to Marlin, but that's where we hit a wall. Side note: I also have my M0/M1 disabled in Marlin, most probably I will never use them.

To be on the safe side, M0_M1 is still disabled in Marlin.
I changed buffer size and flashed firmware again. Additionally flashed an older firmware for the TFT display.
Yesterday on a 10 hour press 2 x M1 stop on the display. (no improvement)
Today I dismantled the power supply and reassembled it because I had to add something.
Now a 12 hour print is running and so far 8.5 hours no M1 on the display.
There is probably no system behind it.
OK There has already been a major change: the printer is no longer in the kitchen, fairly close to the refrigerator, but in the study next to the computer.
Any interference caused by the refrigerator?
The next time I press I can change the location.
Or did I have a problem with the power supply that might have been fixed?
I dont know.

radek8, 2022-02-12:
Please close the duplicate ticket

Seelenkind, 2022-02-12:
> Please close the duplicate ticket

I have tried countless firmwares that have been suggested here. On both sides. Both on the TFT display and on the board. And none helped. I've definitely had the problem with M1 Stop since I installed the SKR Mini E3 V3.0.
I stick to my solution that I deactivated M0_M1 in the firmware. In addition to that, I will in future do all the preparatory measures for printing on the TFT display and the actual printing from the SD card on the Boar in Marlin mode.
Once the print has started, it runs through to the end without any problems. So I use the possibility and comfort of the TFT display.
So the post can be closed here without a concrete solution so far.

closed 2022-02-12 by Seelenkind

rasty777dev, 2022-02-13:
Same thing for me, also with SKR Mini E3 V3.0 - in companion with Octoprint from raspberry. Any soultions now??

radek8, 2022-02-13:
Have you tried applying corrective PR?
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2365

Seelenkind, 2022-02-13:
> Same thing for me, also with SKR Mini E3 V3.0 - in companion with Octoprint from raspberry. Any soultions now??

Edit Marlin gcode.cpp , uncomment M0 M1 .

Seelenkind, 2022-02-13:
> Have you tried applying corrective PR?
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2365

Issue with octoprint top. Not TFT Display.

rasty777dev, 2022-02-13:
> Edit Marlin gcode.cpp , uncomment M0 M1 .

You mean "comment" :) I did it, no problems for now, but is stron random so it's not a sure thing for now...

rasty777dev, 2022-03-01:
Any news? Comment M0 M1 seems to be working, no STOP since that, but it's not a solution...

Kareareawbg, 2022-03-21:
Still have m1 m0 m21 errors randomly. Used cat6e as my cable of choice with foil wrap and a choke and still no joy. 
Works as long as I dont use the touch function or the usb ports on the screen. Can use emulated 12864 and the controller boards sd slot and zero errors.

Random stops, crashes, and pauses for the tft. Wonder if its cause the Coretex M0 offloads some of its tasks to another chip and a function call is missing

Mech131, 2022-06-09:
I'm also having this issue. Is there any solution for it?

chezzandyto, 2022-07-12:
Any news from BTT ? it seems can be a hardware problem. I have two SKR MINI E3 V3, both of them with Random M1 pause.

Qplaze, 2022-08-07:
Hello. 

This is of course a crutch and not a solution. But I noticed: you commented out the code incorrectly. You do not need to comment on the entire processing of the M0 and M1 commands. Since in this case it is treated as unknown.

You just need to remove the processing of the M0 and M1 commands using the M0_M1(); function; which displays a message, suspending work until the user reacts.

Change this string: 
case 1: M0_M1(); break; 

to:
case 1: break; 

or if you want keep this piece of code to

case 1: 
//M0_M1(); 
break; 

You can check that everything works (at least command processing) by yourself by sending the M1 command through the terminal before and after changing the code.

Seelenkind, 2022-10-11:
Comment out M0 in firmware is the only working solution for me.

ricthot, 2023-01-27:
Just FYI - I'm on a Mini E3 v1.2 and also encountering this issue, so it is not specific to the V3.0 release of the board.