bigtreetech/BIGTREETECH-SKR-mini-E3 #619: [BUG] Z-Homing with CR-Touch and skr mini e3 V3 board not working

Loifu, 2022-02-08:
### Description
CR-Touch doesnt Auto home properly using the marlin firmware.
I already tried adjusting some settings in the firmware.

defining USE_PROBE_FOR_Z_HOMING and BLTouchv 

### Steps to reproduce

1. [Install SKR Mini E3 V3.0 in Ender 3 Pro with CR-Touch
2. Install [firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin)
3. Auto-Home

**Expected behavior**
Z-Ho
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8026422/Configuration.zip)
ming success

**Actual behavior**
CR-Touch triggers but z-axis doesnt stop and rams into the print bed.

### Additional Information
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8026422/Configuration.zip) This is my Configuration file.

radek8, 2022-02-08:
Do you have the cables connected correctly?
Send engagement photos.

Loifu, 2022-02-08:
i think i did it righ
[image]
t

radek8, 2022-02-08:
You didn't take a picture of the other side.
However, if the color resolution of the standard (it is not usually provided by products from China), you have it badly connected.
Send a photo of the sensor connection.

Seelenkind, 2022-02-08:
> Do you have the cables connected correctly? Send connection photos

I had the same problem with an Ender 3 (Non Pro).
The firmware on GitHub BTT is not working properly.
Neither the firmware where the BL Touch connects to Probe nor where it connects to the Z Endstop.
I also tried to use the firmware in reverse. Also no success.
The preconfigured firmware already breaks off when homing with BL Touch.
So I downloaded Marlin 2.0.9.3, added the example config files from Marlin and edited everything that BL Touch needs.
Now everything works as it should be. The BL Touch is connected to Probe with all cables. Nothing is connected to the Z Endstop.
Bug with M0_M1 is also temporarily disabled if someone wants to use it with the TFT display.

https://github.com/Seelenkind/Ender-3-SKR-Mini-E3-V3-Marlin-Firmware

radek8, 2022-02-08:
CR-touch apparently supplies probes with a rotated connector
see
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/145

Loifu, 2022-02-09:
[image]
@Seelenkind your firmware sadly doesnt work aswell. I still have the same issue over and over. I also tried re wiring the sensor but then nothing worked neither the led in there turned on but maybe i did something wrong. Thanks for your help

radek8, 2022-02-09:
I tracked the video connection.
https://youtu.be/puAvkQAjRak
The cables are fine. The photos on the internet are wrong.
Maybe the CS-Touch is not compatible with the BL-touch?

radek8, 2022-02-09:
uncomment
#define USE_ZMIN_PLUG

Loifu, 2022-02-09:
Sadly this still doesnt help. What i noticed is that the LED inside is turned on normally but when i try to auto home it the LED turns off

radek8, 2022-02-09:
Does it respond if you select extend and retract the probe from the menu?

radek8, 2022-02-09:
You can try the marlin set up

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
// # define USE_PROBE_FOR_Z_HOMING

And connect the two wires to the Z-Stop connector.

[image]

radek8, 2022-02-11:
Have you tried my last piece of advice?

Loifu, 2022-02-11:
I didn’t have time yet I’ll do it tomorrow

radek8, 2022-02-12:
According to the diagram, pin PC14 does not contain a physical PULLUPS resistor and probably pin PC14 does not support a software PULLUPS resistor.
If it does not contain a CR-Touch resistor, this may cause your problem.
Pin PC2 is equipped with a physical resistor and will not have this problem.

If that were the case, the second solution would be to solder the physical resistor to the board to Pin PC14.

[image]

radek8, 2022-02-13:
Did you change the configuration to marlin?

radek8, 2022-02-13:
can you comment on this line?
//#define Z_AFTER_HOMING

radek8, 2022-02-13:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/623

Seelenkind, 2022-02-23:
> May I check with you that your post firmware suitable for E3 2.0 on Ender 5 with CR touch?
> Thanks and have a nice day.

@chenwenhsi 
Unfortunately I can't say that. I only have the Ender 3 and unfortunately I'm not familiar with the mechanics and end stops of the Ender 5.

chenwenhsi, 2022-02-23:
> 

Thank you, I will try to modify the *.h files and compiler them for Ender 5. -Roger

LiFeTecX, 2022-03-01:
> I had the same problem with an Ender 3 (Non Pro). The firmware on GitHub BTT is not working properly. Neither the firmware where the BL Touch connects to Probe nor where it connects to the Z [...]

indeed this seems to be working for the BL-touch except for the motherboard fan this seemt to make a tikking noice but never starts spinning an idea how to fix this?

MavrEars, 2022-03-03:
Just compiled working firmware for BIGTREETECH-SKR-mini-E3 and CR TOUCH, check it out
[BTT SKR MINI E3 V3.0 CRTOUCH.bin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8181702/BTT.SKR.MINI.E3.V3.0.CRTOUCH.bin.zip)

radek8, 2022-03-04:
Connect an electrolithic capacitor 1uF to the ventilator

exabrial, 2022-03-27:
I have same issue with the Ender 3, SKR Mini 2.0. 

* Works with this firmware: https://www.danbp.org/p/en/node/149
* Does not work with: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch-for-z-homing.bin

dotWee, 2022-06-17:
> 
> Thank you, I will try to modify the *.h files and compiler them for Ender 5. -Roger  

@chenwenhsi did you get it to work? Would you mind sharing your modifications?

valentijnvh, 2022-07-16:
Is there a working precompiled binfile for ender 5 pro and cr-touch?
The one on the website is not werking for me.

enoy19, 2022-08-01:
I surrendered and removed the CR-Touch again and compiled Marlin 2.0.9.5 using the Ender 3 Pro example config (config/examples/Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 3.0).
Marlin let's you Level 9 points of the bed with babysteps and adjusts the print automatically (as far as I know).
It works perfectly - at least for my standards.

This might be an alternative solution until this issue is fixed.

jakedoesgithub, 2022-08-03:
This answer pertains to the V2.0 boards, but if you can find the release of marlin this  firmware corresponds to and find a v3.0 firmware from the same release you have a decent shot of fixing the issue.

"Hi it’s not you it’s the recompiled firmware it’s incorrect here are the older btt version that work https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0"

dotWee, 2022-08-04:
I actually got it working without any issues.

I've used a prebuilt binary from another GitHub repo, but don't remember the name right now.

I'll try to find it and post the link this weekend.

Ak3dprinting, 2022-08-11:
> I actually got it working without any issues.
> 
> I've used a prebuilt binary from another GitHub repo, but don't remember the name right now.
> 
> I'll try to find it and post the link this weekend.

Any luck?

jakedoesgithub, 2022-08-12:
Ya i found one that was an old version of the firmware and it worked.

dotWee, 2022-08-12:
@TheShield167 @Ak3dprinting Yes, check out @Seelenkind 's BTT E3 firmware fork! Pretty sure I'm using his provided binaries for the Ender 5 (Pro), an it's working flawlessly.

HundePupu, 2022-08-21:
> @TheShield167 @Ak3dprinting Yes, check out @Seelenkind 's BTT E3 firmware fork! Pretty sure I'm using his provided binaries for the Ender 5 (Pro), an it's working flawlessly.

Hey can you upload the finished firmware for the ender 5 Pro?
Please

jakedoesgithub, 2022-08-22:
I have the Ender 3 Pro, the fix was to find an older version of the firmware, Marlin [2.0.7.2], to use. So if I were you, I'd look for a version of the Ender 5 firmware that uses Marlin [2.0.7.2] or something close to that. If you go > 2.0.7.2 and it doesn't work, then try going < 2.0.7.2 instead. That version may be the very first one that the user found that worked, so versions above it may not work correctly.

The issue is with the marlin firmware in the newer versions. So an old one that works is your best bet.

dotWee, 2022-08-24:
@HundePupu there's already a precompiled firmware binary for the Ender-5 Pro available in his repo: https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/Ender%205%20pro/firmware.bin  

The firmware I'm using has only some minor & personal adjustments from @Seelenkind's Ender-5 Pro codebase, as it's running on an early Ender-5 frame without the "Pro" upgrades.

Also I agree with @jakedoesgithub, building your own firmware really isn't that hard! There are a lot feature gems available to explore too.

prazy, 2022-09-04:
> I have the Ender 3 Pro, the fix was to find an older version of the firmware, Marlin [2.0.7.2], to use. So if I were you, I'd look for a version of the Ender 5 firmware that uses Marlin [2.0.7.2] or [...]

Hey Jake, I have an Ender 3 pro that I’m switching to SKR mini v3 with CR touch. Would you be able share the firmware you compiled? Thanks!

tom1422, 2022-09-10:
I had the exact same issue as OP but I managed to get it working by downloading marlin bugfix 2.1.x. (I'm using ender3 regular)
Install Auto Build Marlin and Platformio in vscode. Set working directory in ABM to master directory of marlin.
Followed this guide EXACTLY for setting up firmware: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/
In the `@section probes` bit (from article above), I found that these options worked:
`
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
...
#define USE_PROBE_FOR_Z_HOMING
...
#define Z_MIN_PROBE_PIN PC14
...
#define BLTOUCH
`
Then compile and it should work. Setting it to use 5V as well didn't seem to make any difference but didn't destroy the crtouch.
Also make sure if you are using the extruder fan header (and not always on) to set this in configuration_adv.h: (Guide doesn't say to)
`#define E0_AUTO_FAN_PIN PC7`

prazy, 2022-09-11:
> I had the exact same issue as OP but I managed to get it working by downloading marlin bugfix 2.1.x. (I'm using ender3 regular) Install Auto Build Marlin and Platformio in vscode. Set working directory in ABM to master directory of marlin. Followed this guide EXACTLY for setting up firmware: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/ In the `@section probes` bit (from article above), I found that these options worked: `//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN ... #define USE_PROBE_FOR_Z_HOMING ... #define Z_MIN_PROBE_PIN PC14 ... #define BLTOUCH` Then compile and it should work. Setting it to use 5V as well didn't seem to make any difference but didn't destroy the crtouch. Also make sure if you are using the extruder fan header (and not always on) to set this in configuration_adv.h: (Guide doesn't say to) `#define E0_AUTO_FAN_PIN PC7` Also don't enable `PINS_DEBUGGING` for some reason it gives a compile error.
> 
> And another option I would be weary of is `S_CURVE_ACCELERATION` i enabled it and it made it go really slow before stopping. Probably needs extra configuration but I would leave it disabled.

I just compiled Marlin bugfix 2.1.x for my Ender 3 Pro and follow the guide you posted and with your recommendations. everything turned out perfect. Thanks for the suggestion.

jakedoesgithub, 2022-09-12:
> > I have the Ender 3 Pro, the fix was to find an older version of the firmware, Marlin [2.0.7.2], to use. So if I were you, I'd look for a version of the Ender 5 firmware that uses Marlin [2.0.7.2] or something close to that. If you go > 2.0.7.2 and it doesn't work, then try going < 2.0.7.2 instead. That version may be the very first one that the user found that worked, so versions above it may not work correctly.
> > The issue is with the marlin firmware in the newer versions. So an old one that works is your best bet.
> 
> Hey @jakedoesgithub , I have an Ender 3 pro that I’m switching to SKR mini v3 with CR touch. Would you be able share the firmware you compiled? Thanks!

hey this is the one i used I think. https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0 
Please note that this is 2.0. I dunno if they have a 3.0 of the same type or not.

xGloriousLeader, 2022-12-23:
> I just compiled Marlin bugfix 2.1.x for my Ender 3 Pro. I followed the guide you posted and configured based on your recommendations. Everything turned out perfect. Thanks for the suggestion.

Could you share your firmware? I must be doing something wrong. I've followed the guide twice and mine is still not working.

mentat6059, 2023-01-03:
I just downloaded the th3d marlin for an ender 3 and the SKR mini v3 and made these changes for the CR Touch to work (the CR touch wiring is correct and the same as the BL touch wiring - I have both)

configuration.h
#define SERVO0_PIN PA1
#define BLTOUCH_ON_5PIN
#define Z_MIN_PROBE_PIN PC14

then had to finesse the right margin in configuration_adv.h
#define PROBING_MARGIN_RIGHT 50

the only issue I found is when setting the z offset it is incorrect so I just kept it at 0 and then used babysteps to adjust it while doing a bed leveling print 
remember to set this in configuration.h
#define FINE_BABYSTEPPING

stiliajohny, 2023-11-24:
> I had the same problem with an Ender 3 (Non Pro). The firmware on GitHub BTT is not working properly. Neither the firmware where the BL Touch connects to Probe nor where it connects to the Z [...]

Will this work with CR touch ?

Lanmi369, 2024-04-02:
I have a problem with the printer. I bought a new motherboard for it SKR Mini E3 v3.0. And I had cr touch installed on it, now when I start to print something, the height between the nozzle and the plate constantly changes. And if I level it, it starts as it should, it always moves next time.
[image]
[image]