bigtreetech/BIGTREETECH-SKR-mini-E3 #623: [FR] SKR Mini E3 V3.0 no Homing with BL Touch and M1 Stop issue

Seelenkind, 2022-02-12:
Many Ender 3 users with the new board SKR Mini E3 V3.0 have the problem that the officially released firmware does not work.
2 Big problems arise. If BL Touch is plugged into the probe port and the associated firmware is flashed, homing does not work.
I edited Marlin 2.0.9.3 for the board and built and released a firmware that works for me.
The second problem is that an M1 stop error appears sporadically in connection with the TFT display. Printing is paused and waiting for user input. The problem could not be finally solved but there is a temporary solution. The parsing of the M0 M1 code is commented out in the gcode.cpp file and is therefore unknown to the firmware M1. The M1 Stop message no longer appears on the display. Instead, the message M1 unknown appears. But the pressure keeps going.

It would be a great help if the released firmware was officially changed in this regard so that such problems do not lead to frustration in the future.
My config files and the firmware can be found under the following link.
https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3
It would be great if many would deal with the problem and both problems are fixed with the next firmware update.

[image]

radek8, 2022-02-13:
What changes did you make to get the probe up and running?
Multiple users of this board have the same problem with the probe.
Surely they would welcome your help.

Seelenkind, 2022-02-13:
> What changes did you make to get the probe up and running? Multiple users of this board have the same problem with the probe. Surely they would welcome your help.

comment this line ,  z after Homing 
https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/blob/d3f532796dff4cc58291bf5f881a7ae3f83e7097/firmware/V3.0/Configuration.h#L1369

frankeyhe, 2022-02-14:
> > What changes did you make to get the probe up and running? Multiple users of this board have the same problem with the probe. Surely they would welcome your help.
> 
> comment this line : //#define Z_AFTER_HOMING 10 // (mm) Height to move to after homing Z https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/blob/d3f532796dff4cc58291bf5f881a7ae3f83e7097/firmware/V3.0/Configuration.h#L1369
> 
> i think it makes problem with this line : #define Z_CLEARANCE_DEPLOY_PROBE 10 // Z Clearance for Deploy/Stow and #define Z_PROBE_OFFSET_RANGE_MAX 20
> 
> 10+10 =20 and max
> 
> https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/blob/d3f532796dff4cc58291bf5f881a7ae3f83e7097/firmware/V3.0/Configuration.h#L1261

I tried your firmware for my ender pro v1, it got the bed leveling and auto home working but I am running into power loss detected and am unable to print anything, have you ran into this issue?

Seelenkind, 2022-02-14:
>> I tried your firmware for my ender pro v1, it got the bed leveling and auto home working but I am running into power loss detected and am unable to print anything, have you ran into this issue?

in the Marlin firmware "resume after power loss" is possible but not active.  So it can't be the board firmware.  The option is also activated in the TFT display firmware.  Please check if you can disable it on the TFT display.  You probably have voltage fluctuations that trigger this error.
Try to Print fromm Board in Marlin Mode to Test ist.

Favmir, 2022-02-17:
I spent the entire yesterday trying to figure out what was wrong, narrowed it down to the firmware not being able to handle zprobe port input — the zstop has to be physically plugged in or the printer would freeze at G28 or G29.

……I should've looked into this github first. Looks like I'll have to build a custom firmware. Thanks for your help.

Favmir, 2022-02-25:
Uodate: i fixed my issue a few days ago but didn't update on this thread. My symptom varied depending on the wiring, sometimes like this: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/593#issue-1088643236
Or just not working at all.

My conclusion was that the firmware expected you to use zstop, which I pulled out from the board because Crealty's firmware works without it. The standard Skr mini e3 Bltouch firmware expects you to use BOTH the zstop and bltouch, and the 'use z probe as zstop' firmware expects you to plug in bltouch into zstop port(doesnt work if you use bltouch port).

I just found the zstop switch and re-installed it, and the printer has been working fine.

LiFeTecX, 2022-03-02:
Seelenkind i am using your firmware (https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3) but the motherboard fan doesnt work, it seems to make a tikking noice but never starts spinning an idea how to fix this?

i am able to print now woth the BLtouch but am scared my motherboard wil overheet.

Seelenkind, 2022-03-02:
> Seelenkind i am using your firmware (https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3) but the motherboard fan doesnt work, it seems to make a tikking noice but never starts spinning an idea how to fix this?
> 
> i am able to print now woth the BLtouch but am scared my motherboard wil overheet.

Where did you connect the board fan?
Fan1 >> HotEnd cooler
Fan2 >> board fan
Fan2 >> PWM for Filament fan

[image]

LiFeTecX, 2022-03-02:
> > Seelenkind i am using your firmware (https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3) but the motherboard fan doesnt work, it seems to make a tikking noice but never starts spinning an idea how to fix this?
> > i am able to print now woth the BLtouch but am scared my motherboard wil overheet.
> 
> @LiFeTecX Where did you connect the board fan? Fan1 >> HotEnd cooler Fan2 >> board fan Fan0 >> PWM for Filament fan
> 
> I connected the board fan & HodEnd fan together to Fan1. Both start at a nozzle temperature of 50 degrees and turn off when the temperature drops below it again.
> 
> [image]

I connected the motherboard fan to fan2, filament to fan1 and hotend to fan0 with the btt firmware this was working like intended but then the bl touch didint work. With your firmware the bl touch works but one fan doesnt.

edwardryan321, 2022-03-03:
> I connected the motherboard fan to fan2, filament to fan1 and hotend to fan0 with the btt firmware this was working like intended but then the bl touch didint work. With your firmware the bl touch [...]

Same here. I didn't get around to trying the latest release. I tried it almost a month ago.
With the official firmware, BLTouch wasn't working but the board fan started whenever step motors were in action. When I used @Seelenkind's firmware, BLTouch worked but the board fan wouldn't come on.

Seelenkind, 2022-03-03:
I checked the config files again and recompiled them. I uploaded the new file.
The following is stored in [the original pin file](https://github.com/bigtreetech/Marlin/blob/e5347ab9d51d85073e2c53298f6a852cbaf60e65/Marlin/src/pins/stm32g0/pins_BTT_SKR_MINI_E3_V3_0.h#L141). 
Please switch accordingly.
Fan0 >> Pin PC6 >>> PWM for Filament fan
Fan1 >> Pin PC7 >>> Hot end cooler
Fan2 >> Pin PB15 >>> Board fan

LiFeTecX, 2022-03-03:
> I checked the config files again and recompiled them. I uploaded the new file. The following is stored in [the original pin file](https://github.com/bigtreetech/Marlin/blob/e5347ab9d51d85073e2c53298f6a852cbaf60e65/Marlin/src/pins/stm32g0/pins_BTT_SKR_MINI_E3_V3_0.h#L141). Please plug. Fan0 >> Pin PC6 >>> PWM for Filament fan Fan1 >> [Pin PC7](https://github.com/bigtreetech/Marlin/blob/e5347ab9d51d85073e2c53298f6a852cbaf60e65/Marlin/Configuration_adv.h#L588) >>> Hot end cooler Fan2 >> [Pin PB15](https://github.com/bigtreetech/Marlin/blob/e5347ab9d51d85073e2c53298f6a852cbaf60e65/Marlin/Configuration_adv.h#L502) >>> Board fan

thank for trying to fix it so fast i flashed the new firmware but the issiue presists.

Seelenkind, 2022-03-03:
My print will be finished in 6 hours. I will flash the firmware again and test the fans.
It's currently working for me with no problems. As I said, I connected the board fan and hotend fan together to PC7 (Fan1) and they turn on when the nozzle temperature is over 50 degrees and turn off when the temperature drops below 50 degrees. Nothing is connected to Fan2.
I'll report back after testing.

Seelenkind, 2022-03-03:
I tested everything again. Have the same firmware on the board. My filament cooler is plugged into Fan0. My HotEnd fan and board fan are connected in parallel to Fan1. Both go on when the nozzle temperature is greater than 50 degrees.

[image]

https://user-images.githubusercontent.com/75549196/156597472-6bd9dcb4-3ee2-49bd-8349-8c8460ebf1da.mp4

Seelenkind, 2022-03-03:
Then I switched both fans to Fan2 and tested whether the board fan starts up when the axes move.
yes they do
[image]

https://user-images.githubusercontent.com/75549196/156598023-8dd99cf2-73ef-453c-8a44-5b6f1ca6fd58.mp4

LiFeTecX, 2022-03-04:
Thank you for trying but it stil doesnt workt for me. i did what you did and connected both hotend and motherboard to the same working connection and now they both spin when the hotend is 50 degrees.

Seelenkind, 2022-03-04:
> Thank you for trying but it stil doesnt workt for me. i did what you did and connected both hotend and motherboard to the same working connection and now they both spin when the hotend is 50 degrees.

@LiFeTecX 
I updated the firmware and recompiled. Did you download and flash it again?

https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0

EduMota, 2022-03-05:
I'was having the same issue, but no need to coment M0 and M1 in gcode.cpp, actually I think this is unsafe, I just commented the line `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` in Configuration.h and uncommented `USE_PROBE_FOR_Z_HOMING`, the problem is with `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` it forces the a [if macro](https://github.com/bigtreetech/Marlin/blob/e5347ab9d51d85073e2c53298f6a852cbaf60e65/Marlin/src/module/probe.h#L42) in probe.h to uses the `Z_STOP_PIN` instead `Z_MIN_PROBE_PIN`, and now it work fine.

edwardryan321, 2022-03-05:
I just did a test. I loaded up the BTT firmware firmware-ender3-bltouch-for-z-homing.bin. I use the controls to Move X 10mm. The board fan comes on instantly.
Then I loaded up the recompiled firmware discussed here. When I Move X 10mm, the board fan does not move but I hear a clicking coming from the board that only started when I tried to move X.

If I have to wire up the board fan to hot end fan, I'll do it. It's a small price to pay to have the BLTouch working.

Seelenkind, 2022-03-05:
> I was having the same issue, but no need to coment M0 and M1 in gcode.cpp, actually I think this is unsafe, I just commented the line `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` in Configuration.h and uncommented `USE_PROBE_FOR_Z_HOMING`, the problem is with `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` it forces an [if macro](https://github.com/bigtreetech/Marlin/blob/e5347ab9d51d85073e2c53298f6a852cbaf60e65/Marlin/src/module/probe.h#L42) in probe.h to uses the `Z_STOP_PIN` instead `Z_MIN_PROBE_PIN`, and now it work fine.

I read somewhere here in the posts that just leaving the hardware Z endstop plugged in worked.
Presumably this if macro requires what you mentioned, that a limit switch is connected in the Z endstop.

Seelenkind, 2022-03-05:
OK. The whole thing gave me no peace and I started from the beginning.
My firmware is/was original marlin 2.0.9.3with sample configuration files from the original Marlin GitHub page.
Unfortunately, what worked for me didn't work for you.
Now I have followed the hint that BTT has provided an E3 V3 example configuration. Of course, this does not include the BL Touch integration. So stock firmware.
I forked this firmware and added BL Touch.
BL Touch is connected to probe port with all 5 wires. Nothing is connected to Z Endstop.
I just compiled the firmware and tested it successfully.
Now what is so different about this specially adapted Marlin version?
I don't know myself but there are many changes to different files. Some options normally found in Marlin 2.0.9.3 are missing here but are not important.
I hope now that this firmware solves the problem with the fans.
Many many changes in the firmware that definitely have a meaning. Maybe it's an adapted bugfix version?
You can recognize the files I have changed by the description Compiled by Seelenkind.

[image]
[image]
[image]

>> [New Compiled Marlin ](https://github.com/Seelenkind/Marlin/tree/SKR-mini-E3-V3.0-G0B1/Marlin)<< 
Please rename it to firmware.bin after downloading before flashing.

edwardryan321, 2022-03-06:
I hear clicking but this time I see the fan trying to move. I compared your changes to the bugfix code (I missed their Main source code in here link). 

I wonder if these fan changes at the bottom of Configuration.h has anything to do with it. Previously, you had it set to 7 when I heard clicking but didn't see it move. Now at 5, I hear clicking and slight movement.

[image]

I could try making changes around this area and compile this myself and see what I get. I've never done it before so it will take me a while to set up the tools.

Seelenkind, 2022-03-06:
Hello, thanks for the hint.  I activated Soft PWM and set the value to 5 or 7 so that the PWM-controlled fans run more quietly and with less whirring.  It can really be because not every fan works properly with it.  Tonight after work when I'm home I'll disable softpwm and re-upload the firmware.

Seelenkind, 2022-03-06:
I have now deactivated Soft PWM and hope that the fans are working properly. As already mentioned, Soft PWM didn't cause any errors for me and worked wonderfully.
You will find two links to the firmware
[2.0.9.3](https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0) & [Bugfix 2.0](https://github.com/Seelenkind/Marlin/tree/SKR-mini-E3-V3.0-G0B1/Marlin)

edwardryan321, 2022-03-08:
I tried the 2.0.9.3. The board fan works for me now. I heated the hot end to 51C and that fan came on too.
I didn't try a print yet. Maybe I'll have time tonight.
Thanks for your help with this.

yfmichea, 2022-03-13:
I installed the new firmware, but I am still not able to home the Z axis. I am using the SKR mini V3 (3.5" YFT), with a BLtouch v3.1. The BL Touch was working fine with the original Creality motherboard.  When I turn on the printer, I can see a couple of error messages (Echo: Fade Height Off, and Error:Failed to enable Bed Leveling).
I have checked and followed manual instructions to connect the BL touch to a single port, and checked cables' continuity, and everything seems OK. 
The only thing I noticed when trying to home is that the probe is deployed, but the Z axis doesn't seem to move at all. I can still command it to move, but it would not do it when homing. 
Should I just reinstall the original z probe and run both at the same time? What else can I do to troubleshoot this?

Seelenkind, 2022-03-16:
@yfmichea "The BL Touch was working fine with the original Creality motherboard". Exactly. Some BL Touch work very well with the original board and cause problems with other boards. For whatever reason. Sometimes it's just the cable assignment, sometimes it's voltage. But most work without problems. Unfortunately I don't have a solution for this.

Breadleyg, 2022-03-18:
> I installed the new firmware, but I am still not able to home the Z axis. I am using the SKR mini V3 (3.5" YFT), with a BLtouch v3.1. The BL Touch was working fine with the original Creality [...]

I'm having the exact same issue and it's driving me nuts. Let me know if you figure it out, i'm trying all different kinds of connection and setting combinations without success

Favmir, 2022-03-19:
> I installed the new firmware, but I am still not able to home the Z axis. I am using the SKR mini V3 (3.5" YFT), with a BLtouch v3.1. The BL Touch was working fine with the original Creality [...]

You're probably having this problem: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/602#issue-1092131562
Either you re-install the z endstop or make a new firmware instead of using the official firmware. The official firmware is set up wrong.

barrack-n-obama, 2022-03-20:
@Seelenkind your work has been very helpful - i am having the same issues with the Ender 5 pro, is there any way to easily mod what you have packaged to work for my system? Sorry for dumb question.

Seelenkind, 2022-03-20:
> @Seelenkind your work has been very helpful - i am having the same issues with the Ender 5 pro, is there any way to easily mod what you have packaged to work for my system? Sorry for dumb question.

@barrack-n-obama 
try this and give Feedback pls 

https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Ender%205pro

barrack-n-obama, 2022-03-21:
This is the first time i was able to successfully use the "home" feature. It did the full home function whereas i could never do that before. It does start up with an error that says "error: failed to enable bed leveling" but i'm wondering if that's because I simply have not been able to do that before because the bl touch didn't work. I will proceed to see if it is operating but - thank you for this much.

Seelenkind, 2022-03-21:
> This is the first time i was able to successfully use the "home" feature. It did the full home function whereas i could never do that before. It does start up with an error that says "error: failed [...]

this is not an error.  After flashing the firmware, there is still no mesh bed. This message disappears after the first auto bed leveling.

barrack-n-obama, 2022-03-21:
Ok new errors i'm getting i wonder if they are related to the offset of 5 when it used to be ten in the stock marlin before this board. When i run ABL using bl touch it never gets to the point the pointer retracts. 

Then i get "unknown command: g12 p0 s12 p0" it is still retrying the bed leveling

all other functionality appears to work at this time. bed and hotend heated fine.

Seelenkind, 2022-03-21:
> Ok new errors i'm getting i wonder if they are related to the offset of 5 when it used to be ten in the stock marlin before this board. When i run ABL using bl touch it never gets to the point the [...]

Did you fill in your X & Y offset? So the distance between nozzle tip & BL touch tip?
For example my offset X -47 & Y -7
Later you have to determine the Z offset. But you know all that, don't you?
If not, please read carefully: How To ABL

Seelenkind, 2022-03-21:
> Ok new errors i'm getting i wonder if they are related to the offset of 5 when it used to be ten in the stock marlin before this board. When i run ABL using bl touch it never gets to the point the [...]

unknown command: g12 p0 s12 p0

the board Mini E3 V3.0 is buggy.
That won't be the only error you'll get.
It is possible that with a longer press you will suddenly see the following on the display: M1 Stop, wait for user input.
Then I have to remove this command from Marlin.
But it is a known error that does not appear with everyone and not always.
So if yes, please let us know.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/618

Kareareawbg, 2022-03-21:
Follow teaching techs youtube guide, also his webpage for troubleshooting abl including checking what pins are what. I had a similar problem. Turned out to be i didn't have bi linear mesh as the only bed leveling option in config and the wiring wasn't correct for my board. It now self tests correctly.

barrack-n-obama, 2022-03-21:
Ok - I was unfamiliar with the errors and severity, but both responses here make me more confident these are not fatal errors. I will refine my settings and troubleshoot. I didn't know teaching tech had already come out with videos on this board. Thank you for the fw!

Kareareawbg, 2022-03-22:
Just to be clear, its not specifically for the e3v3. But a basis of trouble shooting for any board. He has a couple there an octopus and an skr board. So far for me, many of my problems have been my own darn fault. Sloppy wiring or not understanding enough about what I was changing in the firmware and how it interacted with other parts of the code.

barrack-n-obama, 2022-03-22:
@Seelenkind could you repost the "Marlin 2.0.9.3 Ender 5pro ZIP File"? The link died

closed 2022-04-01 by Seelenkind