bigtreetech/BIGTREETECH-SKR-mini-E3 #624: [FR] (feature request title)

cdowell1978, 2022-02-13:
**Is your feature request related to a problem? Please describe.**
Not sure if it is a problem or a feature, but it's asking to have the Z-offset Wizard when installing the firmware-ender3-bltouch.bin file. Whenever I try and enable it in the firmware, I get errors when trying to compile it in VS Code. it says that MESH_BED_LEVELING and BABYSTEP_ZPROBE_OFFSET are not valid combinations. I am just trying to set the CR Touch close to the bed so it can start leveling from there. 
I am fairly new to this side of 3D printing, and it's my first real attempt at trying to get changes to work in the configuration_adv.h file. 

**Describe the solution you'd like**
Could this option be enabled as a default when downloading the .bin file?

**Describe alternatives you've considered**
Just not using the CR Touch, but would really like to.

**Additional context**
The problems I am having are that after the CR Touch does the 25 point leveling, it either starts printing pretty high off the bed (not even close to the bed/plate), or runs into the plate and scrapes a line into it. After trying to follow the steps on this website, https://3dprintscape.com/z-offset-wizard-complete-guide/ I get the errors I mentioned above.

radek8, 2022-02-13:
Why did you enable MESH_BED_LEVELING?
When you use a probe?
If you have a probe, select another system.

cdowell1978, 2022-02-13:
That’s what was already enabled in the firmware when I downloaded it. I ended up having a friend help me and he got it working for me. Thanks for the input though!

closed 2022-02-13 by cdowell1978