bigtreetech/BIGTREETECH-SKR-mini-E3 #625: [BUG] (flashing bl touch firmware cause screen to go blank)

LichMeMate, 2022-02-19:
### Description

I just go a cr touch for my ender 3 and followed the steps to a tea but now my screen is blank. I tried reflashing the firmware and even using another board. is there something in the firmware that I cant use on the stock screen? I know it successfully flashes cause the file name changes to "FIRMWARE.CUR". Do I need to unplug the sensor first and then flash the firmware?

radek8, 2022-02-19:
Check the BLTouch connection.

LichMeMate, 2022-02-20:
> Check the BLTouch connection.

everything is connected. the cr touch lights up.

radek8, 2022-02-20:
Can you tell me more about your problem?
What version of the SKR board do you have, what is your display?
Can you send configuration, connection photos, or a video with the problem?

LichMeMate, 2022-02-20:
> Can you tell me more about your problem?
> What version of the SKR board do you have, what is your display?
> Can you send configuration, connection photos, or a video with the problem?

Im using the skr mini e3 v2 and im using a stock ender 3 display. when i turn on the power the display is just blue. i cant do anything

radek8, 2022-02-20:
If you disconnect BlTouch, Will the display work?

macmac34, 2022-02-20:
I have the same issue and tried unhooking the BlTouch to no avail

LichMeMate, 2022-02-20:
no the display is still wont work.

radek8, 2022-02-20:
Then you have a poorly configured display.
Submit configuration files.

LichMeMate, 2022-02-20:
i do i get the config files

LichMeMate, 2022-02-20:
> i do i get the config files

how*

radek8, 2022-02-20:
Configuration files from which you compiled FW
confiruartion.h
configuration_adv.h

macmac34, 2022-02-20:
I'll see what I find - thanks!

radek8, 2022-02-20:
Not what I find ....
What FW did you use?
Didn't you compile it yourself?

radek8, 2022-02-20:
Build and compile your own FW.
Then you can solve any problems and add parameters and new functions.
You can build on sample configurations here:

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples

mjulrich, 2022-02-24:
I have the same problem. How do you compile FW? I have no clue.

Madman24, 2022-02-25:
I have the same problem too, file name change "FIRMWARE.CUR". It think everything connected correctly. 
[image]

radek8, 2022-02-25:
> I have the same problem. How do you compile FW? I have no clue. I have an Ender 3, org display, BTT SKR Mini V2.

Visual Studio Code + PlatformIO