bigtreetech/BIGTREETECH-SKR-mini-E3 #626: [BUG] SKR v3 mini step file doesn't include the mounting hole pattern

emaayan, 2022-02-21:
### Description

the step file in the repository does not include the mounting hole pattern on the card itself 

### Additional Information

the mounting hole pattern is a must for anyone wanting to design an enclosure for this card, although there's a 2d schema which does include it's a lot easier if you everything in one place .

anatolyrc, 2022-03-19:
Should be the same as the v2 board, since it's designed to go into an Ender 3. However, I can't even read the v3 step file with Fusion 360, and can't convert it with AnyConv.

benzo1981, 2023-02-12:
Any updates on this? 
just bought one of these for a diy project and would love to have the mounting holes added to the step file.

fedorlspp26, 2023-02-27:
still not on the file, pretty sad.

jmaster03, 2023-03-04:
Theres isn't updates on this yet?

mfaccin, 2023-03-08:
last year I modified myself the mounting holes to design a case, measuring and modifying the step file.
tomorrow I will have access to the pc and see if I still have it available there. don't hesitate to ping me if I don't update it, please.

mfaccin, 2023-03-20:
I tried based on the design files and measurements using my board (SKR Mini E3.0 -- first version, I believe).
hope it helps and appreciates any feedback on the model. 😃 

[SKR mini E3 with heatsink step file](https://github.com/mfaccin/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/3D/BTT%20E3%20SKR%20MINI%20v3.0%20with%20heatsink.step)