bigtreetech/BIGTREETECH-SKR-mini-E3 #628: M48 Command Not Working on latest version of SKR Mini E3 v2.0 BLTouch for Z Homing Firmware

fadetonoir, 2022-03-04:
I have an Ender 3 with an SKR Mini E3 v2.0 board and a 3DTouch (BLTouch clone) ABL probe installed.

I have been using the latest version of firmware from this official repository for the last few weeks, with the BLTouch for Z Homing option as I have removed my original Z stop.

So that's the firmware-bltouch-for-z-homing.bin file found here - https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

In general the firmware seems to work fine, however I've been having problems getting a good first layer even with auto bed levelling. After much troubleshooting I started to doubt the accuracy of my probe - it is a cheap BLTouch clone, after all.

So I decided to do a probe accuracy test using M48. I typed M48 into terminal, hit send and got the message 'Unknown command: "M48"'

It seems strange to me that a firmware compiled for a BLTouch would not have the M48 command for probe accuracy test enabled. I tried switching to firmware-bltouch.bin and the M48 was accepted on that firmware (only it told me I needed to home Z first, which I can't do without reinstalling my Z stop).

So in short, there seems to be an oversight where M48 was not enabled in the Z Homing version of the latest firmware. Hopefully this can be fixed in the next update.

Seelenkind, 2022-03-06:
Try This compiled by me 
https://github.com/Seelenkind/Ender-3-V1-SKR-Mini-E3-V2.0-Marlin-2.0.9.3-BL-Touch

fadetonoir, 2022-03-07:
> Try This compiled by me https://github.com/Seelenkind/Ender-3-V1-SKR-Mini-E3-V2.0-Marlin-2.0.9.3-BL-Touch
> 
> Plug 2 wire at Z endstop [image]

Thanks for replying - I had my probe installed in the dedicated Z probe connector but it was easy enough for me to move it to Z endstop and use your firmware, so thanks to you I was able to run a probe repeatability test for the first time (and I think I got good results - deviations of 0.002500 - 0.004750? I believe BLTouch is rated to 0.005000 so this clone seems to have comparable performance). I guess my bed levelling issue must lie elsewhere...

I'm currently learning how to compile Marlin from scratch so I can fully customise it so this shouldn't be a problem for me in future but I will leave this issue open because BigTreeTech really should have M48 enabled in the Z probe versions of their own firmware.

Seelenkind, 2022-04-28:
> I got the same issue like the one above. @Seelenkind Do you still have the firmware? So I can try?

@VandenOstende i have deleted this firmware, not using it. Pls ask @fadetonoir for download from him.

sunnyxiong2020, 2023-01-30:
Just for the people who might see this issue
You have to enable the "#define Z_MIN_PROBE_REPEATABILITY_TEST" in the configuration.h file in order to use the repeatability test.