bigtreetech/BIGTREETECH-SKR-mini-E3 #630: No Z Change During Prints; Z-Stepper Functions Normally Otherwise

gallowayk, 2022-03-14:
Hi all, I recently installed the SKR Mini E3 V2.0 in my Ender 5.

Everything works as expected except the z-value changing during a print. The printer homes just fine, BLTouch works as it should, but during a print, it'll lay the first layer then never increment the Z-Axis value, thereby continuing to print on the same plane. 

I have scrubbed through the firmware files with little to no luck in finding something that would cause this issue.
I have tried to set Z-stepper current to 860mA, but didn't help.

Any advice, insight, or experience is appreciated.

Thanks!

insuham, 2022-04-17:
I have the same symptoms, but I don't know the cause.
All functions operate normally, but only the Z-axis value changes when printing, and the Z-axis does not actually move.
I would appreciate it if someone could tell me the solution.

ihagens, 2022-05-16:
Got a printer of a friend over to fix. seems to have the same problem but it doesn't always accure. did you find a solution or is this a SKR Hardware problem?

m3kaman, 2023-01-25:
I have the exact same issue. I've tried 2 SKR Mini E3 V3's. Issue persists. 

The thing is, more than a year ago I built a machine using the Ender 5 platform using the SKR Mini E3 V3 and a TFT35. It works beautifully. So much so I wanted to build another.

2 months ago I bought a stock platform and the exact same board, screen upgrade, and the exact same firmware. I got this issue you all are refering to. I troubleshooted various settings and possible hardware issues. Perhaps the z motors were faulty? perhaps the z-steps were set differently. No. No solution. I returned the board and tried another SKR Mini. same issue. I put the new SKR mini into the OLD working machine. The problem followed it. 

Could there be just a bad batch of SKR Mini's out there?? I can't figure out this problem

PaulScar, 2023-04-17:
This has plagued me for three days, I tried all sorts of things to cure it. Today I was watching the first layer of a print when I noticed the Z position reported on the TFT was 1.45 despite the fact that I was printing a 2.8 layer. Stopped the printer and looked through the settings until I found that Z home offset was set at 1.45 for some reason. I changed it to 00.0 and tried again. Glory be the print is running perfectly as I speak.