bigtreetech/BIGTREETECH-SKR-mini-E3 #631: Can anyone make 400mm x 400mm print size firmware for bigtretech skr mini e3 v3.0 32 bit?

plevia, 2022-03-21:
Hello,

"Bigtretech skr mini e3 v3.0 32 bit”

BLTOUCH is already installed.
I am using the "Ender Extender 400XL (for Ender 3 PRO)" kit.
The print area is therefore oversized 400mm x 400mm.
Can I ask you to create the firmware for me?

Best regards.

Seelenkind, 2022-03-25:
x=400 y=400 and z=? 
Only change bedsize and add BL Touch ?

plevia, 2022-03-25:
Hello Seelenkind,

First of all I want to thank you for your reply.
Thank you very much.

The sizes you inquired about are as follows
400mm (X) 400mm (Y) 500mm** (Z) 

For more information, please visit the following URL

https://enderextender.com/products/ender-extender-400xl-for-ender-3-pro

We plan to change the bed size, add BL Touch and install a filament sensor. If you have any other suggestions, I would be happy to add them.

Is it possible to create firmware?

Best regards.

Seelenkind, 2022-03-29:
@plevia precompiled firmware.bin and config files in ZIP
Download [Marlin 2.0.9.3](https://github.com/MarlinFirmware/Marlin/archive/2.0.x.zip) and copy files in ZIP.file do edit / change settings
[image]
[Ender 3 XtendeR BL Touch & Filament Sensor.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8374397/Ender.3.XtendeR.BL.Touch.Filament.Sensor.zip)

plevia, 2022-03-30:
Hello Seelenkind,

I appreciate your kind response during your busy time.
Since you created the firmware, I would like to purchase the main unit and try it immediately.
I'm really thankful to you.

Best regards.

CptKaveman80, 2022-04-16:
Hello, 
I need help. I keep getting false postive on my smart filament sensor after the pruge line. ive tried eveything i can thing of.
M119 im getting triggered.

89coltgt, 2022-06-13:
> 

just want to start by saying thank you for building this , 

But i have been trying to edit this for my own needs but however i keep getting this error every time i try to compile it  can someone please let me know what i am doing wrong 

Executing task: C:\Users\Nitro\.platformio\penv\Scripts\platformio.exe run <

Processing mega2560 (board: megaatmega2560; platform: atmelavr@~3.4; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\configuration.h:23:2: error: #error "Don't build with import-2.1.x configurations!"
 #error "Don't build with import-2.1.x configurations!"
  ^~~~~
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\configuration.h:24:2: error: #error "Use the 'bugfix...' or 'release...' configurations matching your Marlin version."
 #error "Use the 'bugfix...' or 'release...' configurations matching your Marlin version."
  ^~~~~
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:55:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\src\core\drivers.h:83:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\src\core\drivers.h:123:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) || AXIS_DRIVER_TYPE(A,TMC2160) \
                               ^~~~~~~~~~~~~~~~
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\configuration_adv.h:2876:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(U)
       ^~~~~~~~~~~
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\src\core\drivers.h:83:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\src\core\drivers.h:123:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) || AXIS_DRIVER_TYPE(A,TMC2160) \
                               ^~~~~~~~~~~~~~~~
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\configuration_adv.h:2886:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(V)
       ^~~~~~~~~~~
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\src\core\drivers.h:83:51: error: missing binary operator before token "("
 #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
                                                   ^
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\src\core\drivers.h:123:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
 #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) || AXIS_DRIVER_TYPE(A,TMC2160) \
                               ^~~~~~~~~~~~~~~~
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\configuration_adv.h:2896:7: note: in expansion of macro 'AXIS_IS_TMC'
   #if AXIS_IS_TMC(W)
       ^~~~~~~~~~~
**In file included from c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\src\pins\pins.h:541:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\nitro\desktop\marlin-2.0.x (1)\marlin-2.0.x\marlin\src\pins\stm32g0/pins_BTT_SKR_MINI_E3_V3_0.h:173:4: error: #error "DWIN_CREALITY_LCD requires a custom cable, see diagram above this line. Comment out this line to continue."
   #error "DWIN_CREALITY_LCD requires a custom cable, see diagram above this line. Comment out this line to continue."**
    ^~~~~
CalledProcessError: Command '"C:\Users\Nitro\.platformio\packages\toolchain-atmelavr\bin\avr-g++.exe" -D__MARLIN_FIRMWARE__ -DNDEBUG -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\Nitro\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 182:
    env.SConscript(item, exports="env")
  File "C:\Users\Nitro\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Nitro\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Nitro\Desktop\Marlin-2.0.x (1)\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 247:
    apply_features_config()
[... cut ...]
Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:00.854
======================================================================================== 1 failed, 0 succeeded in 00:00:00.854 ========================================================================================
The terminal process "C:\Users\Nitro\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

89coltgt, 2022-06-13:
just want to start by saying thank you for building this ,

But i have been trying to edit this for my own needs but however i keep getting this error every time i try to compile it can someone please let me know what i am doing wrong

> Executing task: C:\Users\Nitro\.platformio\penv\Scripts\platformio.exe run <

Error: Unknown environment names 'STM32G0B1RE_btt'. Valid names are 'include_tree'
The terminal process "C:\Users\Nitro\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

louisjtime, 2022-07-22:
Hello :)
Were you able to install your printer correctly? your filament detector too?
I will soon be in exactly this configuration: Ender 3 pro, SKR mini E3 v3.0, BTT TFT35-E3 V3.0, 400x400x500, BTT SFS V1.0 Smart Filament Sensor Detection Stuck Blocking Filament Module no End stop and BL Touch. Can anyone help me with this homemade setup?
Thank you Best regards

89coltgt, 2022-07-23:
No I haven’t

DanielKulikowski, 2022-09-21:
Hello i need help wyth software build for Ender 3 Max wyth SKR mini e3 v3 board and CR-touch
I managed to modifie Mister Seelenkind marlin build to 300x300 bed and compiled it sucesfuly, the problem now is the auto home is still the same as on the 400x400 bed and the printer thinks the bed is moved 100 mm to the right i dont know wher to change it in marlin software. Its my first Marlin modification. Can someone help or provide a bin file for it?