bigtreetech/BIGTREETECH-SKR-mini-E3 #632: SKR mini E3 V3.0 Third Thermistor Sensor

markxyx, 2022-03-31:
Hi, is there any pin that I can use it as third thermistor sensor for heated chamber?
I tried Z stop and PS-on. No luck.

markxyx, 2022-03-31:
> 

So damn hard...seem like using pi's pin would be much easier.

radek8, 2022-03-31:
Measuring temperature is not the same as connecting a limit switch.
Temperature measurement is performed by measuring the voltage on a resistive divider that contains a thermistor.
You also can't connect a fan to a random pin. The fan must switch Mosfet.

radek8, 2022-03-31:
Caution the processor has 3.3V logic and some pins are sensitive to higher voltages and can be destroyed by 5V voltages!

Jaybeam, 2022-10-18:
I don't know iff you stil need this but I use pin PA5 for my chamber thermistor:

- On a E3 V2 this is the pin labeled SCK on the SPI1 header.
- On a E3 V2 this is the pin labeled CLK on the SPI1 header.

ironhalik, 2023-01-12:
Oh man @Jaybeam - coming back back with an answer half a year later is going above and beyond. Thank you.
I'm going to try it.

Dataro81, 2023-01-12:
[image]
Is this where the definition is? The diagram is confusing.

Dataro81, 2023-01-12:
nm, just found out that sck and clk are different acronyms for the same thing. 
SCK stands for **S**pi **C**loc**K**. 
CLK stands for spi **CL**oc**K**.

Dataro81, 2023-01-12:
> I don't know if you stil need this but I use pin PA5 for my chamber thermistor:
> 
> * On a E3 V2 this is the pin labeled SCK on the SPI1 header.
> * On a E3 V2 this is the pin labeled CLK on the SPI1 header.

For the second line, i think you meant to write E3 V3, right?

NonaSuomy, 2023-01-16:
For the E3 V3.0 you can connect a 3rd thermistor to MOSI on the ISP 8 Pin Header I tested this with the Umbilical board for a Voron 0.1 it does cause issues with the SD Card but it's not too much of problem with Klipper you can just disable it then update firmware etc.
[image]

For Klipper
```yaml
#####################################################################
#   Chamber Temperature Sensor
#####################################################################

[temperature_sensor Chamber_Temp]
sensor_type: Generic 3950
sensor_pin: PA7 # PA7 is MOSI on the SPI header. Has an onboard 10K pullup to 3v3. Connect the other leg of the thermistor to ground
pullup_resistor: 10000
```

[image]

ironhalik, 2023-01-16:
And here's for E3 v2.0 :)

```
[temperature_sensor chamber]
sensor_type: Generic 3950
sensor_pin: PA5
min_temp: 0
max_temp: 90
gcode_id: chamber_temp
pullup_resistor: 10200
```

[image]

Dataro81, 2023-01-19:
Nice! How are you wiring it to the pi though? Not seeing the wire.

NonaSuomy, 2023-01-19:
Goes under the front cross member at the top between the plex panel and the front cross member, down the inside front cross member to the right horizontal cross member, down the right vert member, behind the back panel into the electronics chamber to the RPi.

AxMod3DPrint, 2023-04-22:
Sorry to bump an old post but in a pinch you can also use pin PA1 on the Z probe header on the v2 board, I was hunting around as I have a screen that uses the CLK pin and found Klipper accepted that pin, won't be dead on accurate as it bounces around a little bit like I said, in a pinch. You also need a stupid pull up to get it reading roughly correct. Seems to work though.

Klipper config:
```
[temperature_sensor chamber]
sensor_type: Generic 3950
sensor_pin: PA1
pullup_resistor: 7250000
```

zeljkomarinkovic, 2023-11-24:
Sorry to bump an old post!

@Dataro81 @NonaSuomy 
Did you guys get stable reading from thermistor? 

I have btt skr 3 mini v3 and I want to add third thermistor (ATC Semitec 104GT-2/104NT-4-R025H42G) to it. In my case i got huge fluctuations in temperature (±10 °C) like in picture.

[image]

This is from my printer.cfg

[temperature_sensor Chamber_Temp]
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA7 # PA7 is MOSI on the SPI header. Has an onboard 10K pullup to 3v3. Connect the other leg of the thermistor to ground
min_temp: 0
max_temp: 90
gcode_id: chamber_temp
pullup_resistor: 10000

Have anybody else managed to get it working?

z00mantwo, 2024-07-05:
I tried MOSI. I got big swings too. CLK on SPI header was better. Very steady reading. Only +-0.2C. Pin PA5 maybe? Check docs.

NonaSuomy, 2024-07-05:
[image]

NonaSuomy, 2024-07-05:
Do you have an SDCard in the board?

z00mantwo, 2024-07-06:
> Do you have an SDCard in the board if so remove it?

No SDCard.

NonaSuomy, 2024-07-07:
Maybe your sensor? Are you using the same one as on the Umbilical board?

xMikyx, 2024-08-22:
> For the E3 V3.0 you can connect a 3rd thermistor to MOSI on the SPI1 8 Pin Header. I tested this with the Umbilical board for a Voron 0.1. It does cause issues with the SD Card but it's not too much of problem with Klipper you can just disable it then update firmware etc.
> 
> [image]
> 
> [image]
> 
> For Klipper
> 
> ```yaml
> #####################################################################
> #   Chamber Temperature Sensor
> #####################################################################
> 
> [temperature_sensor Chamber_Temp]
> sensor_type: Generic 3950
> sensor_pin: PA7 # PA7 is MOSI on the SPI header. Has an onboard 10K pullup to 3v3. Connect the other leg of the thermistor to ground
> pullup_resistor: 10000
> ```
> 
> [image]

Can I then use this thermistor to control some free output on the mini e3v3 (PS_ON, PWE_DCT,...), as the thermistor for the hotend and the bed do. Maybe just on/off control so the thermistor would open and close the relay? For example, when I set the temperature in the slicer to 60 degrees, when it reaches it, it turns off the relay. Of course, if it was set to 0 degrees, the heating wouldn't even start. Is this possible?

lxlmongooselxl, 2025-08-12:
Sorry for resurrecting an old thread, but wanted to add an additional option. I had some thermistors laying around that came with my Gigabyte motherboard. Part number [12RH1-001002-01R]. Put a multimeter to them and watched what would happen when holding them in my palm. Discovered they were 10K ohm NTC thermistors. Also did some multimeter testing on the various SP1 pins to see if I could find additional pins with on-board pull-up resistors. Found out PA5 was another candidate. Hooked one thermistor to PA7 and GND and the other to PA5 and GND. Used the following blocks in printer.cfg:

```
[thermistor GIGABYTE_10K_3988]
temperature1: 25
resistance1: 10000
beta: 3988

[temperature_sensor stepper_x_temp]
sensor_type: GIGABYTE_10K_3988
sensor_pin: PA5
pullup_resistor: 10000
min_temp: 0
max_temp: 100

[temperature_sensor stepper_y_temp]
sensor_type: GIGABYTE_10K_3988
sensor_pin: PA7
pullup_resistor: 10000
min_temp: 0
max_temp: 100
```

Now have two additional function thermistors I can use at my leisure. Currently have them strapped to my X and Y stepper motors for testing.

[image]