bigtreetech/BIGTREETECH-SKR-mini-E3 #633: SKR MINI E3 v3.0 on Ender 3 Pro

Kilroy-101, 2022-03-31:
### Description
First of all - i am completely new to the 3d printing. I bought this board since with the standart board stepper motors is very, very loud.
So board expected to print "out of the box", yet - doesn't do anything so far at all. It's just resets itself every time as soon as hotend and the bed getting ready when i'm trying to print literally anything. Tired of looking at "Marlin Bugfix 2.0x" screen. And i found no solution for that so far.
"Please send help" lol

### Steps to reproduce
1. Just turn on the printer and try to print anything.

**Expected behavior**
Print out of the box.

**Actual behavior**
It's just resets itself as soon as hotend and bed got warmed up.

### Additional Information
I didn't modify it anyhow, since i am completely new and i basically have no idea how to do it. All i did - installed it accoring to the instruction and tried to print.

Kilroy-101, 2022-04-01:
> 

Thanks for the advice, really, but i wish i'd knew what is required to be modified in the configuration files lol
You see - besides the new board,  whole printer is in the stock configuration.

radek8, 2022-04-01:
Download the latest Marlin. You can use the main release or with bug fixes.
https://marlinfw.org/meta/download/

Then download the archives with the recommended configuration.
https://github.com/MarlinFirmware/Configurations/archive/release-2.0.9.3.zip
In the folder - Configurations-release-2.0.9.3\config\examples\Creality - find your printer and the used board.

Compare configuration files and transfer changes to Marlin files

Install
Visual Studoi Code and compile Marlin

Read the instructions on Youtube

Kilroy-101, 2022-04-02:
> Download the latest Marlin. You can use the main release or with bug fixes. https://marlinfw.org/meta/download/
>
> Then download the archives with the recommended configuration. [...]

Oh well. First of all thanks for trying to help.
I suppose there is something's off with the board itself. Because i've done everything correctly according to the guides. Changed everything in Conf' files, compiled without any sorts of errors, yet - as soon as i put it in the 3d printer... nothing happens. It's just boots up as usual with that "Marlin Bugfix", when i'm trying to check if any changes were applied - nope, the framware version is the same, and the date of update is the date when it was built. i'll assume that i've bought an defective board. great.
Anyways, thanks for trying to help.

Case closed.

radek8, 2022-04-02:
If you use BlTouch look here.

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/604

Kilroy-101, 2022-04-02:
> 

Thanks, but nah dude. 
Completely standart set, nothing's been replaced, except for, obviously - the board itself. Like i mentioned before - it doesn't ract on a new framware anyhow. Supposed to update through SDcard, but it doesn't do anything. And i have no Arduino and patiance at this point.

radek8, 2022-04-02:
The board will probably be fine.
The problem will be between the keyboard and the chair. :-)
You have to study the creation of your own FW.

Kilroy-101, 2022-04-02:
> The board will probably be fine. The problem will be between the keyboard and the chair. :-) You have to study the creation of your own FW.

Not very nice.
Yet, thanks anyways.

radek8, 2022-04-02:
When you print larger things you will find that leveling soda is necessary. Sooner or later you will buy it anyway.
So add BlTouch to the printer and use some FW that shares users in the links that I sent you.

radek8, 2022-04-02:
I wouldn't want a printer without a probe anymore.

radek8, 2022-04-02:
https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1

Seelenkind, 2022-04-02:
Download Firmware from [here](https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/Ender%203). 
rename to firmware.bin 
copy at SD Card
plug in at Board
turn printer on

Kilroy-101, 2022-04-03:
> Download Firmware xxx.bin from [here](https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/Ender%203). rename to firmware.bin copy at SD Card plug in at Board turn printer on

Thanks. But i already did that countless times by this point. Just won't update. It doesn't react anyhow on firmware on an SDcard at all. I've tried few different stock versions of the firmware.

radek8, 2022-04-03:
Seelenkind has already helped a lot of users. His FW will work.
@Kilroy-101  What is your problem?
Will the BIN file be renamed to CUR on the SD card after the FW update?

Kilroy-101, 2022-04-03:
> Seelenkind has already helped a lot of users. His FW will work. @Kilroy-101 What is your problem? Will the BIN file be renamed to CUR on the SD card after the FW update?

No. It doesn't do anything.

radek8, 2022-04-03:
Sometimes the card reader does not understand the SD card.
Use an 8GB or smaller SD card! Do not use cards larger than 32GB. Format it to FAT32
Copy the FW to the  SD card and restart the printer.
the FW file should be renamed from BIN to CUR.
If that doesn't work, try another SD card. Try more cards.

Kilroy-101, 2022-04-03:
> Sometimes the card reader does not understand the SD card. Use an 8GB or smaller SD card! Never do not use cards larger than 32GB. Format it to FAT32 Copy the FW to the SD card and restart the printer. the FW file should be renamed from BIN to CUR. If that doesn't work, try another SD card. Try more cards <=8GB.

Already done. Besides - i don't have 32gb cards at all, Got a few different SD's less than 8gb - two 1gb and also couple of 8gb's (it's been pointed out about a million times in all the guides that it should be less or at least 8gb sd) i've reformatted every single one of them to FAT32. Doesnt work at all. It just doesn't react anyhow on firmware.

>  Seelenkind has already helped a lot of users. His FW will work. @Kilroy-101 What is your problem? Will the BIN file be renamed to CUR on the SD card after the FW update?
> 
I'm not saying that someone's firmware is bad or anything. All i'm saying: no matters which firmware i'm trying to toss in - it doesn't work. Like i've mentioned before - i've tried most of the versions that has been recommended by other people.
Pretty sure - the problem is with the board itself.

3DRic80, 2022-05-17:
I am in the same boat as Kilroy-101, I have previously upgraded the stock board to the E3 V2.0(which has since died) and configured the firmware for that board so I have some knowledge of what I am doing, however this new V3.0 board will not update the firmware. Tried various SD cards, tried BTT's firmware for BLT and BLT with Z homing. Not worked, tried hitting the reset button still doesn't work. Even found some post stating that renaming firmware.bin in cap' letters works. Nope, De nada, zilcho. Sending back and going to reorder a V2.0 board.

TrevorMatthews, 2022-05-24:
I had a E3 V2.0 board for a few years, a few days ago a vase I was printing would simply stop being printed totally randomly with the print nozzle still touching the layer it was printing, heaters all off. I tried half a dozen times.
Reflashed the firmware, checked all bearings, belts, fans, cpu temp etc etc. Resliced the STL.

In the end I just ordered a new V3.0 board, I fitted it a few days ago and used the BTT pre compiled .bin file for the BLTouch with Z homing. Whilst it all worked the probing started off the front of the bed and needed lots a messing with off sets to get it all right.

I used a 32GB Sandisk card.
The file is renamed firmware.bin
When the machine is turned on you will see the usual No Printer Attached message, but it will flash that message for a little longer than normal as the .bin file is flashed to the board.
Once that has been done you will notice the file has been renamed FIRMWARE can't remember if it was .bin or .cur

I have today printed the 4 and a half hour vase which I hope proved the V2.0 board was indeed faulty.

BTT seem to have got some folders incorrectly named for the Ender 3!

regards
Trevor