bigtreetech/BIGTREETECH-SKR-mini-E3 #635: maple_win_usb_driver

rickbagby, 2022-04-03:
Can someone for the love of God, please tell me how to download this summabichen file? I can see it, I can open it, but it will not let me download it. I need it to install the drivers on my comp for the board.

radek8, 2022-04-03:
[image]

rickbagby, 2022-04-03:
With the exception of a digital signature it seems to have worked. I just have to disable the digital signature requirement on my comp which is no biggie. A real big thank you Radek.

Dentuescher413, 2022-07-01:
Tried downloading the files and the maple.inf file cannot be executed as it is a text file. When I right click the file, it opens in Notepad++. Do you know where I can get the Maple driver file?

radek8, 2022-07-02:
Do not click the inf file. Update your drivers with them.

rickbagby, 2022-07-02:
I couldnt get it to work. I returned it and get one from TH3D.

Dentuescher413, 2022-07-02:
Thanks for the info. What I did to fix my problem is started using  Win10 instead of Win7 then disconnected my printer. Then I went to device manager and brought up the comm port and uninstalled it. Then reboot my pc and plugged in my printer. Win10 auto installed the Maple driver.  I had read that Win10 had the Maple driver and Win7 did not.