bigtreetech/BIGTREETECH-SKR-mini-E3 #636: Next issue: config.h modified

rickbagby, 2022-04-03:
I believe that I have all the config.h files modified for my Ender 3 Pro. I will be running the BTT SKR Mini E3/V2 board. I have dual axis Z steppers, and a much bigger than stock build volume. I have installed the ender extender 400 XL. That takes me to 400mmX400mmX500MM. What I am trying to figure out is how to compose those into a .bin file. I have not used VS Code before but it seems easy enough. I just dont know how to get all that information into a .BIN file. a little help would be appreciated.

rickbagby, 2022-04-03:
So, I have made some progress. I am in the process of "building" but then I get this error message. This is what it tells me... "Building in release mode
Linking .pio\build\STM32F103RC_btt_USB\firmware.elf
c:/users/bagby's/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file ldscript.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RC_btt_USB\firmware.elf] Error 1
=================================== [FAILED] Took 120.69 seconds=========================="
Any ideas?

radek8, 2022-04-05:
Try using the platform
STM32F103RC_btt_maple

rickbagby, 2022-04-08:
Something still is not quite right. I am getting further but now it fails for "arm-none-eabi-g++: error: CreateProcess: No such file or directory *** [.pio\build\STM32F103RC_btt_maple\src\src\gcode\control\M211.cpp.o] Error 1"

This is getting frustrating..

rickbagby, 2022-04-08:
So knowing that my reason for failure changes, I am going to monkey with the board settings and see if I can get anything to work there.

rickbagby, 2022-04-08:
Changing only the board environment, I seem to do better with STM32F103RC_btt_512K_USB. However it still fails. I get error message "c:/users/bagby's/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file ldscript.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RC_btt_512K_USB\firmware.elf] Error 1
=========================================== [FAILED] Took 349.70 seconds" 

It is taking longer to fail. I wonder if I do not have something in Visual studio turned on that I probably should.