bigtreetech/BIGTREETECH-SKR-mini-E3 #637: [BUG] SD Init Fail  using  SD Card Extender (SKR mini E3 V3)

r4ph74, 2022-04-06:
I have Ender 3 and just replace the board with SKR min E3 V3, works flawlessly at the moment.  Two days ago I just bought SD card Extender (1 Meter Cable length), and I have a plan to use it with Toshiba Flashair. I'm getting tired of juggling microSD from PC to the printer and vice versa every time I print.  The issue is when plugging it into the MicroSD slot on the board, The display shows " SD init Fail".  It works fine on Standard Ender 3 pro, and Flashforge AD3 (using a MicroSD Card reader). 

Any help will be much appreciated.

### Steps to reproduce

1. Turn off the machine
2. Plug the SD card Extender into the MicroSD Slot on the board
3. Turn on the machine 

I tried also 
1. Turn on the machine 
2. Plug the SD card Extender into the MicroSD Slot on the board

**Expected behavior**
LCD Shows : "Media inserted"

**Actual behavior**
LCD shows:  "SD Init Fail"

### Additional Information
[image]

regards, 

Rafael

Seelenkind, 2022-04-11:
The use of an SD card extension is more of a matter of luck. I had no problems with the SKR Mini E3 V2. The board V3 is rather bitchy.
When I insert my SD card extension and switch on the printer for the first time with the SD card inserted, I can plug the SD card in and out as often as I want. But if the printer is switched on without an SD card inserted and I reinsert the card later, I have to briefly unplug the SD card extension and reinsert it.

thisiskeithb, 2022-04-12:
Be sure to enable `NO_SD_DETECT` when using an SD extender:

```cpp
  // Enable if SD detect is rendered useless (e.g., by using an SD extender)
  //#define NO_SD_DETECT
```

https://github.com/MarlinFirmware/Marlin/blob/29d3f6a024672a960f7e033e360178f4e80ce98b/Marlin/Configuration_adv.h#L1651-L1652

thisiskeithb, 2022-04-12:
> is it from the firmware? Do I have to reflash the firmware?

Yes and yes. You’ll have to uncomment that line to enable the feature and reflash your firmware.

thisiskeithb, 2022-04-12:
> if this is activated, there is definitely no SD card detection.

That’s exactly the point of the `NO_SD_DETECT` flag. It disables the SD detect pin and you have to mount/unmount the SD card manually through the menu or via goods since the SD detect pin is always engaged when using an extender.

Seelenkind, 2022-04-12:
@thisiskeithb 
"and you have to mount/unmount the SD card manually through the menu or via gcode" doesn't work.
If the SD card was not inserted when switching on, a subsequent update (mount) is not possible. Whether you commented on the definition in Marlin or not doesn't change anything.
The same SD card expansion function on the E3 V2.0 board without any problems. But on the E3 V3.0 an automatic SD card detection is not possible.
Unfortunately, that depends on the board.
Everything already tried. I even got a second expansion. The same thing.

r4ph74, 2022-04-12:
I see.. Ok thanks.

r4ph74, 2022-04-12:
What about using Btt tft screen? Is the SD slot, recognised automatically? Or we still have to edit the firmware?

thisiskeithb, 2022-04-12:
> doesn't work.

Something else is wrong then.

The E3 V3 has an SD detect pin (PC3) and that will be disabled when enabling `NO_SD_DETECT`. Then it will change the behavior of how you use SD cards based on my comments above.

magborresen, 2022-06-25:
Check that your SD Card is formatted to FAT32, this is the only format the board will read, which is also stated here https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Klipper. That was atleast the cause of my problems with this error. Note that if your card is over 32 GB it will never come with FAT32 formatting and you have to use 3rd party software to change the formatting since Windows won't format anything higher than 32 GB to FAT32.

r4ph74, 2022-06-26:
I have tried with other sdcard. I think it gots nothing to do with sd card. My sd card only 8GB and fat 32.
So weird case.some people say just turn off SD check configuration when start up.

KingBain, 2022-07-30:
Confirming with the **two** skr mini e3 v3's that I own. 

These boards appear to be not compatible with the micro SD extenders.

KingBain, 2022-07-30:
Welp, quick update. 

removing/desoldering the resistor fixed my issue. 

Apparently the Raspverry Pi4 have a similar issue.
https://forums.raspberrypi.com/viewtopic.php?t=269457

jdmz1, 2022-08-04:
I tried resolder and tried enabling NO_SD_DETECT with no success. If I power on printer first then insert SD card adapter with card in it all is fine but if I try to boot up with SD card adapter it becomes super laggy.

jdmz1, 2022-08-06:
Update: I downgraded to 2.0.9.3 and it seems to work now.

Hantick, 2022-08-09:
The same on my side. On v2 it worked, on v3 sd card extender does not work, while sd card is read properly

jdmz1, 2022-08-09:
> The same on my side. On v2 it worked, on v3 sd card extender does not work, while sd card is read properly

Try desoldering as the guy above said and using 2.0.9.3 firmware. Worked for me finally.

Hantick, 2022-08-09:
I dont have such skills and braveness to desolder anything

jdmz1, 2022-08-09:
You can just go in there with a tweezer and break it. Does the same thing.

Hantick, 2022-08-10:
Weird thing, I can easily flash a new firmware from SD card extender with connected SD card, but media is not found.

Seelenkind, 2022-08-10:
Insert SD card in extender . Unplug extender on board , plug extender , try
this .

2022, 17:50:

johaleck, 2022-10-12:
> 

I know this is old, but I am having similar issues. Maybe the reason we can flash new firmware using the SD card extender is because the printer is restarted before the firmware will take. I have placed rhe microsd into my SD extender while the printer is on and it doesn't even show an SD card was inserted. Once I power it down then back up, it loads the firmware from the extender. Maybe as mentioned above, with the extender, it acts as though a card is already in amd therefore will not detect anything new.

amd989, 2022-11-08:
Same issue here. Tried with `NO_SD_DETECT` to no avail.

Coming from an E3 SKR DIP board that worked fine with Marlin `v2.0.X` and `v2.1.X`

Switched to E3 SKR MINI v3.0 and it doesn't work after the firmware (`v2.1.X`) boots up.

Firmware flashing works fine with the extender though.

Hantick, 2022-11-09:
Its defenetly a bug coming from Marlin or SKR

cryptopivot, 2022-12-23:
> I have Ender 3 and just replace the board with SKR min E3 V3, works flawlessly at the moment. Two days ago I just bought SD card Extender (1 Meter Cable length), and I have a plan to use it with [...]

So, here I am having the same exact issue, thought the weird thing is that I have mostly the blue 8gig mini sd cards which none will work in the adapter unless flashing with it.  However,  I have a 16 gig card made by sanddisk that works as desired in the SDmini adapter and then in the SDCard to SDmini adapter fixed on the mainboard cover.   Also note both blue 8 gig card does work in the tft35 slot.

cryptopivot, 2022-12-23:
So, here I am having the same exact issue, thought the weird thing is that I have mostly the blue 8gig mini sd cards which none will work in the adapter unless flashing with it. However, I have a 16 gig card made by sanddisk that works as desired in the SDmini adapter and then in the SDCard to SDmini adapter fixed on the mainboard cover. Also note both blue 8 gig card does work in the tft35 slot.

jlwoolf, 2023-01-09:
@cryptopivot Do you happen to have a link to the adapter you're using? I currently can't get mine to work with my Inland brand microSD cards, but if you do have a working combo of microSD and adapter, I'd love to know the brand of the adapter. That way I can get something that at least works, even if I have to use a specific card brand.

Here's a link to the adapter I currently have: https://amzn.to/3ilsmHl

cryptopivot, 2023-01-09:
It’s not the adapter. It had to be the card because I can put that card in anyway microSD to SD card adapter and it works inside of the extension. I will look when I get back to my house. I will not be there for another three days though.I have two types of SD cards two of them are the blue eight gig cards that come with the printers and then I have one 16 gig card, which works even after boot.  I can flash with the 8 gig cards but that’s it. I will give you more information as soon as I get it.~Mike Thompson"A Can Do Guy"Myrtle Beach • South Carolina  (910) 500-2382  (877) ***@***.***

cryptopivot, 2023-01-12:
Here is what I have,  a video showing the card working and others not.1 Videoshare.icloud.com

4yn, 2023-02-04:
> You can just go in there with a tweezer and break it. Does the same thing.
> […](#)

Thanks for this, it worked for my skr e3 3.0, you can pry apart the housing from the SD card side, then remove the resistor with some desoldering (or just snip it off with your filament snippers) and close the housing back. Now the cable works with my 32gb cards on both the printer and my pc as well.

[image]

4yn, 2023-02-04:
> Welp, quick update.
> 
> removing/desoldering the resistor fixed my issue.
> 
> Apparently the Raspverry Pi4 have a similar issue. https://forums.raspberrypi.com/viewtopic.php?t=269457

Thanks for this, it worked for my skr e3 3.0, you can pry apart the housing from the SD card side, then remove the resistor with some desoldering (or just snip it off with your filament snippers) and close the housing back. Now the cable works with my 32gb cards on both the printer and my pc as well.

[image]

3D-Dungeon, 2023-03-06:
> Weird thing, I can easily flash a new firmware from SD card extender with connected SD card, but media is not found.

My mini E3 V3 is doing the same thing, did you ever figure out the problem?

eoyilmaz, 2023-06-01:
@4yn thank you, breaking that resistor worked for me 👍

WeaselKing, 2023-08-17:
Wild. Thanks @4yn! I just removed that resistor like you said, and now it works just fine on the mini e3 v3.  You're a hero!

Headrush69, 2024-05-02:
Removing the resistor worked perfectly. Great find, thanks.

r4ph74, 2024-05-02:
Which resistor is it? 

Yahoo Mail: Search, organise, conquer

Headrush69, 2024-05-05:
The smaller one on the left.

See the picture from **4yn** a few post up.

r4ph74, 2024-05-05:
I see thx

Yahoo Mail: Search, organise, conquer

thk4711, 2024-09-12:
I am using an SKR mini E3 V3 as well and removing the resistor from the extention cable worked for me as well.

sckfred, 2024-12-05:
Thanks a lot, removing the resistor from the extention cable worked for me too.