bigtreetech/BIGTREETECH-SKR-mini-E3 #639: compiling error

rickbagby, 2022-04-17:
OK so I get to the "linking" part of the compiling, and I get this this...
"Linking .pio\build\STM32F103RC_btt_maple\firmware.elf
c:/users/bagby's/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file common.inc: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RC_btt_maple\firmware.elf] Error 1"

Any thoughts or ideas?

radek8, 2022-04-17:
How did you go about creating FW?
Did you download Marlin and edit the configuration files?

A lot of people make the mistake of downloading a new Marlin and copying configuration files from another version that they downloaded somewhere on the Internet.
Using configuration files of another version makes compilation errors.

rickbagby, 2022-04-18:
I got it straight from github. After talking with tech support, they said there was something wrong with the board. So they sent me a replacement. Well, that board powered everything but the bed heater. So it is going back as well. I am done with the cheap crap. Doesn't explain the firmware issue though. I am going to go with a TH3D board when my refund hits from this past mistake.