bigtreetech/BIGTREETECH-SKR-mini-E3 #641: SKR Mini e3 v2 SD init fail

Doormat1, 2022-04-26:
My printer has just developed the above fault,
Switch it on with the sd card in and I get the init fail warning.
SD card is working fine in my pc.

mayder, 2022-09-23:
I have the same problem, my SKR Mini E3 V3 don't start with SD Card and don't read firmware.bin file. My SKR Mini E3 V3 is new and I need update firmware to use BLTouth

mayder, 2022-09-23:
I have the same problem, my SKR Mini E3 V3 don't start with SD Card and don't read firmware.bin file. My SKR Mini E3 V3 is new and I need update firmware to use BLTouch

MGlaab, 2023-03-29:
I had similar issues trying to update my firmware. I solved my problem by using the program sd card formatter to first erase the sd card then I copied the firmware to it. I put it in my machine and it works again. See the link:

https://www.sdcard.org/downloads/formatter/