bigtreetech/BIGTREETECH-SKR-mini-E3 #643: SKR mini E3 V3.0 - CR Touch - Ender 3 V2 - Failure to Z-home.

niblem85, 2022-05-11:
Is there an actual, workable solution for people that have the mini E3 V3.0 with an Ender 3 V2 and CR Touch (specifically)?  After much research, I am gathering that many people are having similar issues with their setups and no good work around for a CR Touch is available.  [I have solved the Ender 3 V2 LCD screen issue by making a custom cable and changing the firmware, thankfully.]

I have attempted many fixes, including but not limited to: re-wiring the CR Touch on the Z-probe connection AND onto the Z-STOP connector (attached below are pics of my current setup), custom firmware changes and precompiled firmware.  So far, I have been unable to make the CR Touch work correctly.  I have specifically followed several other discussions, including: #23395, #619, #623, and #602.  I have also tried to use the most recent firmware from @Seelenkind. 

My current, non-working configurations: [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8670339/firmware.zip)

### Steps to reproduce

1. Turn on Ender 3 V2, the CR Touch appropriately self-tests by deploying and stowing the probe.
2. Navigate to Auto-home all axis - X and Y axis hit limit switches perfectly.
3. Z axis homing begins, Z raises up ~10mm, deploys probe and lowers to the bed.  Once probe touches bed, the LED on the CR Touch will flash red and the nozzle crashes into the bed.

**Expected behavior**
Functional CR Touch probing of the Z-axis without crashing the bed.

**Actual behavior**
During Z-homing, CR Touch deploys probe but when it triggers off the bed the CR Touch flashes red but continues to crash into the bed without stopping.

### Additional Information
[image]
[image]

Seelenkind, 2022-05-11:
@niblem85 pls try this Bugfix Version 
CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty. 

[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

niblem85, 2022-05-11:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

Thanks for the quick reply.
Unfortunately, since I have the Ender 3 V2 and the upgraded LCD screen that comes with it, when I flash that firmware you linked, my screen goes black.  I had to make some custom changes to my previous firmware in order to get it working.  (I was also was using the Jyers variety).

niblem85, 2022-05-15:
@Seelenkind Do you have the original files from the firmware you shared so I can edit them for the ender 3 v2 LCD screen?  Then I can try the firmware you listed.

Kel-Morian, 2022-05-16:
Hi,
I had the same problem and the Youtuber "Aurora Tech" gives the solution in one of her latest videos.
You have to disconnect the red and blue wires from your 5 pin to create a 2 pin that you will connect to the Z-Endstop.
You leave it the 5 pins with only three wires connected to the dedicated port.

I think BTT chose the wrong options when compiling the firmware.

niblem85, 2022-05-16:
> Hi, I had the same problem and the Youtuber "Aurora Tech" gives the solution in one of her latest videos. You have to disconnect the red and blue wires from your 5 pin to create a 2 pin that you [...]

Yes, I have tried this method also!  Unfortunately, it still does not successfully home the Z-axis.  My best guess is that my firmware needs some adjustment, but despite my best efforts to try different settings I cannot produce a good result whether I use all 5 Z-probe wires OR 3 on Z-probe and red/blue on z- endstop.  I was hoping someone here was in my same situation and had figured out a solution with the mini E3 v3, ender 3 v2, and CR touch.

Kel-Morian, 2022-05-16:
> Yes, I have tried this method also! Unfortunately, it still does not successfully home the Z-axis. My best guess is that my firmware needs some adjustment, but despite my best efforts to try [...]

With the firmware "[firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin)" ?

niblem85, 2022-05-16:
> With the firmware "[firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin)" ?

Yes but remember, I am using this on Ender 3 V2, so I cannot use that firmware since it is not compiled to utilize the Ender 3 V2's LCD screen.  I have to customize the firmware to allow me to use my screen.  So I am not sure if that firmware works.

niblem85, 2022-05-18:
(https://imgur.com/a/QdVhav9)
Here is a short clip of my printer displaying the behavior.  CR Touch is triggered during Z-homing, flashes red but continues to move towards the bed and crash.

Kel-Morian, 2022-05-19:
> Yes but remember, I am using this on Ender 3 V2, so I cannot use that firmware since it is not compiled to utilize the Ender 3 V2's LCD screen. I have to customize the firmware to allow me to use my [...]

Yes, sorry, I forgot that detail.
For the shots I have no solution to offer you except perhaps to switch to a TFT35-E3-V3.0 screen?

Sv1jge, 2022-05-21:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/643#issuecomment-1124039050

hello guys
This works for me .Thanx Seelenkind .But I want to compile my own settings and untill now nothings works with vscode.

iCarl, 2022-05-23:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

This works for me with ender 3 pro and skr mini v3.

elotedog, 2022-05-26:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

thank you I finally got the printer to print something although a lot of print quality issues and the bltouch fails like every other time. do you have the configuration files? would like to get a 3x3 grid instead of 4x4

karmavibes7, 2022-05-27:
> Yes, I have tried this method also! Unfortunately, it still does not successfully home the Z-axis. My best guess is that my firmware needs some adjustment, but despite my best efforts to try [...]

same problem too but with ender 5 pro. tried same solution like you but nothing.
good luck, may the force be with us ><

elotedog, 2022-05-27:
> thank you I finally got the printer to print something although a lot of print quality issues and the bltouch fails like every other time. do you have the configuration files? would like to get a [...]

just wanted to update for anyone in my situation, my setup is ender 3 with 3.1 bl touch. the firmware I quoted was the only one that got it printing. as far as my print quality issues, I lowered my print temp to 190 and now I seem to be able to get it going , still needs small adjustments but its up and running now. also just curious. got two boards from Amazon and neither came with a duck and seemed to be opened ;/ not sure if anyone else has this problem ?

PhilMaddox, 2022-05-29:
> same problem too but with ender 5 pro. tried same solution like you but nothing. good luck, may the force be with us ><

You do realise that depending on which way you wire the probe ( 5 pin or 3+2 pin) you need specific firmware for the chosen wiring method !!
I made a comparison chart to identify the differences in the configs https://ibb.co/9w5J34x

ngrhdsgn, 2022-06-25:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

hello, I'm using your firmware to run my ender 3 pro with CR touch and it's working very well. but when I turn on the part cooling fan at low RPM (below 100%) it emits whining noise, but when it is turned on at full speed (100%) the sound disappears. do you have a solution? I read on the prusa forum that it was caused by the settings in the firmware

jakedoesgithub, 2022-08-03:
Hi it’s not you it’s the recompiled firmware it’s incorrect here are the older btt version that work https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0

timgray, 2022-08-19:
thinking that this could be why Aurora tech had issues with the 5 pin, but all configs seem to use PA14 which is correct as far as i know

Schematics and pinout diagrams here show Bltouch sensor pin is PC14 not PA14.   use of PA14 will be wrong and will cause it to not function.

baturayd, 2022-08-23:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

Confirming this firmware works with SKR mini v3.0 + CR touch + Ender 3 Pro. Auto home and auto bed leveling works as expected. @Seelenkind Can you tell us in a nutshell whatthe difference between this firmware and [firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin) is?

Also, as per @jakedoesgithub, flashing v2.x firmware onto v3 board did not fix the issue for me. (hangs during bootloading.)

jakedoesgithub, 2022-08-24:
> Also, as per @jakedoesgithub, flashing v2.x firmware onto v3 board did not fix the issue for me. (hangs during bootloading.)

oops sorry didn't notice this was a v3 question. Look for the same version firmware but the v3 version then if you end up wanting to try it later. I'll edit my original post

dangvitin, 2022-08-27:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

@Seelenkind Can you please share the source of this firmware ?

Niyphelah, 2022-10-20:
> Confirming this firmware works with SKR mini v3.0 + CR touch + Ender 3 Pro. Auto home and auto bed leveling works as expected. @Seelenkind Can you tell us in a nutshell whatthe difference between [...]

Did you have to rewire anything? My CR Touch is unresponsive when I use the normal arrangement of 5 cables and leave the Z Stop ports empty.

Kyuuketsuk1, 2022-10-23:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

I am having the exact same issue. But I am using an **Ender 5** instead. I have installed a CrTouch and put into the z-probe field and have the same issue as OP. it reacts but the plate does not stop. I have also tried without the CrTouch with multiple different firmwares from other parties and the same issue. I have verified that the z-stop is working when plugging it in the x-stop-port I can trigger it. 

This video was with FW: [firmware-ender5-bltouch.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender5-bltouch.binl)

https://www.youtube.com/shorts/SCToQ-Bifuw

With FW: [firmware-ender5-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender5-bltouch-for-z-homing.bin) I am having the exact same issue as OP with an **Ender 5**

Appreciate any ideas what could be the cause for this issue

Sergeyss2022, 2022-11-06:
> @niblem85 пожалуйста, попробуйте эту исправленную версию CR Touch, подключенную к зонду (все 5 кабелей), отключите Z-Endstop и дайте ему разрядиться.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

Dear friend. Maybe there is a movement on the firmware with CRtouch for Ender3?

miketague, 2022-12-01:
Any update on this from BTT. I just installed my board and was rather upset to find that my CR Touch not working. It's a bit annoying, it's not exactly an rare upgrade. I'm considering just sending it back and getting the Creality upgrade board.

PhilMaddox, 2022-12-01:
> Any update on this from BTT. I just installed my board and was rather upset to find that my CR Touch not working. It's a bit annoying, it's not exactly an rare upgrade. I'm considering just sending it back and getting the Creality upgrade board.

Many people have this working !
Its important with the firmware choice to know whether you have all 5 wires in the "probe" port **OR** 3 in the probe port and the other 2 going to the Z min endstop.  Depending on that choice will decide which firmware you need.
Either way the original endstop (switch) needs to be disconnected 
The other issue seems to be that Creality have adopted their own wiring colour designation !  and this needs to be checked with regard to the output wires of the CR touch

Sergeyss2022, 2022-12-06:
Good afternoon. I replaced all the stuffing in my Ender3 with yours. You
can suggest how to set z indent and remove the table with Crtouch from
creality. A lot of settings in the Tft 35 are not clear to me. Thanks!

DaVinci0072, 2023-01-13:
Yep, had the same problem and this firmware fixed it

Ender 3, SKR mini E3 V3.0, CR Touch

Thanks!

smokeandlights, 2023-01-29:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

Thanks. This helped me on my ender 3 pro. Is there a version that fixes this for klipper?

stiliajohny, 2023-11-24:
> @niblem85 pls try this Bugfix Version CR Touch connected at Probe (all 5 cable) , unplug Z-Endstop and let it empty.
> 
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8672098/firmware.zip)

Do you have any firmware that suports normal LED ?

Seelenkind, 2023-11-26:
> Hello. Is there a config of this firmware or the source code of marlin ? Change the configuration of the settings. For my 3d printer?
Sure, search for [Marlin Github](https://www.google.com/search?q=marlin+github) at Google

stiliajohny, 2023-11-27:
@Seelenkind  Could you please help me with that ? 
I cant get it to work 
I connect the CRTouch to the Zprobe and its flashing the firmware bltouch-z-homing and is not doing Z home. it raises by 1cm and try to prob with no success 

thank you in advance

maconly34, 2024-01-07:
> @Seelenkind Could you please help me with that ? I cant get it to work I connect the CRTouch to the Zprobe and its flashing the firmware bltouch-z-homing and is not doing Z home. it raises by 1cm and try to prob with no success
> 
> thank you in advance

2 years after the first !!! 
Same problem with my Ender3 V2, I try to compil with many parameters change, but bad result !

maconly34, 2024-01-08:
Hello,
After many hours, the problem is solved !

**Solution :**
Only plug the old Z-sensor! Grrrrrrrrrrrr
I think it's possible that it's a float state when not plugged...

delong369, 2024-02-17:
Go to mriscoc firmware on github and go to his telegram and you can pay him to compile his professional firmware for it that will work. That's how I fix all my issues.

R-802, 2024-07-23:
First, if you're using klipper, reflash the board with the [firmware provided by BTT](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Klipper/firmware-USB.bin). 

The solution is here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/pull/718