bigtreetech/BIGTREETECH-SKR-mini-E3 #646: btt skr mini v2.0 generic screen problem

malicimen34, 2022-05-21:
hello, I'm trying to use my "DMG85480C050_03W" model screen with btt skr mini v2.0, but I couldn't find how to set it in the configuration.h file. Could you help me with this?
This is the screen I'm trying to use "https://www.aliexpress.com/item/33037017378.html".

radek8, 2022-05-22:
If you don't have an FW for 3D printing for this screen, then the display will be just a piece of iron.
You have to buy a display for a 3D printer. For example from BTT or MKS

radek8, 2022-05-22:
The display communicates with Marlin via the serial port.
If your display has the necessary FW, just define serial port 2 in Marlin
#define SERIAL_PORT 2
or
#define SERIAL_PORT_2 2