bigtreetech/BIGTREETECH-SKR-mini-E3 #647: SKR E3 V3 Error on build after add config files

GDRPrint, 2022-05-24:
I could not get VS  ( Platformio 6.0.0 failed to build Marlin fw ) to build firmware for BTT E3 V3, 
found thisisKeith solution to try dev version of Platformio
Using Marlin Autobuild ext
1. Using dev version of Platformio, 
2. VS code built fw using 2560 selection env ( Marlin autobuild ), no errors
3. Closed VS, added latest config files ( zip file from Github/BTT ) to Marlin folder, no other changes
4. Correct board in Auto build shows STM32GOB1RE_btt
5. Failed build
results
7. |-- Servo @ 1.1.2
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
Building in release mode
*** [C:\Users\GDR Trading\.platformio\packages\framework-arduinoststm32\variants\MARLIN_G0B1RE\PeripheralPins.c] C:\Marlin-2.0.9.3\Marlin-bugfix-2.0.x\.pio\build\STM32G0B1RE_btt\.sconsign39.dblite: No such file or directory
========================================================================================= [FAILED] Took 39.85 seconds ========================================================================================= 

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  FAILED    00:00:39.852
==================================================================================== 1 failed, 0 succeeded in 00:00:39.852 ==================================================================================== 
PS C:\Marlin-2.0.9.3\Marlin-bugfix-2.0.x> echo "done" >"C:\Users\GDRTRA~1\AppData\Local\Temp\ipc"
PS C:\Marlin-2.0.9.3\Marlin-bugfix-2.0.x> 

Very close with developer version of Platformio, had more errors with 6.0
Thanks for any thoughts, and many thanks for all the posts on this issue.

radek8, 2022-05-24:
> 3. Closed VS, added latest config files ( zip file from Github/BTT ) to Marlin folder, no other changes

You must copy the entire Marlin folder.
Not just configuration files.

GDRPrint, 2022-05-24:
Sorry, of course I used the whole file, just meant I pasted the new configs to replace the standard Marlin ones, to build the profile for that particulaer board.
If you open the unedited Marlin folder you will see therre are only 2 config files, 4 are needed to run the printer and screen.

GDRPrint, 2022-05-24:
Update for this post.
I tried to upload the file, it failed just before completion.
I selected the clean button which was activated after the upload attempt.
Clean successful
Rebuilt the file and was successful.
Software flashed and printing working well.
Please close if no further attention needed.
Hope this helps someone along the way.

radek8, 2022-05-25:
Only the author of the ticket can close the ticket