bigtreetech/BIGTREETECH-SKR-mini-E3 #650: [Display 12864 on SKR E3 Mini] How to adapt the BTT 12864 v1 at SKR E3 Mini V3?

Luismiguelfco, 2022-06-01:
### Hello Guys.
Recently i bought the SKR E3 mini V3 and a BTT 12864 display. But after bought that, i saw that's incompatible. :(

So, i started to search about the BTT display pinout and SKR E3 pinout. I found this pinout below:

[image]

           
So i connected the respective pins of BTT on SKR, and i tested all compatibles displays on Marlin, but it doesn't work :(

I think that to display the most important pins are: LCD_RST(D4), LCD_CS(EN), LCD_RS, 5V and GND. The display turn on the backlight led, but no show anything. 

I would like to know if somenone knows if there's a different protocol on BTT display, because SKR support some 12864 displays. Or what is the model of de uC that control de LCD.

I already tested the connection sugested to SKR E3 1.2 and 2.0 and it doesn't works too. :/

Someone to give some support to me? :D

radek8, 2022-06-01:
You can also make the Y cable yourself.

[image]

Luismiguelfco, 2022-06-03:
> You can also make the Y cable yourself.
> 
> [image]
> 

Hi @radek8, thank you for the answer.

I tryed your pinout, but it doesn't work. 😢 The led turn on, but nothing is showed in the screen.

I tested it using "CR10_STOCK_DISPLAY" on Marlin, and after that also i tryed "MKS_12864"

radek8, 2022-06-04:
What model of display do you have exactly?
Can you send photos from both sides?

Luismiguelfco, 2022-06-04:
> What model of display do you have exactly? Can you send photos from both sides?

https://pt.aliexpress.com/item/1005001729266611.html?spm=a2g0o.order_list.0.0.57e9caa4GigTjQ&gatewayAdapt=glo2bra

[image]

radek8, 2022-06-04:
If you have a display
BGTHREETECH-MINI-12864-V1.0,
then I gave you the wrong information.

You have to allow
#define FYSETC_MINI_12864_2_1
or
#define FYSETC_MINI_12864_2_1

In the pins file "pins_BTT_SKR_MINI: E3_common.h" comment line 241:
// # error "CAUTION! FYSETC_MINI_12864_2_1 / MKS_MINI_12864_V3 / BTT_MINI_12864_V1 requires wiring modifications. See 'pins_BTT_SKR_MINI_E3_common.h' for details. Comment out this line to continue."

You have to make your own cables:
Connectors

[image]

Connections

[image]

radek8, 2022-06-04:
Sorry,
your SKR mini E3 in variant V3
there is another connection.
I'll send another connection.

radek8, 2022-06-04:
You will allow
#define FYSETC_MINI_12864_2_1

You comment in line "pins_BTT_SKR_MINI_E3_V3.h" line 261

[image]

You make your own cable:

[image]

Luismiguelfco, 2022-06-04:
> You will allow #define FYSETC_MINI_12864_2_1
> 
> You comment in line "pins_BTT_SKR_MINI_E3_V3.h" line 261
> 
> [image]
> 
> You make your own cable:
> 
> [image]

I got this error on VS Code:

In file included from Marlin\src\lcd\dogm\marlinui_DOGM.cpp:42:
Marlin\src\lcd\dogm\marlinui_DOGM.cpp: In static member function 'static void MarlinUI::init_lcd()':
Marlin\src\lcd\dogm\marlinui_DOGM.h:225:23: error: 'DOGLCD_CS' was not declared in this scope; did you mean 'DOGLCD_      |                       ^~~~~~~~~
Marlin\src\lcd\dogm\marlinui_DOGM.cpp:267:14: note: in expansion of macro 'U8G_PARAM'
  267 |     u8g.init(U8G_PARAM);
      |              ^~~~~~~~~
Marlin\src\lcd\dogm\marlinui_DOGM.h:225:34: error: 'DOGLCD_A0' was not declared in this scope; did you mean 'DOGLCD'?
  225 |     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
      |                                  ^~~~~~~~~
Marlin\src\lcd\dogm\marlinui_DOGM.cpp:267:14: note: in expansion of macro 'U8G_PARAM'
  267 |     u8g.init(U8G_PARAM);
      |              ^~~~~~~~~
Compiling .pio\build\STM32G0B1RE_btt\src\src\lcd\menu\menu_bed_leveling.cpp.o
*** [.pio\build\STM32G0B1RE_btt\src\src\lcd\dogm\marlinui_DOGM.cpp.o] Error 1
=========================================== [FAILED] Took 35.54 seconds ===========================================

radek8, 2022-06-04:
You have a different version of Marin than I do.
That's why you commented on the wrong line.

I wrote that you have to comment on line 261.
So you commented on the wrong line.
Send me your pin set, I'll tell you which line to comment on.

radek8, 2022-06-04:
line 261 leave uncommented

You have to comment on the line I took

 //#error "CAUTION! FYSETC_MINI_12864_2_1 and clones require wiring modifications. See 'pins_BTT_SKR_MINI_E3_V3_0.h' for details. Comment out this line to continue."

Luismiguelfco, 2022-06-04:
> line 261 leave uncommented
> 
> You have to comment on the line I took
> 
> // #error "CAUTION! FYSETC_MINI_12864_2_1 and clones require wiring modifications. See 'pins_BTT_SKR_MINI_E3_V3_0.h' for details. Define NO_CONTROLLER_CUSTOM_WIRING_WARNING to suppress this warning"

Man, it's working!!!!! :DDDD

I tryed the latest version of marlin and did exactly what you said, and it's working!!! Really thank you!

I just need to finish the test regarding neo pixel, but the image is aleady on the screen.

Finishing i post the result's here and i will make a video to help another people.

Thank you a lot radek!!! I'm almost buying another display.

radek8, 2022-06-04:
Unfortunately, this is an atypical display for this board.
The buzzer won't work for you.
The SD card reader probably won't work either.
The pins have nowhere to connect.

Luismiguelfco, 2022-06-04:
> Unfortunately, this is an atypical display for this board. The buzzer won't work for you. The SD card reader on the display will probably not work either. The pins have nowhere to connect.

No problem if the SD or buzzer wont work.

About the Neopixel, i uncommented the lines regarding that, and RGB backlight is working.

But i facing another problem. When the pin DIN is connected to LCD, the display doesn't show anything, only the RGB backlight work; if i disconect the DIN pin, the backlight doesn't work, but the image is showed on the screen. Maybe could it be a contrast setup?

At monday i will buy a flat cable and new connectors to make a permanent cable, i'm using wire jumpers to make the conection today.

Monday i'll be back here with more news about the display :D

radek8, 2022-06-05:
The BTT manual introduces another parameter change.
In the file: Marlin \ src \ lcd \ dogm \ marlinui_DOGM.cpp
change parameter:
void MarlinUI::_set_contrast() { u8g.setContrast(contrast); }
change to:
void MarlinUI::_set_contrast() { u8g.setContrast(255); }

You can try ...

radek8, 2022-06-05:
Other parameters recommended by BTT for this display

[image]

radek8, 2022-06-05:
I found another configuration for your display.

If you need to change the RGB-color manually, then you need to enable the LED_CONTROL_MENU in configuration_adv.h. If you don't need it, you can skip this step.

Luismiguelfco, 2022-06-05:
> The BTT manual introduces another parameter change.
> In the file: Marlin\src\lcd\dogm\marlinui_DOGM.cpp
> change parameter:
> void MarlinUI::_set_contrast() { u8g.setContrast(contrast); }
> change to:
> void MarlinUI::_set_contrast() { u8g.setContrast(255); }
> 
> You can try ...

I changed this function and all is working!!!

Really thank you man! 

[image]

radek8, 2022-06-05:
Congratulations, you endured a successful end.

KingBain, 2022-07-15:
Hello,
I have an skr mini e3 v3 and a BTT mini12864 v1 . With that combination, which pins do I connect to make the reset button work ?

radek8, 2022-07-15:
[image]

KingBain, 2022-07-15:
Ive double checked my wiring and I have what you show above. Reset button isn't working. 

Wondering maybe, if KILL_PIN needs to be defined in marlin for reset to work ?

KingBain, 2022-07-25:
for others going through this, you need to connect both reset pins 
[image]

Wavegreen, 2022-09-24:
Hi, I have tried a few different connections already but can't get my screen to work. BTT support has also not been helpful. What would my connections have to be for a standard CR-20 display and an SKR MINI E3 V3.0? I have added some pictures. 
Thanks in advance!
[image]
[image]

radek8, 2022-09-25:
What type of display is used in the CR-20?

Wavegreen, 2022-09-25:
@radek8 thanks for your reply!
The display is the default display that comes with the CR-20. From forums it is _apparently_ the same as the one from a CR-10 but not sure. 
I tried googling around for shops to get a model number for the screen but haven't had any luck.
A start would be: [https://www.diyelectronics.co.za/store/displays/2301-replacement-lcd-display-for-cr-20-and-cr-20-pro-creality-original.html](url) but it does not give any details that could lead me to a pinout.

It looks extremely similar to the BTT 12864 display without reset button and sd card reader. I did try the connections of that screen with mine but haven't had any luck. I did double check connections but it is possible that I made an error of course.

[image]
[image]

radek8, 2022-09-26:
In the sample configuration for CR-20 it is defined:
#define MKS_MINI_12864
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/CR-20
So it will be his clone.
I assume you are compiling your own Marlin.
Send Marlin configuration files.
Send a photo of the cable to the display on the SKR side.

radek8, 2022-09-26:
According to the picture from your link and the pin layout in the Marlin, the wiring should be as shown.

[image]

Wavegreen, 2022-09-26:
Wow, thanks a lot, it worked without any further changes!
Is there any way to send you some money for a beer or coffee to say thank you? You saved me from further days of headaches!

radek8, 2022-09-26:
Congratulations :-)

radek8, 2022-10-03:
> I changed this function and all is working!!!
>
> Really thank you man!
>
> [image]

@Luismiguelfco 
I suggested the Marlin modification.
Thisiskeithb says no need to edit the file
Marlin\src\lcd\dogm\marlinui_DOGM.cpp
If no characters appear on the display, it is said that resetting the EEPROM is sufficient.
Can you please compile Marlin without this modification and try it?
Thank you

SimonVail, 2022-11-10:
Hi, do you know if it is possible to connect this screen to a SKR E3 DIP v 1.1 ?

radek8, 2022-11-11:
> Hi, do you know if it is possible to connect this screen to a SKR E3 DIP v 1.1 ?

Everything works if you want it and you know how.
Unfortunately, no one has yet defined the pins for this display in the pin file.
If you can do it, define the pins, draw a connection diagram and you can publish it as a PR in Marlin for other users as well.
But if you can't handle it, then you better not buy this HW combination.

radek8, 2022-11-11:
@SimonVail 
If you know at least a little bit what you are doing, see how this display is defined in the pin file for skr mini E3 V2 and V3
Then apply the settings to the SKR E3 Dip.
It won't be as complicated as it looks.
Unfortunately, the SKR dip E3 has only one EXP3 connector, so you won't have anywhere to map the SD card on the display.
But you probably have to take that into account if you bought such a board...

SimonVail, 2022-12-07:
thanks, made it work.

michaelsheridan23, 2023-01-04:
> for others going through this, you need to connect both reset pins [image]

I have the display working and the reset button works too. But my question is after initial boot up I HAVE to hit the reset button to get the display to show anything. I have all of the firmware changes made correctly, I believe, and I have the wiring correct but for some reason the display will not come up unless I hit the reset button. Is this just an issue with this display or am I doing something wrong? 

I have attached my configuration.h, configuration_adv.h, marlinui_DOGM.cpp, and pins_BTT_SKR_MINI_E3_V3_0.h files for reference. TYIA.

[Marlin Files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/10345191/Marlin.Files.zip)

brandonackerman, 2023-01-09:
I have been trying to follow along for about 3 weeks now trying to get my E3v2 back up and running with the BTT SKR Mini E3_V3 and BTT Mini 12864 V1 display but have thus far only have the backlight working. I believe I have everything wired up correctly and have manually edited my config files to match those found here, but still come up with a blank display. @michaelsheridan23 I just tried your files with the latest Marlin 2.1.2, but when compiling I am getting:

In file included from Marlin\src\lcd\dogm\marlinui_DOGM.cpp:46:
Marlin\src\lcd\dogm\dogm_Bootscreen.h:40:12: fatal error: ../../../_Bootscreen.h: No such file or directory
   40 |   #include "../../../_Bootscreen.h"
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32G0B1RE_btt\src\src\lcd\dogm\marlinui_DOGM.cpp.o] Error 1
In file included from Marlin\src\lcd\dogm\status_screen_DOGM.cpp:32:
Marlin\src\lcd\dogm\dogm_Statusscreen.h:49:12: fatal error: ../../../_Statusscreen.h: No such file or directory
   49 |   #include "../../../_Statusscreen.h"
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32G0B1RE_btt\src\src\lcd\dogm\status_screen_DOGM.cpp.o] Error 1

michaelsheridan23, 2023-01-09:
@brandonackerman my build files are based on bugfix-2.1.x not the released 2.1.2. That might help get rid of the errors. I found pull my files from old builds into the latest builds messed with the background files too much and messed up the sanity checks.

radek8, 2023-01-09:
@brandonackerman
You have to add missing files to marlin before compilation, or disable the status and boot screen

brandonackerman, 2023-01-09:
@michaelsheridan23 @radek8 Thanks guys, I got it working. Ended up having to disable the custom boot screen and status screen to get it to compile properly. I now have a working display! Now on to hooking everything else up to the board and seeing if the rest of it works!

michaelsheridan23, 2023-01-09:
@radek8 any idea on my issue above (from 5 days ago)? Is that normal behavior for this setup? Or do I just have something wrong? Thanks!

radek8, 2023-01-09:
I do not own this display or this SKR board.
So I can't try to simulate the mentioned problem.
If you shut down the printer and the display remains blank, what happens when you click on the encoder or turn it? Will the display be rendered?

radek8, 2023-01-09:
Can you try initializing the EEPROM?
Use the M502 command to set the default values and use the M500 command to save them. Reset the printer.
Then try if your problem persists.

radek8, 2023-01-09:
If it doesn't help, try adding the following line (at position 347) to the file "pins_BTT_SKR_MINI_E3_V3_0.h"
and recompile Marlin
#define LCD_CONTRAST 255

[image]

lipe1234645, 2023-03-08:
@radek8, Hello, I am unable to connect my BTT 12864 v3 to my skr mini e3 v2, do you know how to connect?

radek8, 2023-03-09:
BTT 12864 v3?
For the BTT 12864 V1 connection, see here.

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/686

radek8, 2023-03-09:
@lipe1234645 
Use the latest Marlin with bug fixes
Older versions may contain bugs that cause the display to not work.

lipe1234645, 2023-03-09:
@radek8  deu esse erro

In file included from c:\marlin-bugfix-2.1.x\marlin\src\hal\hal.h:30:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
c:\marlin-bugfix-2.1.x\marlin\src\hal\stm32f1\hal.h:172:1: error: expected ';' before 'typedef'
 typedef int8_t pin_t;
 ^~~~~~~
In file included from c:\marlin-bugfix-2.1.x\marlin\src\hal\hal.h:30:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
c:\marlin-bugfix-2.1.x\marlin\src\hal\stm32f1\hal.h:97:3: error: static assertion failed: SERIAL_PORT must be from 1 to 5. You can also use -1 if 
the board supports Native USB.
   static_assert(false, "SERIAL_PORT must be from 1 to " STRINGIFY(NUM_UARTS) ". You can also use -1 if the board supports Native USB.")
   ^~~~~~~~~~~~~
In file included from c:\marlin-bugfix-2.1.x\marlin\src\hal\hal.h:30:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
c:\marlin-bugfix-2.1.x\marlin\src\hal\stm32f1\hal.h:172:1: error: expected ';' before 'typedef'
 typedef int8_t pin_t;
 ^~~~~~~
In file included from c:\marlin-bugfix-2.1.x\marlin\src\hal\hal.h:30:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
c:\marlin-bugfix-2.1.x\marlin\src\hal\stm32f1\hal.h:97:3: error: static assertion failed: SERIAL_PORT must be from 1 to 5. You can also use -1 if 
the board supports Native USB.
   static_assert(false, "SERIAL_PORT must be from 1 to " STRINGIFY(NUM_UARTS) ". You can also use -1 if the board supports Native USB.")
   ^~~~~~~~~~~~~
*** [.pio\build\STM32F103RC_btt_maple\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_maple\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
========================================================== [FAILED] Took 59.58 seconds ==========================================================

Environment            Status    Duration
---------------------  --------  ------------
STM32F103RC_btt_maple  FAILED    00:00:59.579
===================================================== 1 failed, 0 succeeded in 00:00:59.579 =====================================================

 *  The terminal process "C:\Users\felli\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

radek8, 2023-03-09:
@lipe1234645 
Marlin configuration error.
(badly defined serial ports)

You have to download Marlin.
Edit the configuration files according to your configuration and compile the FW.
If you want to update new versions often, use Desktop GitHub, which will synchronize changes and new versions with your settings.

A common mistake is to download a new Marlin and copy the configuration file from the old version of Marlin into it.
Configuration files of different versions of Marlin are not compatible with each other.

lipe1234645, 2023-03-09:
Eu tentei compilar novamente e deu certo, só o display que não aparece nada

radek8, 2023-03-09:
Use English.
Use the thread related to the same HW as you have.
This thread deals with connecting the display to the SKR mini E3 V3
You write that you have an SKR mini E3 V2. The V2 version has a thread here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/686

PangiLukas, 2023-04-21:
Hi, Recently i bought the BTT SKR E3 mini V3 and i have home MKS LCD12864B V1.0.  in latest marlin 2.1.2 i define it as lcd and use mks mini12864 pinout for wiring but it doesnt work. Only 2pins what im trusting is 5V and GND. I try finding pinout individual for this lcd but nothing found. Can someone please help me?[image]

[image]

Thanks so much

radek8, 2023-04-21:
@PangiLukas 
Bad combination of display and control board
The SKR has an EXP3 connector
The display has an EXP1 and EXP2 connector

You would have to use a Y cable.
The card reader on the display will not work in this connection.

PangiLukas, 2023-04-21:
I have alright mks tft32_L v4.0 can it work?

radek8, 2023-04-21:
The MKS LCD12864B V1.0 display will work in Marli mode.
The SD card reader will only work on the SKR

radek8, 2023-04-21:
The TFT display will also work but it will be connected via the serial port.
Unfortunately the display is MSK and the Marlin board is from BTT.
Each uses a different connector. So you have to make your own cable.

radek8, 2023-04-21:
configuratin
https://www.banggood.com/cs/MKS-LCD12864B-Intelligent-LCD-Display-Smart-Display-3D-Printer-Part-p-1641871.html?cur_warehouse=CN

PangiLukas, 2023-04-21:
Yeah, all this setting i still do. But at lcd is not show, it only light up. If there is no way to configure with working sd.it is not usable for me. I try to find display from btt. Thanks for all @radek8

radek8, 2023-04-22:
You misunderstood me.
The SKR E3 mimi V3 board has only 1 EXP3 connector.
You can use it to connect the display and use Merlin mode.
You can use the SD card on the SKR, but not on the display, the SD card communicates via the EXP2 connector.
You would have to buy a board that has two connectors (EXP1 and EXP2).

The second option is to use the SD card on the TFT touch screen.
The TFT display has its own processor and communicates via a serial port.
So you can use your MKS TFT32_L v4.0. Your SD card will work there.
 However, MKS TFT32_L v4.0 does not have Marlin mode, but that doesn't matter.

PangiLukas, 2023-04-23:
So for tft 32 its enough to connect via tft serial port 5v/gnd/tx/rx and in firmware set number of serial port and baudrate?

radek8, 2023-04-23:
Yes

[5 more replies]