bigtreetech/BIGTREETECH-SKR-mini-E3 #653: [BUG] Blue screen with skr mini E3 V3

Webdevpistolet, 2022-06-15:
Hello, 
I installed the BTT skr mini E3 V3 on the Ender 3 pro, I made two prints then I changed the fans and installed the micro-swiss hotend. I turned the printer back on and since then it stays on a blank blue screen. I unplugged everything on the card and turned it back on but it does the same thing. I downloaded the ender 3 firmware on the github, copied it to my sd card firmware.bin and nothing happens, when I read the sd card again, the file is still in bin format, I'm starting to despair because I can't find any solution to this problem.
Can you help me to solve this problem?

radek8, 2022-06-15:
If you connect the printer to a counter, does it recognize it?. 
Communicates?

Webdevpistolet, 2022-06-15:
I connected it to octopi but no connection is detected.

radek8, 2022-06-15:
Try to load the FW from the SD disk again. Try more SD drives.
The SD drive should be up to 8G and FAT32 format.

If it does not connect, the display does not show anything and even the new FW from the SD disk will not be loaded, so it looks like a damaged SKR.

Forsyth-Creations, 2023-02-27:
I'm having this same issue. Here are some specifics:

I am using Marlin branch bugfix-2.1.x on a CR-10 with the BTT SKR v3, and have had great success flashing firmware through the SD card, as well as the binary transfer option given through Marlin/Octoprint. Recently I encountered this issue too. I have tried the base image for the BTT board, attempting to flash it though the SD card. In all cases, I see the firmware.bin change to a FIRMWARE.cur, which indicated that the board took the firmware. However, the board will not connect via USB to Octoprint, and the screen remains blue. Is there any solution in the works for this, as I have seen many such issues on Reddit and in other communities.

radek8, 2023-02-27:
Send the configuration files.

Forsyth-Creations, 2023-02-27:
[Configurations.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/10836206/Configurations.zip)

Here are the configurations I used!