bigtreetech/BIGTREETECH-SKR-mini-E3 #658: [BUG] (short description)

srettew, 2022-06-21:
I just installed an E3 V3.0 on my Ender 3 Pro. This is regarding the 3 fan assignments and getting them to work correctly. I confirmed the following by checking function:

FAN0 (PC6) = Part cooling fan
FAN1 (PC7) = Hotend cooling fan
FAN2 (PB15) = Control board cooling fan

FAN0 turns on correctly by using fan speed control on TFT35
FAN1 turns on correctly when hotend temp is goes above 50 degrees and turns off when lower than 50 degrees
FAN2 is the problem. This is supposed to turn on when a driver is activated. I get nothing when I manually move any of the axis using the movement control on TFT35.

Where in the config_h file can I find PB15 and what is the correct code fix to make it work?

radek8, 2022-06-22:
Can you send your configuration files?

srettew, 2022-06-22:
It’s the config that is available on GitHub for Ender 3 Pro and SKR Mini E3 V3.0.

StrawberryD24, 2022-07-24:
Having the same issue on an ender 3. The part fan won't turn on at all

radek8, 2022-07-24:
You have to allow
#define USE_CONTROLLER_FAN
and define the pin
#define CONTROLLER_FAN_PIN PB15

StrawberryD24, 2022-08-14:
@radek8 which file(s) need to be modified for the pin definitions? Can you share an example config?

radek8, 2022-08-14:
file Configuration_adv.h

[image]

StrawberryD24, 2022-08-18:
@radek8 
Thanks for your  for the insight but I have already done that. I have no issue with the board controller fan and hot end fan. my Issue is with the part cooling fan supposedly PC6 in the firmware.

I compared the pinout on the Box and user manual for the V3 board vs those published on their site and the firmware. Turns out on the pin diagrams shown on both the box and the user manual on their github,  the pins are listed as :
HB - PC6
E0 - PC9
Fan2 - PB15
Fan0 - PC7
Fan1 - PC0

Whereas the pins in stm32g0 defines them as
#define HEATER_0_PIN                        PC8   // "HE"
#define HEATER_BED_PIN                      PC9   // "HB"
#define FAN_PIN                             PC6   // "FAN0"
#define FAN1_PIN                            PC7   // "FAN1"
#define FAN2_PIN                            PB15  // "FAN2"
[image]
[image]
[image]

StrawberryD24, 2022-08-18:
Here is the Box
I think it's a bigger issue than just firmware 
[image]
[image]
[image]
e

radek8, 2022-08-18:
@StrawberryD24 
Yes, you are right, the description of the pins on the cse does not match the definition of the pins in the pin file.

@bigtreetech, can you comment on the fan pin definition issue and fix it?¨

radek8, 2022-08-19:
if you think that there is a poorly defined set of pins in Marlin, create a ticket in the Marlin repository with a request for a correction, or propose a correction yourself.

https://github.com/MarlinFirmware/Marlin/issues

StrawberryD24, 2022-08-20:
@radek8  thanks a mil for your help. It will be my first request for correction on GitHub hope I do it right lol

WebSpectrum, 2023-03-08:
iS THIS XORRCT?:

WebSpectrum, 2023-03-08:
Is this correct:

WebSpectrum, 2023-03-08:
is this correct?:

FAN0 (PC6) = Part cooling fan
FAN1 (PC7) = Hotend cooling fan
FAN2 (PB15) = Control board cooling fan

or this:

FAN0 (PC6) = Part cooling fan
FAN1 (PC7) = Control board cooling fan
FAN2 (PB15) = Hotend cooling fan

I see a mix of information

radek8, 2023-03-09:
@WebSpectrum 
The part cooling fan is defined in the pin file.
#define FAN_PIN PC6 // "FAN0"

You define the hot end cooling fans and control boards in the configuration file configuration_adv.h
#define E0_AUTO_FAN_PIN
#define CONTROLLER_FAN_PIN

WebSpectrum, 2023-03-09:
Hello,

 

In the manual it shows FAN0 as PC0  ?

 

 

Regards

 

Paul Maher

Web Spectrum -  <http://www.webspectrum.co.uk> www.webspectrum.co.uk 

22+ Years Professional Web Design/Development & High-End SEO

Adobe Community Professional

Phone: 07837 903056

radek8, 2023-03-10:
Send a link to the mentioned manual.

WebSpectrum, 2023-03-10:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/Hardware/BTT%20SKR%20MINI%20E3%20V3.0%20user%20manual.pdf

 

 

Regards

 

Paul Maher

Web Spectrum -  <http://www.webspectrum.co.uk> www.webspectrum.co.uk 

22+ Years Professional Web Design/Development & High-End SEO

Adobe Community Professional

Phone: 07837 903056

radek8, 2023-03-10:
> @radek8 děkuji mil za vaši pomoc. Bude to moje první žádost o opravu na GitHubu, doufám, že to udělám správně lol

@StrawberryD24 
Can you send a detail photo of your SKR board?
It looks like your board is not from BTT, but an imitation from another manufacturer. :-)

radek8, 2023-03-10:
It seems that BTT has chaos in the documentation

SKR mini E3 V3.0

[image]

SKR mini E3 V3.0
[image]

SKR mini E3 V3.0.1
[image]

radek8, 2023-03-10:
This layout will probably be faulty.

[image]

Wully616, 2023-04-24:
Just purchased today from amazon and seeing the same issue, the board documentation makes no sense.
Can someone confirm the fan pin layout?

mianos, 2023-04-30:
I used 
```
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN PC8 // PB15
```
Based on the PCB 3.0.1 screen print above and I seem to have a 'whole system' fan as fan 3 in the menu that can be be controlled with PCM 0-100%.

The board is marked 3.0. The box is marked 3.0 but the D/C is 2023 1.6, right from the fab by the look of it.
(under the main text it says 'new' in Chinese.

mogfink, 2023-05-01:
So I'm having a related but dissimilar issue if anyone has any insight... I'm using the io pins to control rgbw LEDs through MOSFETs, everything works except that when the board fan is on the blue LEDs light, and when I switch on the blue LEDs it tries to run the board fan for a moment until both fan and blue LEDs go off, I'm using pin pd4 for the blue LEDs and the board fan is on fan1

Gavisann, 2023-07-06:
I have been facing the same issue with fan headers and this is what worked for me using the precompiled firmware that shipped with my Mini E3 V3 and Ender 3 Pro.

FAN0 - Part cooling fan - Confirmed by sending `M106 S255` via Octoprint's Terminal which runs the fan at 100%.
FAN1 - Hotend fan - Confirmed by heating hotend to 200c, fan kicks on at 50c
FAN2 - Controller fan - Confirmed by using Octoprint to move the X/Y motors and the fan runs for 1-2minutes.

tim-speight, 2023-12-28:
HOW DO YOU KNOW IF YOU HAVE A 3.0 OR A 3.0.1? My boards case say's New and manufactured 2022.12.31

radek8, 2023-12-28:
According to the wiring diagram, it looks like the difference is in the CPU used
V3.0 - STM32G0B1RCT6
V3.0.1 - STM32F401RCT6

tim-speight, 2023-12-31:
OK It definitely is a v3.0.  But I am wondering if it is a fake.I used the Marlin firmware distinctly for a v3.0 and the pinouts are running the wrong elements.  I wonder if there were a dodgy batch of boards BIQU manufactured in 2022.
The Box wiring with random PC0 & PC6 assignments is worrying.  How did U get on Strawberry?