bigtreetech/BIGTREETECH-SKR-mini-E3 #659: BLTouch doesn't work on SKR E3 v3

Luismiguelfco, 2022-06-24:
Hello guys. I bought a SKR E3 mini v3, and i've been facing problems with BLtouch.

Sometimes it works, but almost all times that i try to do the bed leveling(G28 amd G29), it stops during the leveling after a ghost trigger and report: STOPED on the screen.

With Arduino and Ramps the same sensor is working fine.

I tested all functions regarding bltouch on marlin to try to correct this failure.

I tested in 2 inputs, the servo conector and the Z min plug

Anyone can³ help me?

J4nRyb1n, 2022-06-25:
Just try connect BLTouch to z-stop pins - I had a same problem before...

Luismiguelfco, 2022-06-25:
> Just try connect BLTouch to z-stop pins - I had a same problem before...

Hello J4nRyb1n,

I already tested this. I connected the BLTouch on the Z min end stop input, and also in the probe input, and it still not working well.

My problem is the "ghost" trigger. 
See the log below, the bltouch think that it was triggered on the height 1.20mm, but there's nothing in the bed.

">>> g29
SENDING:G29
Error:!! STOP called because of BLTouch error - restart with M999
Error:!! STOP called because of BLTouch error - restart with M999

Error:Probing Failed
Error:Probing Failed

echo:endstops hit:  Z:1.20 "

And this problem ONLY occurs in the second measurement (The slow measurement).
I've teste all configs on the marlin for BLTOUCH.

On the oscilloscope, i can see a lot of pulses on the BLtouch switch pin, while it only should pulse one time, when the sensor touchs the bed. If i disconnect the "servo pin" from board, the ghost pulses on the switch pin stop.

J4nRyb1n, 2022-06-26:
May be I have a same problem with another firmware...

If you will be make a new firmware, look at:
1033 - #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and uncomment it

it's helped me...

haqbany, 2022-12-02:
Working firmware is here 
 Please read the instructions under the video 

https://youtube.com/shorts/EK_teBw-dPw?feature=share