bigtreetech/BIGTREETECH-SKR-mini-E3 #661: Hotbed heating on startup

GMDoola, 2022-06-27:
Installed the SKR mini E3 V3 on my old Hanbot DRT printer. Seems to be working fine except for the hotbed. It starts heating up from the time you power up the printer. The heat keeps going up even if you try dropping it. I updated the firmware as well but no luck. All connections seem fine also. Can anyone assist.

radek8, 2022-06-28:
Just for info.
The positive voltage is permanently connected, the negative potential is switched

radek8, 2022-06-29:
Did you short the hotbed? Are you sure you know what you're doing?
This is how you must burn a Mosfet.
Did you use Merlin for Ender? Was this FW designed for your control board?
If you are upgrading, it is best to compile your own FW.

GMDoola, 2022-06-29:
OOPs no there is no short between hotbed and frame

radek8, 2022-06-29:
How you used Merlin. Do you have the configuration files you used?

radek8, 2022-06-29:
Do you have a Volt and Ohm meter?

GMDoola, 2022-07-02:
> Check Mosfet: Unplug the printer. Disconnect the bed heater from the SKR Perform the ohm measurement as shown.
> 
> [image]
> 
> How much did you measure?

Hi radek8

At a setting of 20 ohm on multimeter I get 0.05 and 20k ohm - 5.14 ohm

radek8, 2022-07-02:
If you had unplugged the bed, it means that you have burned mosfet.
There are two solutions. 
Solder a new mosfet, or use an external mosfet.

GMDoola, 2022-07-02:
Hi radek8

At a setting of 20 ohm on multimeter I get 0.05 and 20k ohm - 5.14 ohm

GMDoola, 2022-07-02:
what is the mosfet part number and where do I solder it.
I did not touch or short anything that could cause mosfet to blow.
there is no shotting I can see

radek8, 2022-07-02:
Yes, if the board is under warranty, repairing the board will void the warranty.

If the board is under warranty, contact the seller and request a warranty replacement.

GMDoola, 2022-07-02:
Thank you will do that

Thanks for all the time and help

radek8, 2022-07-02:
Close the ticket if you no longer need it.

GMDoola, 2022-07-03:
Just installed another new board. doing exactly the same.
heating bed at statup
Made sure that nothing can touch to cause and sort of shorting

radek8, 2022-07-03:
> OOPs no there is no short between hotbed and frame

You have a short circuit between the hopny teles and the printer frame. You wrote that you controlled it.

GMDoola, 2022-07-03:
how do I do this

do you have a video or photos

GMDoola, 2022-07-03:
could it be the old bed or sensor settings

radek8, 2022-07-03:
Do you have the bed connected via an external mosfet?
But you didn't tell me that... :-(
Draw a block diagram of how you have it connected.

radek8, 2022-07-03:
They also sent photos of the connection

GMDoola, 2022-07-03:
did not know. found it wwhen was checking cables.

sending phoyos shortly

radek8, 2022-07-03:
Why are you using an external mosfet?
Does the bed have a different voltage or too much current?

GMDoola, 2022-07-03:
the hanbot I have came with it

GMDoola, 2022-07-03:
[image]
[image]
[image]

GMDoola, 2022-07-03:
[image]

radek8, 2022-07-03:
Where did you connect this connector?

[image]

radek8, 2022-07-03:
What resistance does your bed have?

GMDoola, 2022-07-03:
> What resistance does your bed have?

with printer on or off

GMDoola, 2022-07-03:
at a setting of 20k ohm - 9.88 ohm

radek8, 2022-07-03:
12 : 9,88= 1,21
1,21 * 12 = 14,5
What did you measure?
Your bed would only have 14.5W
I want to know what power your bed has. What current flows into the ostele?

GMDoola, 2022-07-03:
> 12 : 9,88= 1,21A 1,21 * 12 = 14,5 W What did you measure? Your bed would only have 14.5W I want to know what power your bed has. What current flows into the ostele?

how do I messure this?

GMDoola, 2022-07-03:
[image]

this is what I tested

radek8, 2022-07-03:
I probably won't get much information from you.

Your power supply gives a maximum of 17A. So for the bed it will be a maximum of 15A
Even the Mosfet on the SKR should withstand that.
Try disconnecting the external Mosfet and connect the bed directly to the SKR board on the HB connector

radek8, 2022-07-03:
[image]

GMDoola, 2022-07-04:
Thanks that worked
Now getting error below as soon as you preheat and temperature goes above 210deg

[image]
[image]

do you have instructions how to level the bed for the skr mini e3 v3

radek8, 2022-07-04:
This error occurs if the thermal runaway protection on the ecttruder, which is integrated in Marlin, is activated.
There can be more problems that it causes.

Either you have the protection set to fairly critical values, or there is a problem with the thermistor or the heating element.

Send the Marlin configuration files you used to compile Marlin.

You wrote that you changed the print head...
Did you happen to use a printhead with a 24V heater?
Did you set the right type of thermistor in the Marlin?

GMDoola, 2022-07-04:
check the bed leveling as well

radek8, 2022-07-04:
No! No! No! 
OMG 12 V neb 24V
You bought perhaps only one in particular!

GMDoola, 2022-07-04:
> No! No! No! OMG 12 V neb 24V You bought perhaps only one in particular!

?

radek8, 2022-07-04:
Also send me the file
Configuration_adv.h

GMDoola, 2022-07-04:
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/9040555/Configuration.zip)

radek8, 2022-07-04:
What voltage is the print head for?

radek8, 2022-07-04:
In Marlin, you have the power supply wrongly defined
set up
#define CHOPPER_TIMING CHOPPER_DEFAULT_12V

GMDoola, 2022-07-04:
BIQU H2 V2.0 Extruder Kit

Extrusion method: Double gear extrusion
Maximum printing temperature: 270℃
Weight: 211g (including heating aluminum block, fan)
Maximum extrusion force: 7.5kg (depending on filaments)
Extrusion capacity (based on existing): 600mm³/min(depending on filaments)
Number of pulses: 932steps/mm for stepper motor drive at 16 subdivisions  (micro-steps) (further correction is required)
Recommended motor current: 800mA
Filaments' diameter and tolerance: 1.75±0.05mm

Does not specify voltage nor on the box

radek8, 2022-07-04:
Is there a fan included in the print head?
Did the fan come with the head?

radek8, 2022-07-04:
Or can you measure the electrical resistance on the heating element?

[image]

GMDoola, 2022-07-04:
[image]

You are right the fan is 24v

GMDoola, 2022-07-04:
should I upgrade psu to 24v or change fan to 12v

GMDoola, 2022-07-04:
> Or can you measure the electrical resistance on the heating element?
> 
> [image]

At a setting of 20 ohm on multimeter I get 15.5 and 20k ohm - 0.01 ohm

GMDoola, 2022-07-04:
the heating block was not tapped so can't install grub screws, so I secured it with cable ties so cable can't slip out especially thermistor

radek8, 2022-07-04:
> the heating block was not tapped so can't install grub screws, so I secured it with cable ties so cable can't slip out especially thermistor

I don't understand what you write.

radek8, 2022-07-04:
Your problem is clear
Extruder heating is for 24V
If you connect it to 12 V, the heating element that should have 40 W / 24 V only has 10 W / 12 V
10W is not enough to heat the Hotend to a temperature lower than 210°C
Therefore, the printer stops.
You have to buy a new 12V heater and fan

If you would like to change the power supply to 24V (24V is better :-),
you would have to make a bed too. The bed is designed for 12V and with a 24V power supply it would consume 4x more...
From the current 110W bed, it would be 440W and that would burn up the heating.
24V source + 24V bed is an expensive upgrade

radek8, 2022-07-04:
After you solve the power problem, I recommend calibrating the thermistor.
you have type 1 set in FW, but that certainly won't match your thermistor.
When I changed my thermistor, the display showed 210°C, but I measured 230°C on the hotend with an external sensor.
After calibration to the correct value, the temperature is against reality with an accuracy of 1°C.

When you have the right temperatures, you can deal with leveling.

Unfortunately, I don't use MESH_BED_LEVELING. 
That's why I won't advise you how to use it.
It is too imprecise.
I recommend you buy a BL-Touch or a 3DTouch clone.

radek8, 2022-07-04:
I recommend enabling this option
#define M115_GEOMETRY_REPORT

radek8, 2022-07-04:
Before using the printer, calibrate the extruder and do a PID calibration of the Hotend and the bed.
Then save the calibrated values to the FW and recompile the FW

GMDoola, 2022-07-04:
Will try all this and let you know

AustinA30Lover, 2025-06-27:
Hi, 
Am having a similar issue- did you solve this problem?
thanks