bigtreetech/BIGTREETECH-SKR-mini-E3 #663: [BUG] No Printer Attached-- SKR Mini E3 V3.0 and TFT35-E3 V3.0

tjteagle, 2022-07-07:
Config files and wiring photos below
Ender 5 plus w/ BLTouch
SKR Mini E3 V3.0
TFT35-E3 V3.0

I assembled my hardware about 1 week ago and have been fighting serial communication issues since. I started troubleshooting by flashing updated firmware from BTT github without success (BIGTREE_GD_TFT35_V3.0_E3.27.x.bin and Marlin 2.1 and 2.0.x branches). I started reading up on experiences by others on github and reddit and I could not find anything conclusive. Here are my steps up until now:

- Began configuring existing ender 5 plus firmware (in case I want to use Marlin mode in the mean time)
- Confirmed all Marlin dependencies are defined in my Marlin firmware
- Configured serial port 1 at -1 to 3 and serial port 2 at -1 to 3 and possible matches of that set
- Flashed baudrate at 115200 in Marlin and changed on TFT through settings>connections>serial ports
- Followed same procedure with baudrate @ 250000
- Flipped TFT connector around and this was clearly wrong-- two lines across screen and messed up visuals

Anything else I could be missing here? Any help would be appreciated

https://pastebin.com/VHYEQ7f1
https://pastebin.com/a7EJqhQa
[image]
[image]

radek8, 2022-07-07:
You have the wrong serial port in Marlin.
You have to set serial port 2
for example:
#define SERIAL_PORT 2

tjteagle, 2022-07-07:
Apologies, I thought I set the serial port settings back to original before posting. Just flashed this config file to no effect:

https://pastebin.com/DQXg35M1

New ribbon cables just arrived so will be attempting those now.

radek8, 2022-07-07:
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120

radek8, 2022-07-08:
I still have a question.. 
Will the bin file be renamed to cur after the marlin update? 
After switching to marlin mode, is marlin functional?

radek8, 2022-07-08:
If you have ruled out all problems and the possibility remains that there are faulty pins, then it is best to solve the problem with the seller as a guarantee.
The problem is that you don't know whether to complain about the TFT bebo SKR. A faulty pin can be on one or the other. If you could try the display on another printer, you would find out what is wrong.
I've encountered bad pins on SKR as well, but the problem is more common on TFT.
There are several threads where this has been addressed.

barabuzza, 2022-09-08:
> Config files and wiring photos below Ender 5 plus w/ BLTouch SKR Mini E3 V3.0 TFT35-E3 V3.0
>
> I assembled my hardware about 1 week ago and have been fighting serial communication issues since. I [...]

Hello. I also have the same motherboard and display, I'm trying to connect it all to an Ender 5 plus. Did you manage to solve your problem? I also have problems with the firmware

tjteagle, 2022-09-08:
> https://pastebin.com/VHYEQ7f1 https://pastebin.com/a7EJqhQa [image:
> IMG-0100]
> <https://user-images.githubusercontent.com/108652016/177667458-b4d40a76-51da-4728-9c8e-c5c3368e19bd.jpg> [image:
> IMG-0101]

Hello there,

After pulling my hair out with this issue over a month or so, I bit the
bullet and grabbed a rpi 3b+ from eBay and installed klipper/fluidd.

I did see some things online about a new firmware update but I did not dig
in enough to see if it was a solution for my BTT hardware. After all of my
troubleshooting with help from GitHub/Reddit/Facebook, I was led to the
assumption that a TFT pin was faulty. I would be curious to know if you
find a solution but I will be sticking with klipper.

Good luck

radek8, 2022-09-08:
@barabuzza
Create a new ticket. Write what problems you have, attach the configuration and photos of the connection.
We will try to find out what your problem is and how it can be solved.

radek8, 2022-09-08:
@tjteagle
Have you tried remapping the serial port pins to the TFT?

adamtoczek, 2022-12-13:
I went through all the connections problems between SKR mini e3 v3 and tft35. 
What worked for me:

#define SERIAL_PORT 2 //this is default setting for Ender 3 SKR mini e3 v3 from marlin configurations
#define SERIAL_PORT_2 1 // this was -1, tried different values from -2 but it would not compile. 

the bound rate is 250000, but you can change it any time from tft settings

hope it will work for you

natepotato, 2023-01-03:
Your serial port should be set at -1 because the skr mini e3 v3 is 32 bit and -1 is what the 32 but boards are set to. I had the touch screen working but messing with the firmware to try to get the creality touch screen working I changed what screen I had enabled and now I don't remember which one worked. Also you need to have serial port 2 configured correctly to

natepotato, 2023-01-03:
Ok so just tried something and both modes work. I have a custom built printer the same size as the Ender 5 plus with the skr mini e3 v3 and the btt 35 tft screen, and what worked was serial port-1, serial port 2 2, with both baud rates at 115200, and enabling the cr10 screen

AnnelotJ, 2023-07-03:
Hi, 
I got it work following natepotato post however when I put out instructions on the screen the printer does nothing. Any advise for me ?

radek8, 2023-07-03:
@AnnelotJ
what is "napotato"?
Give more information. Printer board type, display type.
Does the display show the temperature of the nozzle?

obsidiansro, 2024-08-05:
Hi,

I am having similar issue.

No printer attached!

Printer: Creality Ender 3 V2 with BL Touch
Board: SKR Mini E3 V3.0
Display: TFT35 V3.0

It was working with SKR Mini E3 V2.0 but after upgrade to SKR Mini E3 V3.0 board and display refuse to communicate.
I have tried to change Serial ports and matching baud speed with no success.

I would appreciate if someone with same setup could provide correct config advice.

tjteagle, 2024-08-05:
Hello,

I came to the conclusion that one of my pins was non functional. Instead of
working through customer service to replace, I simply transitioned to
klipper via fluidd

obsidiansro, 2024-08-05:
I was connecting the display to the wrong pins on the board.
So now it is working but only with stock BTT firmware.
When I try to upload compiled firmware, I still can not get it communicating.
I don't know how Serials variables should be set up.

SERIAL_PORT
SERIAL_PORT_2
LCD_SERIAL_PORT
AUTO_ASSIGNED_LCD_SERIAL

obsidiansro, 2024-08-05:
I think I will switch to klipper as well.