bigtreetech/BIGTREETECH-SKR-mini-E3 #664: support for mks mini 12864 v3 on SKR mini E3 V3

l0v3byte, 2022-07-10:
Hi,
I need help for wiring and correct setup for a good configuration on SKR mini E3 V3 and LCD MKS mini 12864 v3 someone can help me
Thanks to everyone

radek8, 2022-07-11:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650

l0v3byte, 2022-07-20:
@radek8 A thousand thanks I have created an adapter that allows you to use the DuPont from 1 to 5 pins without problems and that can keep them firmly in place. After doing some testing I will put it on thingvarse. Thanks again

KingBain, 2022-07-25:
@l0v3byte  whats youre adapter look like ? I went with a 3dprinted 2x5 dupont style connector. 

https://www.thingiverse.com/thing:3392377

l0v3byte, 2022-07-25:
@KingBain No i use the normal DuPont can you buying online or in local elettronic shop and with my adapter intersects with the original connector. Now I'm waiting for August 1st that I take my paycheck and buy the nozzles and some filaments, otherwise I can't finish it and on the 6th or 7th I put everything online don't worry

l0v3byte, 2022-08-05:
@KingBain and @radek8 This is my first version the next weekend i put the CAD file in my GitHub and to make it easy for everyone to make a cable that suits them even if they have never used a cad
[image]
[image]
[image]

l0v3byte, 2022-08-05:
To save time someone can tell me the order of the cables according to the color

DheerajKhajuria, 2022-12-03:
Took me a while to figure out the klipper configuration .

[board_pins]
mcu: mcu
aliases:
  EXP1_1=PA1,EXP1_2=PA15, EXP1_4=PA10,EXP1_6=PC14,EXP1_3=PA9, EXP1_10=<5V>,EXP1_5=RST, EXP1_9=<GND>,
  EXP2_2=PB8, EXP2_6=PD6 , EXP2_1= PD8, EXP2_5=PB5 , EXP2_3=PB9

[display]
lcd_type: uc1701
cs_pin: EXP1_3
a0_pin: EXP1_4
#rst_pin: PD6
contrast: 62
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2
## Some micro-controller boards may require an spi bus to be specified:
#spi_bus: spi
## Alternatively, some micro-controller boards may work with software spi:
spi_software_miso_pin: EXP2_1
spi_software_mosi_pin: EXP2_6
spi_software_sclk_pin: EXP2_2

[output_pin beeper]
pin: EXP1_1

[neopixel mks_mini12864]
pin: EXP1_6
chain_count: 3
color_order: RGB
initial_RED: 1.0
initial_GREEN: 1.0
initial_BLUE: 1.0