bigtreetech/BIGTREETECH-SKR-mini-E3 #665: Skr mini e3 v2 dead after running kilpper over uart

aSocialMenace, 2022-07-11:
### Description

Skr mini e3 v2 died after one print running octoklipper over UART. Octoklipper was running on a raspberry pi 0 w2 and was communicating and powered by the tft pins

### Steps to reproduce

could not reproduce.. only have one e3 v2 board :(

**Expected behavior**
expected to run klipper on my octoprint server and power it via the 5v on the TFT pins while communicating via UART also using the TFT pins

**Actual behavior**
printer spit out one respectable benchy and died while trying to heat up for a second print 

### Additional Information
I purchased this board a couple of months after it was released. 

It spent a few months running marlin then i switched over to a traditional klipper set up where i ran octoklipper on a raspberry pi 4 which was powered by usb and connected to the printer via usb. It ran like this for well over a year flawlessly like this

 (side note.. klipper killed a stepper motor driver in my skr mini e3 v1.2 i had running on a different machine but that was a known issue.. well not to me at the time)

 Any ways.... I wanted to clean up the look and i HATE usb cables.. So i switched over to a pi 0 w2 and printed a nice case for it to mount to the frame and set it up to run octoklipper. I used the TFT pins to both power the pi and handle the communication. and it looked beautiful! everything ran as expected... for one print.. then she died.

I have tested power coming into the board and i get the expected 24v and the fuse is good. However the board has no leds lit up and no power coming out of the board at all.. Any pins the should be putting out 5v are all dead. I get nothing from the screen.. no fans... shes just dead

it may also be worth mentioning that when i was running marlin i was useing a BTT TFT screen with it. but ditched the touch screen as they are not compatible with klipper when i switched over to klipper.  

I belive the board died trying to power my pi 0 w2. I chose the pi 0 w2 as i had power delivery concerns and i knew the pi 4 would be to power hungry for the skr mini board to power.. 

I will probably be ordering the skr mini e3 V3 here soon to replace this board. But can the V3 handle powering the pi better? or will it have the exact same power delivery system as the V2?

radek8, 2022-07-11:
If the problem is only with the 5V supply, the SKR could still be saved.
You ask if the SKR controller should power the Pi, but you didn't say what the Pi's consumption is (the maximum current it draws).

radek8, 2022-07-11:
Look here
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/556
Another board similar problem.
The 5V power supply has burned out.
Replaced by an external 5V module.

aSocialMenace, 2022-07-11:
> If the problem is only with the 5V supply, the SKR could still be saved. You ask if the SKR controller should power the Pi, but you didn't say what the Pi's consumption is (the maximum current it draws).

I am pretty sure all the power is dead on it. none of the fans work. but i am curious as to how the board can be saved if its just the 5v lines that are dead.

So i think the max current is rated at 2.5. However there is an article here https://www.cnx-software.com/2021/12/09/raspberry-pi-zero-2-w-power-consumption/ that does a deep dive on the raspberry pi o w2 power consumption and these guys struggled to draw 1.8 amps. The conclusion of the article was that a 1.5 amp power supply should be more than enough for most applications.

I dont know how much power i was using. I havent seen any information about how much power this pi draws when running octoklipper .. but thats all she was doing.. i didnt have any thing else running on it.. no camera or any other accessories. 

But that leads me to another question. What is the max output of the 5v on the TFT pins? how much current can it safely power? 

Thanks for responding so quickly too!

aSocialMenace, 2022-07-11:
> Look here [bigtreetech/BIGTREETECH-SKR-V1.3#556](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/556) Another board similar problem. The 5V power supply has burned out. Replaced by an external 5V module.
> 
> [image]

ok so after reading that post i tried a couple of things to see if my board could be saved to. First i plugged the main board into a usb power supply and the screen lit up! It was blank but still lit up... Life? well lets see

i noticed the pi still wasnt receiving power tho so i plugged it into its own power supply and she fired up and suddenly the printers screen came to life with the usually klipper display!! oh fingers are crossed!!

Then i opened a browser and pulled up the octoprint ui and clicked connect.. and wouldnt ya know it she connected! I mean i am reciving temp data and everything! full on MCU communication here.. holly sh!# this might work!!

still tho.. no fans running.. oh well loaded up a print and let her rip! but sadly she didnt print... the bed temp never changed and she eventually canceled print with an error message.. bed not heating as expected... the temp stayed flat the hole time..... 

any thoughts??

Bazmundi, 2022-08-29:
Thoughts?! Going [by this](https://www.youtube.com/watch?v=l7gjxqTY1Rk) I just set up an RPi Zero 2W with OctaPrint, and did wire up the RPi to my BTT SKR MINI E3 V2, just waiting on a power supply.  Then found this "issue", so I am back tracking a little on this setup.  Although, it looks as much as the BTT boards generally may have problems?  But the idea that you could just jump on the serial port to drive a SBC may be poorly thought out.   Do we know what current you can actually draw on the serial port power pins?

aSocialMenace, 2022-08-29:
I reached out the BTT  and ask them. Their response was that you could not power the pi. 0w2 from the skr mini as the 5v it supplies is only capable of pulling 1.5 amps. However I did it anyways lol.. it has been working just fine for me for a a couple of months now. I've been meaning to come back to this as I am fairly confident that this is not what killed my E3 V2. After digging the board out of the printer and taking it closer, look at it it appears that the positive screw terminal coming into the board melted And that was the cause of failure. I just don't think the extra amp that pie is consuming would have been enough to cause this malfunction. I believe not taking the time to add Ferrell connectors to these wires was a contributing factor. It was a very well used printer and I just had the factory tinned wires screwed down into terminals. But like I said, I've been powering my Raspberry pi 0w2 from the 5v pin on my TFT header for a while now.. probably printed a couple hundred hours on it so far. I haven't once experienced a notification inside a octoprint that my pie is underpowered nor have I experienced any malfunctions from running it like this. Officially they tell you it's unsupported but the pie is unlikely to More than 1.5 amps anyways. I know octoprint will inform you if your pie is underpowered.. I used a pi 4 with another setup I have and I just used an old phone charger for the power supply I kept getting a constant nagging notification in the top right window of my octoprint informing me that my power supply was insufficient.. until I bought a new power supply that is. Hope this helps

Bazmundi, 2022-08-29:
LOL! Now you tell me, I've two daughterboards coming anyway.   

And thanks for the wiring warning :)

aSocialMenace, 2022-08-29:
Lol sorry.. ya I've been meaning to come back to this and post an update.. but I kind figured that it was such an uncommon thing.. No one's probably doing it.. we wouldn't help anybody to update my status.. so i just kinda put it off.. but if your anything like me it won't take you long to put those other boards to use