bigtreetech/BIGTREETECH-SKR-mini-E3 #668: CR10 S5 - BTT SKR mini E3 V3 - BLTouch not working

d-e-f-p, 2022-07-17:
I have upgraded my CR10S5 to BTT SKR mini E3 V3 but I am having no luck to make it work. I tried several firmware (marlin, marlin from BTT) with different configurations or connections (bltouch port, z-axis) but I always have one of this errors on homing:
1. No axis movement - Corrected switching BLTouch signal cables.
2. Seems to work fine, deploys probe, starts lowering z, but it doesn't stop and crashes to bed.

Any advice? Are you using this boards with same printer? How can I check or monitor what is going wrong?

I'm starting to give up and thinking about using stock board again :(.

Thank you in advance!

d-e-f-p, 2022-08-03:
The problem was a defective extension cable.

closed 2022-08-03 by d-e-f-p

seiya22, 2022-08-06:
Hello I just found your comment on GitHub while doing research on the Internet I am in the same situation as you I have a creality cr 10 s5 with a BTT SKR MINI E3 V3.0 motherboard and a BLTouch. I use visual studio code to generate the firmware for my machine, but I always have an error message

#error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
on line 81.

I used file Marlin-bugfix-2.1.x and Configurations-bugfix-2.1.x.
can you help me find a solution or possibly share me your Files, thank you very much for your help. cordially Benjamin from France

seiya22, 2022-08-06:
@d-e-f-p 
Hello I just found your comment on GitHub while doing research on the Internet I am in the same situation as you I have a creality cr 10 s5 with a BTT SKR MINI E3 V3.0 motherboard and a BLTouch. I use visual studio code to generate the firmware for my machine, but I always have an error message

#error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
on line 81.

I used file Marlin-bugfix-2.1.x and Configurations-bugfix-2.1.x.
can you help me find a solution or possibly share me your Files, thank you very much for your help. cordially Benjamin from France

d-e-f-p, 2022-08-07:
You have to manually take values from configuration file to Marlin branch, except for CONFIGURATION_ADV_H_VERSION and any other parameter that is only in one of the files (that parameters are from the different version).

I used a comparison program like Winmerge to do it. I can't share the files if you still need them.

seiya22, 2022-08-07:
thank you for your answer I managed to find why the Version of Marlin was incompatible with the ADV configuration file, there was a typo on the version which was in 2.1 instead of 2.1.1 , from there the files have compiled correctly in visual studio code I flashed my motherboard but unfortunately no command works on my machine with this frmware for the moment I am totally lost and I do not know how to fix all this thank you for your reply and sorry for bothering you with all this.

d-e-f-p, 2022-08-07:
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/9277783/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/9277785/Configuration_adv.txt)

This are the files I used. BLtouch configured for z-homing with 3 servo pins on skr specific connector and 2 switch pin to z-endstop.

seiya22, 2022-08-08:
thank you very much this should get me out of my bad situation, my cables are connected in the same way.
thank you so much.

d-e-f-p, 2022-08-08:
Review z-end-stop pins polarity. If they are reversed motors won't move (at least in my machine it was like that)

seiya22, 2022-08-08:
ok i will change that, i have a massage on the top of my screen that says no printer attached!
there is definitely something out of place, I'm going to search the forums for this error message, it's because of this that nothing reacts on my machine.

seiya22, 2022-08-08:
[image]
[image]

d-e-f-p, 2022-08-08:
It seems that the screen is not detecting the motherboard. There are some configurations to be made in firmware to make it work, but I'm using stock screen so I can't help with that. Sorry :(.

seiya22, 2022-08-08:
that's the impression I had there are a lot of forums on the question, I should be able to find my answer, thank you I'll let you know

seiya22, 2022-08-09:
hello, d-e-f-p
Everything works on marlin, all I have to do is fix the communication problem between the tft 35 touch screen and the motherboard.
the marlin function is connected with the cable from the port of the old screen and the touch part with another connector on the tft port, I think that in marlin the com port of this connection is poorly defined.

seiya22, 2022-08-09:
[image]

seiya22, 2022-08-09:
[image]