bigtreetech/BIGTREETECH-SKR-mini-E3 #669: [BUG] PWM fan on SKR mini E3 V3

SEModer, 2022-07-21:
Hi!
I'm try add hardware PWM to Marlin config To fan0 or fan1 or fan
And get this error.
[image]
firmware compiled correctly for STM32G0B1

When fans are connected, with software PWM, the fans either shake hard or they squeak if try change cycle_time.
tried Sunon 24v maglev, junon 4020 dual ball 24v

Pmant, 2022-07-22:
Hardware PWM currently is not implemented for STM32G0 chips. Check [my fork](https://github.com/Pmant/klipper/tree/stm32g0_hardware_pwm) which enables hardware PWM for the fan headers on a SKR mini E3 V3. I'm running a Sunon 5015 for part cooling like this:
```
[fan]
pin: PC6
off_below: 0.07
cycle_time: 0.00003125
hardware_pwm: True 
```
The RPM range is not great for these coolers but still noticable. 

PC6 and PC7 can't have different cycle times when using hardware PWM on both. Usually PC7 should not use hardware PWM as it is used for heatbreak cooling.

daviddodd64, 2023-03-11:
Hi, I am also looking to use hardware PWM, but on an SKR Mini E3 v2.0. Does your branch work with this board? I want to hook up a laser to PC6 and the PWM frequency isn't high enough for raster engraving.