bigtreetech/BIGTREETECH-SKR-mini-E3 #673: [BUG] USB & UART not showing at times

inOut47, 2022-08-02:
### Description
My new E3 V3 board is, most of the times, not responsive. it means it will not show up on the USB device list or via UART.

its possible to flash the firmware and the .cur file is created.
at first i thoght that the board is bricked but after getting home and trying again on a different printer with a different PI it connected. 
however one or two power cycle later it's back to not showing up.

kernel log shows nothing (not even identifing the USB device is connected) and no klipper device in /dev/serial/
UART is not responsive (after flashing the appropriate firmware).

### Steps to reproduce

1. plug board to PI
2. Power up pi
3. power up board

**Expected behavior**

board shows up as a serial device

**Actual behavior**

nothing shows up

### Additional Information
it is possible to flash the device.
i tried flashing the firmwares from this repo (both klipper & marlin) and custom built firmware (klipper) but no luck, same results.

murphytodd4, 2022-08-03:
I have seen this as well. I started a support ticket and they sent me another board. Almost instantly it started to do the same thing. I have since stopped using the board because support became useless. I am starting to think this is more of a hardware issue than a software issue.

inOut47, 2022-08-03:
i have used a total of 4 boards and only one has showen this symptom.
i think i found the problem.
it looks like a few legs of the MCU are covered shorted with some tin.

i scrapped it off with a screwdriver and it seems ok but i am afraid that it may brake during work.
[image]