bigtreetech/BIGTREETECH-SKR-mini-E3 #674: Please make a firmware for ENDER 5 plus with E3 3.0

Propeople2, 2022-08-04:
Pleaseeeeeeeee!!!!!

Scoutprime, 2022-08-18:
Hi, same request :) i'm trying to create firmware for E5plus with CR10 stock display and BLtouch. So far, no luck. Still having issues with Z Axis. It would be great if BigTreeTech will create Marlin for E5plus

mcxpress80, 2022-09-22:
yes it would be nice

Pseudo-Nym, 2023-01-17:
Agree

Dazza2012, 2023-02-02:
I'm having issues with Bed Levelling stopping at different times during mesh setup attempts using this recently purchased SKR E3 MINI v3.0 board and BTT TFT 35 v3.0.1 screen. BigTreeTech states this board is suitable for Ender 5 Plus but there is no firmware available.
I cannot find any suitable Marlin Firmware or Config files on BigTreeTecch GitHub so far. 
Firmware files for this would be good including BL Touch options. Marlin Config file samples to work with suited for Ender 5 Plus is what I am really after! 
Help!
Thanks.

Gixie69, 2023-05-26:
Yeah for sure, We need some firmware for the E5+ Skr mini3 v3.0.1 Ive had to use TH3D firmware but its still not having it. Wont home, wont level Will try to print but the extruders not working. Im guessing its all in Marlin but im a trucker not a computer wizz. Real people need helping out BTT.