bigtreetech/BIGTREETECH-SKR-mini-E3 #676: E0_AUTO_FAN_PIN not set correctly in 2.11 release & bugfix branch

mikensan, 2022-08-09:
### Description
In the marlin configurations repository under Ender3 and Ender3v2, the configuration_adv.h has the E0_AUTO_FAN_PIN incorrectly set to -1 (disabled) where it should be PC7.

I assume bigtreetech maintains / contributes to the marlin/configurations branch for their boards, if this issue needs to be posted under Marlin's github page I can do so. 

### Steps to reproduce
1. Clone MarlinFirmware/Configurations
2. Compile firmware
3. Watch hotend fan never turn on at any temperature

**Expected behavior**
Expected Hotend fan to turn on at and above 50c as defined in the config file.

**Actual behavior**
Fan does not spin >=50c hotend temp.

### Additional Information
Corrected configuration_adv.h file using PC7 for E0_AUTO_FAN_PIN:
https://pastebin.com/ewq2Pdrd

maueroats, 2022-08-10:
You should specify which example configuration. Including a link to the wrong code on Github makes it easier to verify.

Stock Ender-3 tree has the PC7 as you say it should:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/71fee0824d60a2364ada46df6b831c0e9dcec7ed/firmware/V3.0/Marlin/Ender%203/Stock/Configuration_adv.h#L610

In fact, all three of the Ender-3 configurations in this repo have the correct pin.

maueroats, 2022-08-10:
Example of wrong config line in Marlin Configurations repository (Creality V427 board):
https://github.com/MarlinFirmware/Configurations/blob/b11c8225ce4df5d742fd6922d739c75aaa72f781/config/examples/Creality/Ender-3/CrealityV427/Configuration_adv.h#L631

Example of correct config line, different board same repo:

https://github.com/MarlinFirmware/Configurations/blob/b11c8225ce4df5d742fd6922d739c75aaa72f781/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%203.0/Configuration_adv.h#L631

mikensan, 2022-08-10:
Apologies - I was writing in a frame of mind and forgot the simplest thing, a link lol. 

This is regarding the Configurations repo in Marlin's github - but since it is for a BTT board I wasn't sure if I should post here with the understanding you guys maintain it or directly on Marlin's github?

This can be seen in both release and bugfix, but here it is in release: https://github.com/MarlinFirmware/Configurations/blob/release-2.1.1/config/examples/Creality/Ender-3%20V2/BigTreeTech%20SKR%20Mini%20E3%20v3/MarlinUI/Configuration_adv.h

Line 631

jkcab95, 2022-08-26:
I fixed this in my marlin config h adv and the hot end fan still does not come on however the controller fan 2 PB15 comes on when the hot end reaches 50c?

mikensan, 2022-09-08:
> I fixed this in my marlin config h adv and the hot end fan still does not come on however the controller fan 2 PB15 comes on when the hot end reaches 50c?

You may just have them physically connected in the opposite spots. Try switching them physically or swapping them in the config.

jkcab95, 2022-09-08:
Mike,
The pin assignments are wrong. Bigtree has them listed incorrectly. I changed them in the configuration and they now work correctly.
Thanks,

Jeffrey Kratz
Facilities ManagerAura Badge Co264 Clayton AveMonroeville, NJ 083431-856-881-9026 Ext: 125

mikensan, 2022-09-09:
Just a FYI - since you replied via email, your signature got attached and your business address and phone # were published.
Yup that's what I'm hoping this bug report fixes - the incorrect PIN assignments.

jkcab95, 2022-09-09:
Mike I edited out my info could you please do the same in your reply?
Thanks, Jeff