bigtreetech/BIGTREETECH-SKR-mini-E3 #677: Bed heater stuck on and no power to hot end

jmartens2, 2022-08-20:
Hi, I have a problem or two I would like to request help with. I was running a print and came downstairs to check and the printer was moving but I thought it was a clog. As I was trying to clear nozzle I notice the pla on the bed was really soft. I tried to turn hot bed off and the temp kept going up. Now I’ve run into a land of  fact finding. 
1 my hot bed is stuck on and heats at startup
   I’ve checked mofset on board according to a website and it is within range. I’ve checked wiring to hot bed, there no physical damage, I get continuity through the wires and a resistance of 2.8 ohms (seems low?), can this be a problem? It’s an Ender 3 pro bed
2 after I took board out of printer I just hooked up power and hot bed still on (bad mofset??) but it looks good with meter?
3 with board out I noticed a burn mark, it looks like mofset for fan 1 so I must have shorted the fan some how?  This shouldn’t affect the hot bed correct?
[image]
 3 also it seems as if the hot end stopped heating, I get a thermal runaway error on activation from touchscreen. I’m still working on this one. But it’s mofset looks good also.

I would like to request some troubleshooting help from anyone! 
Thank you if you read this!

jmartens2, 2022-08-21:
looks as if the mc74hct controller chip is not working. i cant control fans from pronterface. i cant control hot end either and hot bed is stuck on. ive tried flashing stock firmware from BTT website and since i have the board out of machine its hard to tell if it took. if the channel on the mc74hct chip is shorted closed that would keep the hot bed on correct? ive had my meter on the channel it is indeed on with 5vdc is what i get from the pins.  mofset seems to be fine. how can i check processor being stuck on?

dvlopez2010, 2022-10-25:
I am having a similar issue with my mini e3v3. The hot ends goes into thermal runaway. I have been playing with firmware to see if it is a firmware issue, no luck. Please let me know if you find a solution.

jmartens2, 2022-10-25:
No luck on my end. I bought a new board. I could communicate to it but the PWM interface chip wouldn’t output anything. Too small for me to solder

dvlopez2010, 2022-10-25:
Did you buy another skr or did you go with a different brand?

jmartens2, 2022-10-25:
Yup it was my fault I probably shorted out a fan or something. And I’m familiar with the board now. No new learning curve

jmartens2, 2022-10-25:
Sorry same board.

dvlopez2010, 2022-10-26:
Thank you. I reached out to btt and hopefully can get a warranty
replacement.

jmartens2, 2022-10-26:
Let me know if you even get a reply. I’m still waiting

Menethoran, 2024-04-21:
did anyone get a resolution to this? Im experiencing a similar issue and can absolutely guarantee it isnt my wiring (im running a sprite pro and i built the loom out, sealed all wires and ran wiring outside of machine body to avoid possibility of pinching wires.) My runaway is on the hotend. This is the second board im encountering this with (first had a thermal runaway to failure... i lucked out it wasnt a fire).

dvlopez2010, 2024-04-21:
The issue I found was the board itself was bad. Once I changed the board it
resolved the problem.