bigtreetech/BIGTREETECH-SKR-mini-E3 #678: [BUG] Adding 5th stepper in UART mode crashes firmware

RoP09, 2022-08-30:
### Description

I Added a 5th stepper motor driver (TMC2209 V2.0) to my SKR Mini E3 V3 using the  Marlin based FW from [Mrisoc](https://github.com/mriscoc/Ender3V2S1) .

Configuring the driver in Standalone mode works fine, but using UART mode crashes the FW after flashing resulting in an endless loop.

### Steps to reproduce

Configured like below:
[image]

**Expected behavior**
TMC2209 works in UART mode

**Actual behavior**
After starting up the printer the FW ends up in endless loop

### Additional Information
This happens regardless if the driver is connected or not. Non UART (Standalone) mode using the same pins works fine.

radek8, 2022-08-31:
But PD0 and PD2 pins are not RX2 and TX2.
It would be more appropriate to use
PD2 //RX5
PD3 //TX5

[image]

RoP09, 2022-09-01:
> But PD0 and PD2 pins are not RX2 and TX2. It would be more appropriate to use PD2 //RX5 PD3 //TX5
> 
Thanks for the reply! The comments are wrong...
Actually I was initially trying RX2/TX2 from the TFT port on the board (since they are "real" UART pins) but that lead to the same behaviour (board reset).

But I will try with these pins as well!

RoP09, 2022-09-02:
Same thing. Also results in endless loop of the FW if I connect it to PD2/PD3...

RoP09, 2022-09-04:
Update: Connected it still same as above to PD0 PD2 PD3 PD4 and PD5 now (for enable pin), and set the config differently now to mserial5 like below:

[image]

Now the board boots up, but then displays TMC CONNECTION ERROR on the display. The printer works fine for a couple of seconds, I can move the other steppers or home it etc. but without the 2nd Z Stepper and after some seconds it reboots again.

Something wrong with my connection?

Connected MS1 and MS2 to 3.3V, bridged both PDN pins and connected them to the RX/TX pins with 1K on TX and the rest like above. Anything I am missing?

RoP09, 2022-09-04:
Update: I now soldered also the small solder pad to enable UART on the Pin 5 like described in the thread linked in the previous post and also here https://de.aliexpress.com/item/33025720969.html?af=701906&gatewayAdapt=glo2deu.

Now I can even control the stepper from the printer menu even if the TMC connectino error is still displayed. But it seems like it is only half the speed of the other stepper now in this configuration?

radek8, 2022-09-04:
Did you use the same wiring as the other controllers?

[image]

radek8, 2022-09-04:
Did you enable it in the configuration.h file?
#define Z2_DRIVER_TYPE TMC2209

RoP09, 2022-09-05:
> Did you enable it in the configuration.h file? #define Z2_DRIVER_TYPE TMC2209

Yes exactly. If I put it as _STANDALONE, the driver works fine but not in UART mode then (with same wiring).

RoP09, 2022-09-05:
> Did you use the same wiring as the other controllers?
> 
> [image]

Not exactly. I have no 100 Ohms before TX/RX (only the 1K resistor I have). Also DIAG pin I have not connected and CLK + PDN_UART are also not connected to GND (did not read anywhere that this has to be done).

radek8, 2022-09-05:
> 
> Not exactly. I have no 100 Ohms before TX/RX (only the 1K resistor I have). Also DIAG pin I have not connected and CLK + PDN_UART are also not connected to GND (did not read anywhere that this has to be done).

So that's okay.

radek8, 2022-09-05:
Something is wrong here.
Active parameters should not be grayed out

[image]

RoP09, 2022-09-05:
I am not sure, think that was just a display issue of VS code here, cause normally if I don't define Z2_SERIAL_TX/RX_PIN it will give me compilation errors. If I define Z2_HARDWARE_SERIAL it seems it does not need those RX/TX pin defines and compiles fine.

radek8, 2022-09-05:
Have you also tried swapping the RX and TX pins?
Sometimes the problem is that the wires to the pins are reversed.
Does not connect TX to TX.
RX connects to TX and TX connects to RX. The one who transmits must connect to the one who listens and vice versa.

Next, I would try to use TX2 and RX2 and on the TFT connector and define it as MSerial2

RoP09, 2022-09-05:
No did not try swapping them, but I will try that and also give it a try with MSerial2 later!
Thanks for the hints!

RoP09, 2022-09-05:
Swapped PD2 and PD3 defines now without using MSerial, but only the RX / TX defines - same story (FW freezes, as written in the initial post).
Next I will try to exchange the actualy RX / TX cables and use MSerial5 again...

RoP09, 2022-09-06:
No success. Also swapping RX TX physical pins results in same error as before...

radek8, 2022-09-06:
If it doesn't work for you in UART mode, you can remap E0 as Z2 and use the external controller for the extruder in STANDALONE mode.
It's not ideal, but it's usable.

RoP09, 2022-09-06:
Thanks for the idea, at least E0 I need for LIN_ADVANCE in UART mode, but maybe another stepper (X or Y) could be remapped indeed...