bigtreetech/BIGTREETECH-SKR-mini-E3 #682: PA3 and PA2 not in pinout pdf

EduMota, 2022-09-09:
I'm trying to identify the pins PA3 and PA2 on the skr mini e3 v3, but they are not listed at board pinout pdf.

I want to check the connections to make sure that everything is ok, because I follow the setup recommended [here](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Klipper), but without result, my board not respond to klipper, and continue getting the error 
`mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer`

radek8, 2022-09-09:
PA2 and PA3 
There are RX and TX pins on the TFT connector.

EduMota, 2022-09-11:
Then I think some thing is off.

My printer.cfg 
[image]

My /boot/config.txt
[image]

My /boot/cmdline.txt
[image]

My connections on skr mini.
[image]

My connections on raspberry pi 3 b+
[image]

Pinout of raspberry:
[image]

radek8, 2022-09-12:
I won't help you. 
I don't use Raspberry or Klipper.

EduMota, 2022-10-21:
Ok, now it's working, for someone with the same issue, that is how I figured out.

I plugged an usb/ttl adapter on raspberry and my notebook, on Linux host executed:
~~~
minicom -b <baud rate> -D <serial>
~~~

I could find out that my raspberry was with a high baud rate 300000, so I just recompile Klipper and flash it on printer board, and in `printer.cfg` added the 
~~~
[mcu]
baud:300000
~~~

closed 2022-10-21 by EduMota