bigtreetech/BIGTREETECH-SKR-mini-E3 #683: After installing bl touch and updating firmware it does not want to auto home or level bed

masterfenix87x, 2022-09-12:
After having the skr mini v3 for awhile i decided to add a BL touch. Installed that and the firmware from github but i cannot get the ender 3 3d printer to auto home the z axis nor auto level the bed. it just sits there after auto homing the x and y axis.

DanielKulikowski, 2022-09-25:
if you dont have a marlin fix you might need to plug 2 cables from BL touch to a Z endstop
https://www.youtube.com/watch?v=dtwWnz2-5VQ

haqbany, 2022-12-02:
Working firmware is here 
 Please read the instructions under the video 

https://youtube.com/shorts/EK_teBw-dPw?feature=share

brianvoo, 2023-07-14:
Took me a while to get mine working. https://github.com/brianvoo/BIGTREETECH-SKR-MINI-E3-V3-BLTouch-ZHoming give this try.