bigtreetech/BIGTREETECH-SKR-mini-E3 #686: BTT SKR MINI E3 V2 with BTT MINI12864 V1.0 - only backlight working

Legiotech, 2022-09-29:
**Hi**
I'm trying to connect SKR MINI V2 with Mini12864, but best I can get is backlight working. I've made a custom cable to connect display to the board - then i checked 10 times if i made it right (according to #650/info in Marlin files). I have also followed recomendations for commenting/uncommenting different lines from #650 topic. I have set the contrast in _Marlin\src\lcd\dogm\marlinui_DOGM.cpp_ to 255: 
void MarlinUI::_set_contrast() { u8g.setContrast(255); }
[image]

If I am correct my cable is working fine, otherwise i wouldn't get backlight (at least gnd, +5 and one more cable are fine). Do you know any way to make my screen work?

radek8, 2022-09-30:
If I look at the pin file for your board, there is something I don't like...

Legiotech, 2022-10-01:
I checked it and indeed there is two pins in mainboard, that are different than in Marlin files. Sadly i don't see any pins for PA9 and PA10 on mainboard. Does it mean that it won't work? 
[image]
 files.

radek8, 2022-10-01:
I didn't mean this.

I don't have the SKR mini E3 board, nor the BTT MINI12864 display.
That's why I can't test the settings.
Send me your configuration files and I'll try to suggest changes if you want to test them.
configuration.h
configuration_adv.h
pins_BTT_SKR_MINI_E3_common.h

radek8, 2022-10-01:
The swap is here

[image]

RX2 = PA3
TX2 = PA2
According to the pin file, it's the other way around.

[image]

radek8, 2022-10-01:
Some pins ( PA9 , PA3) are defined twice. That won't be good either.
I would try to comment on these items:
#define LCD_PINS_RS PA9
#define LCD_PINS_ENABLE PA3
#define LCD_RESET_PIN PA10

[image]

Legiotech, 2022-10-02:
Thank you very much! I made changes that you talked about and it's working fine!!! 

[image]

radek8, 2022-10-02:
Congratulations. 
That's great. So what have you changed? I would suggest changing it in Marlin so that it works for others and they don't have to look for a problem as well.

Legiotech, 2022-10-02:
Modyfications in Marlin that I have done:
- In Marlin\src\lcd\dogm\marlinui_DOGM.cpp:
`void MarlinUI::_set_contrast() { u8g.setContrast(255); }
`
- Configuration.h:
`#define FYSETC_MINI_12864_2_1` - in line 2814

-pins (Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_common.h):
  after `#elif ENABLED(FYSETC_MINI_12864_2_1)` i've added `#define NO_CONTROLLER_CUSTOM_WIRING_WARNING`
and i've commented three lines: 
```
    //#define LCD_PINS_RS                     PA9     // CS
    //#define LCD_PINS_ENABLE                 PA3     // MOSI
    #define LCD_BACKLIGHT_PIN               -1
    #define NEOPIXEL_PIN                    PB8
    #define LCD_CONTRAST                    255
    //#define LCD_RESET_PIN                   PA10
```
From the Marlin files that was it i guess. In wiring i swapped two wires in TFT connector:
[image]

I think it was all. Thank you very much once again.

radek8, 2022-10-03:
@Legiotech 
I suggested the Marlin modification.
Thisiskeithb says no need to edit the file
Marlin\src\lcd\dogm\marlinui_DOGM.cpp
If no characters appear on the display, it is said that resetting the EEPROM is sufficient.
Can you please compile Marlin without this modification and try it?
Thank you

Legiotech, 2022-10-03:
I have compiled Marlin without changes in marlinui_DOGM.cpp and as it was before, there was no characters on the screen. I have plugged the usb cable and tried to connect to the board using Pronterface, but it haven't worked. After that I've installed a driver from [https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/mapleDrv-USBdriver](url) and it haven't worked as well. 
I have also mount mainboard into the printer and it turned out that the screen work only when i power the board via usb (but it may be solved by reseting eeprom as Thisiskeithb said). When i power it from the usb and then turn on power supply it works fine and i can use screen normally.

Sporkforce, 2022-10-27:
Hello, I am also trying to use the BTT Mini 12864 V1.0 on a BTT SKR Mini E3 V2.0. I am using the Marlin 2.0.8.2x SKR Mini E3 V2.0 Firmware and there is no 
#elif ENABLED(FYSETC_MINI_12864_2_1
Entry in 
Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_common.h

Do I manually insert that entry along with

 //#define LCD_PINS_RS                     PA9     // CS
    //#define LCD_PINS_ENABLE                 PA3     // MOSI
    #define LCD_BACKLIGHT_PIN               -1
    #define NEOPIXEL_PIN                    PB8
    #define LCD_CONTRAST                    255
    //#define LCD_RESET_PIN                   PA10
?

I'm no stranger to rewiring connectors or tweaking firmware, but for some reason I can't get my head around this one. Thanks for any help.

Legiotech, 2022-10-28:
I had the same issue at first, so I decided to use Marlin 2.1.x and used WinMerge to rewrite configuration.h and configuration_adv.h. It was quite fast operation. I don't know if adding some lines by hand will help, but probably not. You can try it and report if it worked. 
BTW I still have problem with connecting via usb to the mainboard, I will try to solve it, but now I have to put it aside for a bit.

Sporkforce, 2022-10-29:
I got Marlin 2.1.1 and updated its Config/Config_adv, along with the pins_BTT_SKR_MINI_E3_common.h and marlinui_DOGM.cpp. 
Havn't found anyone yet able to use USB connectivity though, after prowling the googles for a good bit.
Otherwise everything is accounted for and seems ready to go, custom cable is made, but for the life of me I can't get it to compile. It claims the wrong board is defined in Configuration.h . I read this may be a platformio board definition mismatch but Im still in the process of tracking this issue down. Were you able to get your firmware compiled without any fuss?

Legiotech, 2022-10-29:
Yes, I compiled my firmware without issues. I can't check my settings now, but if you won't solve the issue till wednesday I can check it for you. I was looking for some information about usb connection too and it seems to be not working.

radek8, 2022-10-29:
@sporkforce,
send the exact wording of the error.
Send all the files you edited:
configuration.h, configurtion_adv.h, pin file and more...

Sporkforce, 2022-10-29:
Oh thanks! I used Winmerge to copy over the custom settings. The printer is a uniqe setup so misc sizes and hotend/thermistor/extruder wont match a typical Ender 3 or CR10
Full error message after attempt to compile:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:247:8: error: #error "CAUTION! FYSETC_MINI_12864_2_1 / MKS_MINI_12864_V3 / BTT_MINI_12864_V1 requires wiring modifications. See 'pins_BTT_SKR_MINI_E3_common.h' for details. (Define NO_CONTROLLER_CUSTOM_WIRING_WARNING to suppress this warning.)"
  247 |       #error "CAUTION! FYSETC_MINI_12864_2_1 / MKS_MINI_12864_V3 / BTT_MINI_12864_V1 requires wiring modifications. 
See 'pins_BTT_SKR_MINI_E3_common.h' for details. (Define NO_CONTROLLER_CUSTOM_WIRING_WARNING to suppress this warning.)"    
      |        ^~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:315:2: error: #endif without #if
  315 | #endif // HAS_WIRED_LCD
      |  ^~~~~
Error: MOTHERBOARD is not defined in Configuration.h
[Marlin 2.1.1 Edited Files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/9894701/Marlin.2.1.1.Edited.Files.zip)

radek8, 2022-10-30:
The error says that you have used a display that requires a special connection cable.
You have made the cable, now you need to accept this warning and comment out this line in the pin file:

[image]

radek8, 2022-10-30:
Attention!
Your pin file appears to be corrupted.

radek8, 2022-10-30:
Use this pin file
[pins_BTT_SKR_MINI_E3_common.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/9895360/pins_BTT_SKR_MINI_E3_common.zip)

Also check the cable connection
there was a wiring error in the original pin file.

[image]

Sporkforce, 2022-10-31:
@radek8 I appreciate the help! With your file it compiled just fine and I was able to flash the firmware without issue. The screen still doesnt work unfortunately. I attached
[image]
 a photo of the pinout I am using. The encoder LED Lights as does the LCD backlight, but the screen is completely blacked out by the LCD. I changed the contrast from 255 to 125, but the blacked-out LCD does not change.

@Legiotech I did get USB connectivity working with pronterface. I had to go into Device Manager, into "Ports (COM&LPT) and I found the STMicroelectronics Virtual COM Port that showed up when I connected the board. I went into "Properties" with a right click, and then to the "Port Settings" Tab at the top. I switched the Bits per Second from 9600 to 115200. It was detected in pronterface just fine after that. Verified connectivity by moving a stepper back and forth. I was not able to upload firmware to it via USB though.

Sporkforce, 2022-10-31:
UPDATE: It is fully functional! I can't thank you guys enough. I had incorrectly put the RESET pin on Board Connector EXP1-4 to the Display's RESET pin EXP2-3 as seen in the previous post, It should go to LCD_RST EXP1-6. My final Pinout is as pictured. 
[image]
I also had forgotten to enable support for the Neopixel LED Driver on PB8 in Marlin 2.1.1 Configuration.h:
Line 3267
//#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE          NEO_GRBW
Was changed to:
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE          NEO_RGB

Everything seem to be good, I'll test for a while and see. So far I'm thrilled. Thanks again

Legiotech, 2022-11-22:
UPDATE: I have connected my screen as @Sporkforce showed in last post and now im after week of testing and all is working fine. When i wired my screen correctly even USB communication started to work.
Thank you both very much for help!

FilamentFused, 2022-12-13:
Hi there,

I have followed everything but i am still getting these errors

In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h:61,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:543,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h:247:8: error: #error "CAUTION! FYSETC_MINI_12864_2_1 / MKS_MINI_12864_V3 / BTT_MINI_12864_V1 requires wiring modifications. See 'pins_BTT_SKR_MINI_E3_common.h' for details. (Define NO_CONTROLLER_CUSTOM_WIRING_WARNING to suppress this warning.)"
  247 |       #error "CAUTION! FYSETC_MINI_12864_2_1 / MKS_MINI_12864_V3 / BTT_MINI_12864_V1 requires wiring modifications. See 'pins_BTT_SKR_MINI_E3_common.h' for details. (Define NO_CONTROLLER_CUSTOM_WIRING_WARNING to suppress this warning.)"
      |        ^~~~~
Error: Failed to parse Marlin features. See previous error messages.
======================================================================================= [FAILED] Took 1.64 seconds =======================================================================================

Environment      Status    Duration
---------------  --------  ------------
STM32F103RC_btt  FAILED    00:00:01.636
================================================================================== 1 failed, 0 succeeded in 00:00:01.636 ==================================================================================
xman@Xanders-MacBook-Pro Marlin-bugfix-2.1.x 2 % 

any help would be appreciated

stilkata8888, 2023-02-10:
> UPDATE: It is fully functional! I can't thank you guys enough. I had incorrectly put the RESET pin on Board Connector EXP1-4 to the Display's RESET pin EXP2-3 as seen in the previous post, It should [...]

i wired up like this dude and again i dont have anything on display every setting i change like him and still dont have display.Can someone help me ??

radek8, 2023-02-10:
@stilkata8888 
What version of the display and SKR do you have?

stilkata8888, 2023-02-10:
@radek8 e3 v2

stilkata8888, 2023-02-10:
and the cersion of the display is v1.0

radek8, 2023-02-10:
BTT MINI12864 V1.0?

stilkata8888, 2023-02-10:
yes !

radek8, 2023-02-10:
Send me your Marlin configuration files that you used to compile.
configuration.h
configuration_adv.h
pins_BTT_SKR_MINI_E3_V2_0.h
pins_BTT_SKR_MINI_E3_common.h

stilkata8888, 2023-02-10:
like zip files is it okay ?

radek8, 2023-02-10:
ok

stilkata8888, 2023-02-10:
[files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/10710001/files.zip)
there we go

radek8, 2023-02-10:
I have to go, I'll check it out later.

radek8, 2023-02-10:
The configuration looks fine.
Check if you have cables connected to the management.

stilkata8888, 2023-02-10:
i check the cables 5-6 times and its connected correctly to the display i only get backlight and nothing on the screen ;(

radek8, 2023-02-10:
So you can still try to edit this file.
And recompile.

V Marlin\src\lcd\dogm\marlinui_DOGM.cpp:

void MarlinUI::_set_contrast() { u8g.setContrast(contrast); }

Rewrite to

void MarlinUI::_set_contrast() { u8g.setContrast(255); }

stilkata8888, 2023-02-10:
i edit this file and show me error.The error is:#error "Please select only one LCD controller option." and i can't compile it.

radek8, 2023-02-10:
This modification should not cause this error.
Haven't you made another adjustment yet?

stilkata8888, 2023-02-10:
can the versions of the marlin can cause these errors ?

radek8, 2023-02-10:
If you use all files of the same version, then no

stilkata8888, 2023-02-10:
okay im gonna make the change you speak about and im gonna update you what happend.

stilkata8888, 2023-02-10:
again the same nothing on display and no backlights 😞

radek8, 2023-02-11:
@stilkata8888
I found one more error in your configuration.
You have to enable it in Configuration.h
#define NEOPIXEL_LED

stilkata8888, 2023-02-11:
man its okay everything i fixed it the display is running up perfectly and everything is just fine.Later im gonna upload the things i edited in firmware.

P1ayer2, 2023-02-24:
Hello all! 
After trying to use the BTT V1.0 12864 display on the newest marlin version (2.1.2) I discovered that the (Marlin-2.1.x-V1/Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h) file is completely different compared to the file referenced in this thread. I have found that DOGLCD_CS should be what was previously known as LCD_PINS_RS and DOGLCD_MOSI what was LCD_PINS_ENABLE. These had the PA9 and PA3 values respectively, and has been commented as it is supposed to be. However, I can not find the equivalent of LCD_RESET_PIN PA10. In fact, I can not find a PA10 command at all in this version. Is it something I should ignore? Have I commented the correct things otherwise? I am pretty new to marlin and I am doing a completely custom build so any help is much appreciated. Thanks in advance! 
// Navil.

stilkata8888, 2023-02-24:
does your display is turning on ? Have you got something on the display ?

P1ayer2, 2023-02-24:
> does your display is turning on ? Have you got something on the display ?

Hey! Thanks for your reply. I haven't received the display yet, I'm just doing the configuration of the board right now to use directly after receiving it. Do you want be to reply once I have it and can test? It is hopefully arriving this Monday.

stilkata8888, 2023-02-24:
which board are you using for your build ?

P1ayer2, 2023-02-24:
I am using the BTT SKR MINI E3 V2.0 for my rebuild with the BTT 12864 V1.0 Display as mentioned before :D

stilkata8888, 2023-02-24:
okay im gonna make u shematic of wiring and im gonna tell me you which things u need to change in the firmware to activate the display :)

stilkata8888, 2023-02-25:
> No way! Thank you so much!

when your display arrives connect it with the board and tell me does it turn on

shahabaz, 2023-04-04:
Same situation here, apparently I have got SKR Mini E3 V2.1 which seems to be similar to V2.0. I just got my display will try this out and update here

stilkata8888, 2023-04-04:
make things in this thread and ur diaplsy is gonna work :)

shahabaz, 2023-04-04:
> make things in this thread and ur diaplsy is gonna work :)

Sure I'm on it. I got side tracked by a solution from Smith3d [here](44a4bxedxPwGQtSnEzGzdejnCdtrDJqkJ6uqWwGbXmdx5zrjGdHivmr9pyd7x8rVQMgumwGPCfuwseieCg8tuN5jNNeMJPU) I was wondering which to go from. I will try things from here first and then get back.

Since I highlighted this can you confirm if both say the same ?

P1ayer2, 2023-04-04:
Hey again! I got it to work pretty easily. All I had to do was to wire it like it says in the Marlin documentation. It worked on the first try!

shahabaz, 2023-04-05:
It's working for me too. @Sporkforce v2 wiring helped me. SD card doesn't seem to be working plus need to check on USB..

[image]

radek8, 2023-04-05:
The SD card cannot work.  Your board has only the EXP1 connector

shahabaz, 2023-04-05:
> The SD card cannot work. Your board has only the EXP1 connector

How about the error speaker does it work ?

radek8, 2023-04-05:
The speaker is also not connected and configured.
Looking at the schematic, with a little modification to the wiring and pin set, the speaker could be wired up.

There probably won't be enough free pins on the SD card. So you have to use the SD card on the SKR board.

[34 more replies]