bigtreetech/BIGTREETECH-SKR-mini-E3 #694: [FR] Sprite extruder support

thethirdthree, 2022-10-25:
Running into an issue with the new Sprite extruder with my BL touch.  The much further forward mount is positioning the sensor off the build plate during the auto level feature.  

Is there a firware in the works for this?

dvlopez2010, 2022-10-25:
That is a issue with the probe offset. You should be able to adjust that in the settings to get it on the build plate. I used the offset for the s1 configuration when compiling the firmware.

thethirdthree, 2022-10-25:
I'm a little on the stupid side when it comes to that, could you point me in the right direction?

dvlopez2010, 2022-10-25:
I need more information on where I lost you? In the settings menu you
should be able to find a probe offset setting that allows you to adjust the
offset. If not you will need to compile firmware to adjust the offset.

thethirdthree, 2022-10-25:
I've got the z-offset dialed in, I'm having an issue with the y shift of the mount.   Where did you find that setting for the S1 config?

dvlopez2010, 2022-10-25:
I opened the example configuration folder for marlin and scrolled through
the 3300 lines of code until I found them. You can also search for offset
and it will take you there faster.

dvlopez2010, 2022-10-25:
> Ok thanks!
>
> just a sanity check you mean
>
> Firmware > V3.0 > Marlin > Ender 3 > bltouch-for-z-homing >
> configuration.h or configuration_adv.h and find where the BLTouch Y offset
> is defined?

Yes. I looked up the default offset I had sent to someone else a while
back:

The probe offset default on the s1 firmware is -37, -39

I did not send the z offset but there is a wizard for the z offset.