bigtreetech/BIGTREETECH-SKR-mini-E3 #696: [BUG] (short description)

getz97, 2022-11-08:
Hi, i need some Help !! I have SKR MINI E3 V2.0 motherboard, when my hotbed going close to desire temperature( about 1 degree before "t"') motherboard start "make noise like "'click clik click"'. 
The Led diode on board flick with same frequency.
Any idea why this hapens?
I use marlin 2.1.1 compilation, Ender 3.
Thank you.

radek8, 2022-11-08:
Can you send a video?

getz97, 2022-11-09:
i remove kabel from hotbed, no "klick" sound, The "klick" sound is hearing only when power is switch on to Hotbed.

radek8, 2022-11-09:
Send video with sound and Merlin configuration files

getz97, 2022-11-10:
https://youtube.com/shorts/qziJ2tTlAGg?feature=share

radek8, 2022-11-11:
Try changing the frequency of the PWM and bed heating.

getz97, 2022-11-13:
i find it :) - some time ago I put an EMF filter on one cable coming from the power supply. It rezonate to Impulses from PSU when PID working.

radek8, 2022-11-20:
Please close the ticket