bigtreetech/BIGTREETECH-SKR-mini-E3 #698: SKR Mini E3 V3.0.1 missing in active branch G0B1

michalwu, 2022-11-21:
V3.0.1 board is missing from default and active branch. 
If this is on purpose, then could you please explain if v3.0.1 will get any updated firmware/code?

ardiramelan, 2022-11-30:
> V3.0.1 board is missing from default and active branch. If this is on purpose, then could you please explain if v3.0.1 will get any updated firmware/code?

I cant build/compile the skr mini e3 v3 using platformio but for the other boards runs perfectly fine. Did you face the same issue ?

GreaterJoe, 2022-12-29:
I tried building Marlin for it for days and it just won't work. Same story with Klipper. Seems there's only one way to use this board, and that's the stock firmware without auto bed levelling.