bigtreetech/BIGTREETECH-SKR-mini-E3 #700: [FR] Add pre-compiled Marlin 2.X Firmware for Creality Ender 5 Plus.

LevelUp-ErikJ, 2022-12-01:
I am planning on replacing the mainboard for my Creality Ender 5 Plus with a BTT SKR Mini E3 V3.0, and I am having extreme difficulties figuring out how to configure the Marlin Firmware to work with my Ender 5 Plus. I see that there is an available firmware for the Ender 5 and Ender 5 Pro, and I was wondering if a pre-compiled Marlin build could be created and uploaded to make this upgrade easier.

pumasean, 2022-12-06:
me too lol

Dizzybird, 2023-02-13:
Is there a solution to this yet? I have been wanting to do this for awhile now. I would like to use the E3 V3 and the stock screen on the ender 5 plus. I know there are a lot of pwopl e with Ender 5 plus that would like to see an easy firmware install with BLtouch and stock screen.

Dix92, 2023-04-29:
Did you still waiting for a compiled firmware for an Ender 5 Plus?

I have one, but with a bigtreetech Touch display.

Let me Know if you are interested.

LevelUp-ErikJ, 2023-04-29:
I'm good for now. I'm using the Kersey Fabrications Firmware on my Mini
E3v2.

Erik H. Jacob

CEO/President, LevelUp Computers & Technology
***@***.***
(404) 585-8324
www.levelupcomputersandtechnology.com

ThunderClap86, 2023-06-28:
> Did you still waiting for a compiled firmware for an Ender 5 Plus?
> 
> 
> 
> I have one, but with a bigtreetech Touch display.
> 
> 
> 
> Let me Know if you are interested.

I am interested, but I have a voron clockwork 2/stealthburner and the BTT TFT35 v3.0.1 screen. I know how to make the adjustments to the screen, just need to tune the extruder.