bigtreetech/BIGTREETECH-SKR-mini-E3 #708: Bootloader for the SKR Mini E3 V3 (for STM32G0B1)?

Magpie-81, 2022-12-23:
I might have bricked my SKR Mini E3 V3.0.0 with some firmware updates.
It still shows power LED, and SD slot LED when powered and with a card, but the display does not show anything, USB does not show up as COM-Port on PC and firmware.bin files on the SD card dont get flashed (no status blinking LED) and don't get renamed to .CUR.
Of course i've verified the USB cable, SD card and everything else with different boards that i have available, and everything works fine with original creality and makerbase boards that i have.

Used an original STLink/V2 to connect to the microcontroller over SWD, and that seems to work fine too. It identified the mcu and can read out all the flash into a bin file, erase all the flash, and write + verify the bin file back in the chip without any errors.
I then tryed to do a complete chip erase again, and load one of the precompiled .bin files from the bigtreetech GitHub into the processor. (Starting at adress 0x08000000). That did not seem to resolve the issue tho, still no reaction from the board over USB/serial, nothing displayed on an LCD, still does not seem to flash from SD card...

Not quite sure what to do with the board, i suspect it is dead before it ever printed 1mm of filament. 
But the last thing i would like to try is to restore the original bootloader into the chip. So if someone could give me a .bin file with a full flash dump of his board, or at least the lower adress part where the bootloader is stored, that would be great.

swaaws, 2023-03-24:
I use a manta m8p with the same stm32. My solution build the .bin without the bootloader and flash with `make flash FLASH_DEVICE=0483:df11`

deckape964, 2023-03-24:
well done
                       -=Guy Fallon=-

cb-agotchi, 2023-03-26:
In need of stm32g0b1 bootloader for SKR Mini e3 v3 as well

ufoDziner, 2023-03-30:
Does this solve the klipper firmware_restart issues?  My Mini E3 V3 does not reliably restart without removing power.  Thanks.

cb-agotchi, 2023-03-30:
> Does this solve the klipper firmware_restart issues? My Mini E3 V3 does not reliably restart without removing power. Thanks.

I have no idea. I've not run into that issue nor heard of it.

ufoDziner, 2023-03-30:
> > Does this solve the klipper firmware_restart issues? My Mini E3 V3 does not reliably restart without removing power. Thanks.
> 
> I have no idea. I've not run into that issue nor heard of it.

Okay, thanks.  For reference: https://github.com/Klipper3d/klipper/issues/2675#issuecomment-619039980

Magpie-81, 2023-03-31:
> 

Can you upload that file somewhere on github (maybe your personal repository) and send me the link? I would like to try and fix my SKR Mini with it.
Thanks :)

cb-agotchi, 2023-04-08:
> > 
> 
> Can you upload that file somewhere on github (maybe your personal repository) and send me the link? I would like to try and fix my SKR Mini with it. Thanks :)

https://github.com/cb-agotchi/Bootloaders.git

puzari-amit, 2025-07-24:
> > > 
> > 
> > 
> > Can you upload that file somewhere on github (maybe your personal repository) and send me the link? I would like to try and fix my SKR Mini with it. Thanks :)
> 
> https://github.com/cb-agotchi/Bootloaders.git

Hi. Could you kindly share the bootloader, please?