bigtreetech/BIGTREETECH-SKR-mini-E3 #709: [BUG] BLTouch for Z-homing does not work on latest firmware release for SKR Mini e3 V3.0

TheJoshGriffith, 2022-12-30:
### Description

I have downloaded `firmware-ender3-bltouch-for-z-homing.bin` from here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin

Trying to home the Z axis using this firmware, the axis would raise by 10mm, and do nothing more.

If I manually sent a G29 command from either my BTT TFT35, or via OctoPrint, or any other means, it would result in everything working as intended, until the probe retracts, then I'd get an error from the mainboard itself simply saying "probing error".

I downgraded to an older build (a3076873ab0b34e1e33d44e56bc86c861c7e3432), and the same behaviour results in Z homing just fine.

### Steps to reproduce

1. Install the latest build for Ender 3 with BLTouch for Z homing
2. Try to home Z axis using BLTouch

**Expected behavior**

The BLTouch homes, and the home is set.

**Actual behavior**

The printer reports a "Probing Error" and calls kill().

### Additional Information

Here is a video of the G29 command being executed over OctoPrint to the printer running that firmware:

https://user-images.githubusercontent.com/6074574/210101271-4dbfde4a-cdb1-4e48-a976-a3364a404e52.mp4

As you can see, the probe deploys, it lowers until it hits the bed, and as it retracts the motherboard sounds an audible error, as "Probing Failed" appears on the screen (not visible in the video).

LaddiusMaximus, 2023-01-10:
I sort of found the fix for this if you are still looking.

AMiketta, 2023-01-11:
> I sort of found the fix for this if you are still looking.

please let us know what the Fix is...

LaddiusMaximus, 2023-01-11:
Aurora Tech on youtube https://youtu.be/dtwWnz2-5VQ  She breaks it down well. However the config changes she found on github will overwrite your configuration on your board. And when you flash your own config changes, it will wipe out the fix and you will be back to square one. I dont know how to resolve that part.

Turtleball, 2023-01-25:
I believe you have to go into VS code or some other editing software and edit the firmware to fix that. I believe there  is a g28 code in there somewhere that you need to uncomment and it will fix that issue. I just watched a video on that today but I can't remember which video it was or I would reference it for you. Good luck and let me know what eventually fixes your problem!

brianvoo, 2023-07-14:
Took me a while to get mine working. https://github.com/brianvoo/BIGTREETECH-SKR-MINI-E3-V3-BLTouch-ZHoming give this try.

Blisk, 2024-05-28:
I still have the same problem and can not solve it. Any idea?