bigtreetech/BIGTREETECH-SKR-mini-E3 #710: Unable to flash Klipper on BTT SKR Mini E3 V3.0

enricoxg, 2023-01-04:
Hi all i just wanted to update from my post from yesterday where i was and still unable to flash Klipper on my BTT SKR Mini E3 V3.0 
UPDATE 1/3/2022: 
I AM ABLE TO UPDATE TO A STOCK BTT SKR E3 V3.0 MARLIN FIRMWARE FROM THE GITHUB REPO. (CONFIRMED BY file rename from firemware.bin to FIRMWARE.CUR. Now the issue is that the board is not recognizing either my freshly compiled Klipper firmware or a generic firmware from the BTT SKR E3 V3.0 Kllipper.bin firmware.
1/2/2023 Post 
Hi all,   I am a serious standstill trying to install Klipper on my SKR BTT MINI E3 V3.0 (TFT35 installed as well) with CB1 and Pi4B adapter. I am currently at the part where I compiled my Klipper.bin firmware and renamed it to firmware.bin in an empty MicroSD card.   With the printer off, I insert the sd card, and turn on the printer. However, when I take the memory card out and put it back into my computer I still see that the file name DID NOT CHANGE to firmware.cur.   Things I tried:   -I tried using a pre compiled Klipper.bin from the BIGTREETECH SKR MINI E3 V3.0 firmware repository.   -Formatted and partitioned microsd card to 4GB and formatted to both FAT AND FAT32.   I am so confused at what is wrong.UPDATE 1/3/2022:

Kylejvh1, 2023-01-24:
I am also trying to do the same. No matter how I have tried to do the configuration the cb1 and the skr mini e3 v3 will not communicate.

mcrrein, 2023-02-19:
Experiencing the same problem here. Any updates so far?

Hirashi3630, 2023-02-25:
Same issue here, any updates?

burt23, 2023-03-31:
☝️

harsh1592, 2023-04-16:
Same issue, was able to revert back to marlin but cannot flash klipper. Going to have to send this back.

burt23, 2023-05-13:
I also had an extremely difficult time getting klipper onto the SKR Mini E3, but finally found a solution: log into UI interface of the Pi to resolve config issues in your printer.cfg file

In my case the screen did not work right out of the box, and left me thinking I bricked my printer. What I did to fix it was find the raspberry pi on the network ```nmap 192.168.1.1/24 -v```, which simply runs nmap across your assumed IP range. Change the address to match your network if necessary. Alternatively you can assign the Pi a static IP upon creation using the advanced settings inside the Raspberry Pi Imager tool. Once you've found the IP log into Mailsail UI (which I'm using in this case, highly recommend it, also available through the RPi Imager tool). You'll want to add a hostname to your PI so you can easily access it from the browser, as that's where you'll be making all your config changes. Make sure your using a high quality cable to connect your Pi to your printer. 

Once I logged in I was able to see that klipper was indeed on the printer, but the printer.cfg file wasn't being read correctly. The UI gave me very specific errors which I could google and add into the config. Most of them were required fields from the klipper docs. I recommend making your changes through the Machine tab and editing the file in your browser. Knock out the errors one by one and your printer will be working with Klipper. 

I hope this helps

TerranEarth, 2023-10-14:
The board does not tolerate an emergency power off, so the firmware process should be done slowly, disconnect all USB connections, turn off the board, give it time to stand for 5 minutes, then insert the sd with the prepared firmware and turn it on, watching the diode and behavior, let it stand for a minute and turn it off, remove the sd.