bigtreetech/BIGTREETECH-SKR-mini-E3 #711: [BUG] SKR mini E3 v3 - USB connection issues

kubanskyy, 2023-01-04:
### Description

When I connect board to RaspbberyPi/ PC via micro USB port, RaspberryPi/PC nothing happens, just like the USB port isn't working.
I have tried different usb cables, different size sd cards, sd cards formatted in different bytes, connecting 24v power to SKR, verifying klipper make, installing octoprint on a new sd card, etc. I have verified the klipper make config.
So the SKR will rename the firmware.bin to firmware.cur, however running a lsusb doesnt show any devices connected. Even when plugging in either SKR into my windows 10 desktop it doesn't even recognize a usb device. No audio notification or unknown device in device manager.

Not sure what to try next and doing some googling i've seen some similar issues but with the older versions. Anyways I'm here now seeking some help and hopefully can figure out how to get these working.

i found that producer could send a firmware which restores function of the USB port.

### Steps to reproduce

1. Turn off board and RPi 
2. Power up board
3. Power up RPi

**Expected behavior**
New USB device connected

**Actual behavior**
Nothing

JosephHobbs, 2023-01-23:
Have you tried shortly the USB_5V jumper?  I had this issue with my SKR mini E3 V3 connecting to Ubuntu on a Beelink T4 Pro and shorting that jumper fixed it...

CyreX541, 2023-03-31:
Had a similar issue when i upgraded my Ender 3 V2 to SKR Mini E3 V3.

While pretesting the SKR Board using USB, no other connected printer cables, shortened USB Power using Jumpers,
everything seemed fine. The Board showed up in device manager as expected.

In the next step i swapped the old Creality 4.2.2 Mainboard against the new SKR, mounting it inside the Printer.
Later on i connected earch relevant cable to the appropriate Slot e.g. Motors, Heater cartridges, endstop switches, CR Touch etc.
Then powering up, doing the chores for calibration and started an test print.
For now, everything works as expected using the SKRs SD Card Slot for printing.

As i wanted to try Octoprint, i connected an USB Cable.
Boom, similar problems as you mentioned.
Spent several days for troubleshooting and contacting the Support via email.

While connecting the USB Cable, the USB Port slighty touched the Metal Frame of my Printer, shorting the USB Port.
This led to an permanently damaged / dead USB Port.
In my Case using the SKRs SD Card Slot for printig, it still works correctly.

Lateron i ordered a new SKR Board, did the same procedure as described above.
But befor mounting it inside the printer i used a few stripes of electrical tape, isolating the Edges of the printers Frame.
Since then i´m happily using the USB Port for Octoprint.

apriady, 2023-07-16:
hi @CyreX541,
In your case above, are you still able to communicate with your printer via the UART port?

because I had the same problem as you (the USB port was damaged by touching the metal frame) but when I try to connect with the UART port it doesn't seem to work either.