bigtreetech/BIGTREETECH-SKR-mini-E3 #716: Ender 3 SKR Mini E3 V3+BigTreeTech TFT 35-E3 V3.0

lacaa75, 2023-01-09:
Dear fellow forum members!

I would like your help because I am a complete beginner.
I have an Ender 3 machine with an efy SKR mini E3 V3 motherboard and a BigTreeTech TFT 35-E3 V3.0 display.
Would someone be so kind as to make me an installation firmware.bin that I could use to install marlin 2.1.2? It currently has the 2.0.x bugfix.
Thank you in advance for your help!

Sergeyss2022, 2023-03-12:
https://github.com/Sergeyss2022/BIGTREETECH-SKR-mini-E3-CRtouch-SmartfilameynBtt.git

Sergeyss2022, 2023-03-12:
Сам только вчера собрал, вот тестирую.

Dummy-up, 2023-05-13:
You can find precompiled versions around the place but a lot simply don't work.  it is easier to do it yourself by following  ay youtube video or some other on line how to, you don't need much knowledge except there's nothing much on that comination and I am still scratching my head getting it going and have been trying for weeks. I can get it to compile but it won't flash or I can get fw files to flash but they are not what I want exactly. Very frustrating that the hardestg th ing to get working is two products from the same damn company.