bigtreetech/BIGTREETECH-SKR-mini-E3 #719: [ISSUE] SKR E3 Mini V3.0 & BTT TFT35 V3.0.1 on Ender 6

xIExodusIx, 2023-01-30:
Hi there, 
I own an Ender 6 and wanted to get better prints, more and better settings and more information about printer and printings.
That's why I bought the BTT SKR Mini E3 V3.0 and the BTT TFT35 V3.0.1. 
My issues, I have several, are:
1. When I switch the printer on or press the reset button on the TFT35 V3.0.1 the TFT comes up instantly and tells me: "No printer attached!" for about 5 to 20 seconds.
	Which are the correct serial ports (TFT and Mainboard) and baud?
2. The Ender 6 is a CoreXY printer, I got it all work with MArlin 2.1.2,  the BLTouch works too. But if there is some writing on the first layer(s) the writing is mirrored and so not really readable.
	How can I solve this problem? I tried with uncountable different setups for about three weeks now.
3. I have the problem that the hotend cooling fan (FAN0) runs the whole time and never stops. The other two fans working correct, the part cooling fan (FAN2) stops when not printing and the mainboard cooling fan (FAN1) only runs when the steppers are activ. 
         How can I solve this? I want the hotend cooling fan to stop when not needed.

Thanks for helping.

radek8, 2023-02-01:
external MOSFET 

https://www.aliexpress.com/item/1005002307486977.html?spm=a2g0o.detail.1000014.16.42384728kJ4alS&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40050.281175.0&scm_id=1007.40050.281175.0&scm-url=1007.40050.281175.0&pvid=861b2dd5-50ec-494f-bd9a-a25eaaa76bb9&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40050.281175.0,pvid:861b2dd5-50ec-494f-bd9a-a25eaaa76bb9,tpp_buckets:668%232846%238110%231995&pdp_ext_f=%7B%22sku_id%22%3A%2212000020011458939%22%2C%22sceneId%22%3A%2230050%22%7D&pdp_npi=2%40dis%21CZK%2133.79%2130.41%21%21%21%21%21%400b0a23ae16752800150115545e40f5%2112000020011458939%21rec

xIExodusIx, 2023-02-02:
@radek8 
Thanks a lot for your help. :-)

The aliexpress link does not work, because the item is not avialble at my region! I am german.

About the issue with the FAN0, to verify that it is the MOSFET, I will change it to an other port. 
I hope that ist not the mainboard or the MOSFET.

How to get the mirrored writing solved, you got no clue?

radek8, 2023-02-02:
You can't remap FAN0 to some random port.
you can only remap it to the port that switches the Mosfet transistor.
This port should also be capable of PWM regulation.
If you are not going to use the electronics or extruder fan switching, you can use (remap) FAN1 or FAN2.
If the transistor on FAN0 were faulty, it should not be a problem to resolder it.
[image]

radek8, 2023-02-02:
Mirroring prints.
I have no experience with the CoreXY printer.
Probably some choice in the configuration will solve it, but I don't know what.

I would try to turn the direction of one motor on the XY axes by swapping the two wires in the connector and then modify the Marlin to match the travel directions and end stops.

radek8, 2023-02-02:
https://groups.google.com/g/h-bot-and-corexy-3d-printers/c/eopGME-S3Y4

KosPar01, 2023-08-16:
Hello, did you get a working firmware? I've been struggling for 2 months and I can't get them to work together. Can you dump your files, please?

xXMephisto82Xx, 2024-10-01:
Hallo hätte eine frage hat irgendwer eine funktionierende Firmware SKR Mini E3 V3.0 & BTT TFT35 V3.0.1 auf Ender 6 oder SKR Mini E3 V2.0 den mir ist das original Board kaputt gegangen.. Ich kenn mich nicht so wirklich gut damit aus das selber zu schreiben die Firmware da ich noch recht neu bin mit 3d Druck. Vielleicht kann mir wer weiterhelfen. Lg

NeroSGV, 2024-11-02:
i have ender 6 with btt skr mini e3 v3 board and tft35-e3 touch screen with a biqu h2 v2s lite extruder with the biqu micro probe i have tried marlin and klipper to configure the firmware and they never work mainsail gives me the same error every time and when i i put marlin firmware on sd card and name it firmware.bin and making all the changes to the two config files on vs code it never works ever screen just says no printer attached and i use old macbook as a raspberry pi i build firmware with kiauh and mainsaill never can connect to board im not good at code and have limited skills with computersbut i have been trying for months and nothing works im exhausted and about to give up on everything and throw everything in the trash i have had my 3d printer my ender 6 for like 2 years and printed on it maybe five times     and why are there no good upgrades for the ender 6 its like they built it and sold it and then just forgot about it they make all these firmware files for ender 3 ender 5 but never ender 6 if i cant find help i will be launching this printer by catapult right in the dumpster a football field away as well as lighting it on fire and placing a curse on it as well cause im on my last ;eg with this thing please help

xIExodusIx, 2024-11-13:
Hello NeroSGV,
My Ender 6 with BTT SKR Mini E3 V3 Mainboard and BTT TFT35 V3.01 Display is working with Marlin FW 2.1.2.4, not 100% perfect but it works. It took me 3-4 month with googling and testing to get it work. 
If I can help, let me know how and I will try to help.

xIExodusIx, 2024-11-13:
Hello xXMephisto82Xx,
I have made me a Marlin Firmware 2.1.2.4 for my Ender 6 with BTT SKR Mini E3 V3 and BTT TFT 35 V3.0.1 that works. Not 100% perfectly but it works.

xXMephisto82Xx, 2024-12-11:
Hallo xIExodusIx
ich bin über jede hilfe dankbar die ich bekommen kann..würde deine version gerne testen.
LG

xIExodusIx, 2024-12-12:
> Hallo xIExodusIx ich bin über jede hilfe dankbar die ich bekommen kann.. würde deine version gerne testen. LG

Hallo xXMephisto82Xx ich kann Dir gerne meine Configurations-Files zur Verfügung stellen. Ich darf hier aber nur max. 25 MB hochladen. Die ZIP-gepackten Configs sind leider größer. 
Kennst Du Dich mit VSCode aus? Ist eigentlich kein Hexenwerk wenn man einigermaßen gut Englisch kann.

LG

xXMephisto82Xx, 2024-12-12:
> > Hallo xIExodusIx, ich bin über jede Hilfe dankbar, die ich bekommen kann.. würde deine Version gerne testen. LG
> 
> Hallo xXMephisto82Xx, ich kann Dir gerne meine Konfigurationsdateien zur Verfügung stellen. Ich darf hier aber nur max. 25 MB hochladen. Die ZIP-gepackten Configs sind leider größer. Kennst Du Dich mit VSCode aus? Ist eigentlich kein Hexenwerk, wenn man einigermaßen gut Englisch kann.
> 
> LG

Ja kenn mich dort ein wenig aus werde das schon hin bekommen..Hab jetzt e mehr Zeit im Urlaub..

xIExodusIx, 2024-12-13:
> Ja kenn mich dort ein wenig aus werde das schon hin bekommen.. Hab jetzt e mehr Zeit im Urlaub..

Alles Klar, wie kann ich Dir die Dateien schicken?

spcasey6907, 2025-05-02:
Did you get this going ? With stock display

spcasey6907, 2025-05-02:
> > 1. When I switch the printer on or press the reset button on the TFT35 V3.0.1 the TFT comes up instantly and tells me: "No printer attached!" for about 5 to 20 seconds.
> >    Which are the correct serial ports (TFT and Mainboard) and baud?
> 
> This is normal behavior. The display boots faster than the board with Marlin. Therefore, a message appears for a few seconds that the display is not connected to the printer.

Is the Ender 6 Stock display not compatible with  BIGTREETECH sku mini e3 v3?

spcasey6907, 2025-05-02:
Is the Ender 6 Stock display not compatible with BIGTREETECH sku mini e3 v3???